ros2/ros2 #734: ROS 2 Eloquent Elusor

wjwwood, 2019-06-19:
This meta ticket aggregates the tasks targeted for the next ROS 2 release, Eloquent Elusor.

Release date is TBD.

Tasks will be updated as they progress.

----

### Planned for Inclusion

- rclcpp:
  - general improvements
  - refactor executor
- rclcpp/rclpy parameters:
  - passing key-value parameters on CLI
- launch/launch_ros:
  - XML frontend
- ros2cli:
  - make daemon “reset” itself when the IP address changes (e.g. by DHCP)
- rviz:
  - support for interactive markers
  - finish migration of wrench display (torque ring)
  - support for use_sim_time
  - using message_filters (https://github.com/ros2/rviz/pull/375)
- rqt:
  - port parameter plugin
- diagnostics:
  - continue porting
- rosbridge_suite:
  - continue porting
- rosbag2:
  - continue improving
- buildfarm:
  - RPM packages for Centos
- Performance testing
 
----

### Stretch Goals

*We think that these items are important but don't currently have the bandwidth to commit to them; if you can help out, let us know!*

#### Architecture

- Node / participant mapping, move into context

#### Improvements

- Increase test coverage to reduce release-time manual testing
- Improve documentation content
- macOS support
- powershell support in ament_cmake
- performance of sourcing large isolated workspaces

#### Feature “holes”

- rviz/image_transport: support image transport
- Parameter: expanding constraints, set allowing to override
- 3/N-part type names in the resource index
- Expose QoS settings, graph API
- log4cxx revisit

#### Technical Debt

- ros2interface CLI tool (replacement for ros2msg/ros2srv)
- rclcpp improvements, high level structure
- Use pybind11 in rclpy, best practice
  - existing proof of concept: https://github.com/ros2/rclpy/pull/173
- API review

#### Infrastructure

- support rebase from forked repos in CI
- generate package-specific docs
- versioned documentation

#### Demo

- Support TurtleBot 2/3 with Dashing

----

We will continue to refine this list as we consider more input.

vmayoral, 2019-06-19:
Find below a first peek into @AcutronicRobotics contributions (to be edited and detailed as we progress):

- RT QA buildfarm
- Information models
- Manipulation
- Security

o-olalde, 2019-06-19:
From @aliasrobotics we have these contributions in mind:

- Best security practices for robot hardening

- Security tool development

- Security testing for applications on top of ROS-Eloquent

- Security workshops to the community

gbiggs, 2019-06-23:
From Tier IV we are looking at the following contributions initially, but we expect this list to change as we figure out more about our needs and resources.

- (Firm goal) Debian as Tier 2 platform
- (Fuzzy goal) Contribute to implementing deterministic publish/subscribe (coordinated via Real-time WG)
  - Possibly implement a wait-list-based executor (coordinate with ?)
  - Possibly implement callback groups functionality (coordinate with @wjwwood)
- Implement an `rmw` library for RTI's real-time/certifiable DDS (coordinate with Real-time WG/Apex.AI)
- Sample structured safety case for a robot application (coordinated via Safety WG)

robotpilot, 2019-07-16:
These are the efforts **ROBOTIS** is working on for ROS 2 Eloquent:
- [TurtleBot3](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/):
  - Support ROS 2 Eloquent
  - Support ROS 2 binary packages for TurtleBot3
  - Support Linux distro image based on Raspbian for ROS 2 Dashing and Eloquent with TurtleBot3
  - Porting ROS 1 package of existing TurtleBot3 to ROS 2 
    (https://github.com/ROBOTIS-GIT/turtlebot3)
- [turtlesim](https://github.com/ros/ros_tutorials):
  - Porting to ROS 2 (https://github.com/ros/ros_tutorials/pull/53)
- [ros2arduino](https://github.com/ROBOTIS-GIT/ros2arduino):
  - Support ROS 2 Eloquent and add examples and documentation
  - Support Arduino and [OpenCR](http://emanual.robotis.com/docs/en/parts/controller/opencr10/) (provide completely open-source hardware and software for ROS 1 and 2)
- ROS 2 Tools (rqt plugin):
  - We would like to join the development of the existing ROS 2 Tools based on rqt and contribute to the more diverse ROS 2 tools. 

(to be edited and detailed as we progress)

Karsten1987, 2019-07-17:
Bosch's contribution for Eloquent:

* diagnostics
  * obtain maintainership of the ros2-devel branch
  * finish porting core packages
  * releasing packages into Eloquent (and Dashing if possible)

* rosbags
  * continuing maintaining rosbags
  * further feature improvements (start, stop, looping)
  * handling QoS settings correctly

* ros2 executor 
  * Further research on possible ros2 rclcpp executor models
  * Release of EDF (earliest deadline first) executor 
  * rcl executor model for embedded ros2

JaimeMartin, 2019-08-04:
eProsima's contribution for Eloquent (to be edited and detailed as we progress):

Fast RTPS (Ongoing - Already partially supported in Fast RTPS 1.9 beta) 
- Static Memory Allocation
- Non-Blocking Calls
- Discovery Server
- Automated performance Tests on CI
- Dynamic Types Support.
Fast RTPS - Planned:
- Shared Memory Transport.
- Better DDS compliance
- Modern DDS C++ API

Micro-ROS (Planning now - TBD)
- Eloquent Support
- XRCE-DDS Peer To Peer comms

Integration Service (Planning now - TBD)
- New Release 
- Dynamic Types Support.

allenh1, 2019-08-05:
> Shared Memory Transport.

👏

thomas-moulard, 2019-08-06:
* **Developer Productivity**
    * Implement a mechanism to cross-compile ROS 2 nodes easily
* **Security Work**
    * Implement a ROS 2 node isolation method
    * Implement a ROS 2 security testing tool
* **Node Telemetry**
    * Expend ROS 2 so that node can provide performance (message latency, size...) information.
* **ROS 2 Maintenance**
    * ARM-HF architecture support (monitoring the CI and fixing as needed)
* **Improved ROS 2 support on selected robotic platforms**
    * Support TurtleBot 3 with Dashing with ROS 2 running in the robot [currently discussing with ROBOTIS how to share work]

Other considerations:
* Proof of concept for runing rviq/rqt in a browser

Those are only tasks we plan to contribute in Q3. Tasks for Q4 are TBD.

seanyen, 2019-08-06:
Microsoft's contribution for ROS 2 Eloquent: (to be edited and detailed as we progress)

* [Visual Studio Code extension for ROS](https://marketplace.visualstudio.com/items?itemName=ms-iot.vscode-ros)
  * Support ROS 2 Eloquent for existing functionality.
* [ROS on Windows build pipeline](https://github.com/ms-iot/ros-windows-build)
  * Host ROS 2 Eloquent Chocolatey binary release.
* [Azure IoT Hub Relay for ROS](https://github.com/microsoft/ros_azure_iothub)
  * Enable ROS2 support.

dejanpan, 2019-08-08:
Apex.AI contributions to ROS 2 E:

1. [ ] Documentation testing extension to launch_testing (in which the documentation files written in markdown are instrumented such that they become integration tests)
1. [ ] Support for ROS2 QNX (compilation and testing (we actually got python and thus colcon to run on QNX) => we will start with the design document first
1. [ ] Static code analysers (we are evaluating LDRA, TrustInSoft)
1. [ ] Upstream the velodyne driver from https://gitlab.com/AutowareAuto/AutowareAuto/tree/master/src/drivers to https://github.com/ros-drivers
1. [ ] Provide compile time option to include and link DDS types when the corresponding ROS 2 type is included/linked 
1. [ ] Add read_n/take_n/waitset to rclcpp. This will require EProsima to add some functionality. Apex.AI to be involved in the design document creation.
1. [ ] Loaned messages (ability to ask middleware for memory, related to read)
Apex.AI to be involved in the design document creation. 
1. [ ] additions to https://github.com/ApexAI/performance_test => details TBD
    1. collaborative effort (OR, Amazon, eProsima, Apex)

esteve, 2019-08-08:
@dejanpan I started a discussion for contributing our Velodyne driver to ros-drivers in https://discourse.ros.org/t/contributing-apex-ais-ros-2-velodyne-driver/10223/5

joespeed, 2019-08-12:
ADLINK contributions for ROS 2 Eloquent:

[rmw_cyclonedds](https://github.com/atolab/rmw_cyclonedds) and [cyclonedds](https://github.com/eclipse-cyclonedds/cyclonedds)
-   contribute [rmw_cyclonedds](https://github.com/atolab/rmw_cyclonedds) to ROS 2
-   [cyclonedds](https://github.com/eclipse-cyclonedds/cyclonedds) refactored to work with [ROS 2 CI buildfarm](https://ci.ros2.org/)
-   both integrated with ROS 2 CI buildfarm
-   documentation
-   Linux, OSX, Windows, [FreeRTOS](https://github.com/eclipse-cyclonedds/cyclonedds/tree/master/ports/freertos-posix). QNX planned 4Q 2019 for RT WG
-   added binding for test tool used in Real-Time WG test farm
-   DDS Security v1.1 planned for 4Q 2019

[rmw_opensplicedds](https://github.com/ros2/rmw_opensplice) and [opensplice](https://github.com/ADLINK-IST/opensplice)
-   ROS 2 Eloquent related enhancements & fixes
-   DDS Security v1.1 via commercial edition. Is work in progress for [opensplice](https://github.com/ADLINK-IST/opensplice) open source

tutorial with source
-   focused on AMR migration to ROS 2 Eloquent
-   joint effort by Rover Robotics, Intel, ADLINK, AWS RoboMaker, Open Robotics
- tutorial also via ROScon19 [day 1 talk](https://roscon.ros.org/2019/#day-1-october-31st) and interactive exhibition

shoemakerlevy9, 2019-08-13:
Rover Robotics contributions for ROS 2 Eloquent:

Driver 
- Support our ROS driver for Eloquent

Navigation2
- Testing
- Create PR in upstream packages as needed
- Create tutorials

bshin-lge, 2019-08-15:
Here are the planned LG Electronics contributions:

[meta-ros](https://github.com/ros/meta-ros) improvements and maintenance:
- Support for Eloquent distro
- Restructure repository to more easily support adding ROS1 packages to a ROS2 image

Target Tier 2 platform support for OpenEmbedded:
 - Buildfarm for OpenEmbedded images

mkhansen-intel, 2019-08-15:
Intel Navigation2 contributions / enhancements for Eloquent:
 - Navigation2 working with Cartographer
 - Multiple robots running Navigation2 simultaneously in same environment
 - Ability to set location on startup via launch parameter
 - Ability to save robot location as robot moves, reload last known position at launch
 - Enable Navigation2 system test in ROS build farm
 - Improve system stability to 95% system test pass rate

SteveMacenski, 2019-08-16:
Samsung contributions for Eloquent 
- Runtime configurable recovery actions
- Nav stack running TF for positioning & SLAM integration
- Refactoring slam toolbox & ros2 demos for mapping 
- continue maintenance of image pipeline and getting through backlog as time permits 
- Initial Nav2 interface definitions for plugins

jatowler, 2019-08-22:
CCDC GVSC contribution for Eloquent:

- [ ] Multi-machine launching (ros2/launch#79)

We additionally expect to target other launch and parameter issues in support of this goal.

kyrofa, 2019-09-04:
Canonical's contributions for Eloquent:

* Enhancements and design
    * [x] Migrate SROS 2 to cryptography library instead of openssl (https://github.com/ros2/sros2/issues/109)
    * [ ] Node IDL design
* Tools
    * [x] mypy ament linter (https://github.com/ament/ament_lint/pull/154)
    * [x] `ros2 param dump` (https://github.com/ros2/ros2cli/pull/285)
    * [x] `ros2 pkg xml` (https://github.com/ros2/ros2cli/pull/280)
    * [x] `ros2 topic type` (https://github.com/ros2/ros2cli/pull/272)
    * [x] `ros2 service find` (https://github.com/ros2/ros2cli/pull/274)
    * [x] `ros2 service type` (https://github.com/ros2/ros2cli/pull/273)
    * [x] `ros2 topic find` (https://github.com/ros2/ros2cli/pull/271)
    * [x] ros2 CLI interface prototypes autocomplete (https://github.com/ros2/ros2cli/pull/299, https://github.com/ros2/ros2cli/pull/300, https://github.com/ros2/ros2cli/pull/301)
* Ports
    * [x] ROS 2 port of teleop_tools (https://github.com/ros-teleop/teleop_tools/issues/34)

We're also doing our best to support multiple nodes per DDS participant in SROS 2, although this is dependent on the "node/participant mapping" stretch goal, and is also a stretch goal. This is in addition to various bugfixes and quality improvements across the stack as a result of scans and code analysis (e.g. https://github.com/eProsima/Fast-RTPS/issues/630, etc.).

TSC21, 2019-10-14:
Is ROS2 Eloquent supposed to be available by apt? I find over 430 possibilities to `ros-eloquent-*` except `ros-eloquent-desktop`, which is usually the one used for install all the other base packages. Is this on purpose or am I missing something? Thank you!

clalancette, 2019-10-14:
> Is ROS2 Eloquent supposed to be available by apt? I find over 430 possibilities to `ros-eloquent-*` except `ros-eloquent-desktop`, which is usually the one used for install all the other base packages. Is this on purpose or am I missing something? Thank you!

We're still in the alpha stage for Eloquent, with beta coming in the next couple of weeks.  You can view the full schedule at the bottom of https://index.ros.org/doc/ros2/Releases/Release-Eloquent-Elusor/

TSC21, 2019-10-15:
> We're still in the alpha stage for Eloquent, with beta coming in the next couple of weeks. You can view the full schedule at the bottom of [...]

@clalancette I see. Trying to understand the timeline: November 22nd, 2019 is the definite release date, but when exactly is `ros-eloquent-desktop` going to be made available? Before November 22nd or only by that date?

mjcarroll, 2019-10-15:
@TSC21 The variants package has now been released, so `ros-eloquent-desktop` will become available after the next sync: http://repo.ros2.org/status_page/ros_eloquent_default.html?q=desktop

TSC21, 2019-10-15:
> @TSC21 The variants package has now been released, so `ros-eloquent-desktop` will become available after the next sync: http://repo.ros2.org/status_page/ros_eloquent_default.html?q=desktop

Excellent! Thans @mjcarroll!

vmayoral, 2019-10-21:
Following from the commitments and the entry Alias Robotics [submitted a few months ago above](https://github.com/ros2/ros2/issues/734#issuecomment-503457205), here's an update from our side:

We originally committed to:
- [x] Best security practices for robot hardening
  - [Contributed to design article about Threat Modeling and](http://design.ros2.org/articles/ros2_threat_model.html)
  - [Disclosed in coordination with Acutronic Robotics a threat model for an industrial manipulator arm](https://news.aliasrobotics.com/collaborative-robot-insecurities-exposed/)
  - Disclosed a threat model for a ROS 2-natively powered robot (https://news.aliasrobotics.com/threat-modeling-a-ros-2-robot/)
- [x] Security tool development and contributions to
  - [aztarna](https://github.com/aliasrobotics/aztarna), a footprinting tool for robots
  - [ROS](https://github.com/aliasrobotics/ros1_fuzzer) and [ROS 2](https://github.com/aliasrobotics/ros2_fuzzer) fuzz testing packages
- [x] Security testing for applications on top of ROS-Eloquent
  - Pseudo-automatically reported [236 security flaws](https://github.com/aliasrobotics/RVD/issues?utf8=%E2%9C%93&q=label%3A%22robot+component%3A+ROS2%22+-label%3A%22invalid%22+) (ROS 2-specific)
  - Processed [22 security flaws](22) and submitted mitigations for those that applied (ROS 2-specific)
- [x] Security workshops to the (security and robotics) communities
  - [Small seminar](https://discourse.ros.org/t/talk-by-alias-robotics-on-performing-a-security-assessment-over-rosin-packages/8704) on how security is being applied to ROS-Industrial packages as part of the QA sessions
  - (upcoming) Talk about the attacker's model in robot cybersecurity @ the Spanish National Cybersecurity Instititute (https://www.incibe.es/en/enise/13enise/program/enise/workshops)
  - (upcoming) Participation in the ROS2 Security Workshop @ ROSCon 2019
  - (planned) Robot Cybersecurity Workshop at the European Robotics Forum 2020

dirk-thomas, 2019-11-21:
> Processed 22 security flaws and submitted mitigations for those that applied (ROS 2-specific)

@vmayoral The link pointing to the processed security flaws is wrong. Can you please fix it to see which ones have been processed so far?

vmayoral, 2019-11-21:
Updated, my bad and thanks for the ping @dirk-thomas.

We're actively developing RVD (and while there're still no instructions nor guidelines to use it just yet),  ROS 2 specific processed flaws can also be listed using [rvd_tools](https://github.com/aliasrobotics/RVD/tree/master/rvd_tools) as follows:
```bash
rvd list --label "weakness" --label "robot component: ROS2"  --isoption closed
Starting rvd, the CLI tool for managing the Robot Vulnerability Database...
Listing all open flaws from RVD...
  ID  Title
----  ---------------------------------------------------------------
 436  moveit_core: memory leak in loadGeometryFromStream
 435  rcl: test_graph memory leak, __interceptor_calloc
 334  nav2_map_server: detected memory leaks, __default_reallocate...
 167  nav2_util: detected memory leaks, rcl_lifecycle_register_tra...
 166  rclcpp: SEGV on unknown address, rclcpp::Node::get_node_topi...
 164  rcl_action: SEGV on unknown address, rcl_node_fini
 163  rcl: SEGV on unknown address, rcl_node_fini
 153  test_communication: detected memory leaks, rosidl_generator_...
 152  test_communication: detected memory leaks, rosidl_generator_...
 151  test_communication: detected memory leaks, rosidl_generator_...
 150  test_communication: detected memory leaks, rosidl_generator_...
 149  test_communication: detected memory leaks, rosidl_generator_...
 148  test_communication: detected memory leaks, rosidl_generator_...
 146  test_communication: detected memory leaks, rosidl_generator_...
 145  test_communication: detected memory leaks, rosidl_generator_...
 144  test_communication: detected memory leaks, rosidl_generator_...
 143  test_communication: detected memory leaks, rosidl_generator_...
 142  test_communication: detected memory leaks, rosidl_generator_...
 141  test_communication: detected memory leaks, rosidl_generator_...
 140  test_communication: detected memory leaks, rosidl_generator_...
 117  rclcpp_lifecycle: detected memory leaks, __default_reallocat...
 107  rcl_lifecycle: detected memory leaks, __default_reallocate
```

FYI, a few things have changed since we posted that last update. I'm listing them as well as other items that are planned to be completed within 2019:
- [x] Got first round of feedback regarding the RVD proposal ([discourse discussion](https://discourse.ros.org/t/introducing-the-robot-vulnerability-database/11105/3), ROSCon 2019 conversations)
- [x] Got `rvd_tools` CLI implemented to facilitate its integration with other third-party proprietary tools (improving rapidly)
- [x] Transferred open (and future) tickets to a machine readable format. This has been highly inspired by prior-work ([robust](https://github.com/robust-rosin/robust) project) where mitigation of bugs was at the center of the work done.
- [ ] Re-implement syntax validation (prior `action` has been disabled after the machine-readable format update)
- [ ] Implement automatic identification of duplicates
- [ ] Re-enable automatic dynamic testing reports for ROS 2 base packages (currently disabled until we address relevant feedback)
  - [ ] tagging things appropriately indicating that comes from "automatic dynamic code analysis"
- [ ] Enable automatic static testing reports (currently generating **many** tickets and lots of noise)
  - [ ] tagging things appropriately

SteveMacenski, 2019-11-21:
@vmayoral I don't want to bog down this thread with too much discussion, but how are these being reported? I see some navigation comments here but I've never heard from Alias on them or direct reports. What's the feedback or solution-fixing mechanism you're proposing?

vmayoral, 2019-11-21:
> I see some navigation comments here but I've never heard from Alias on them or direct reports.

By comments, I presume you mean tickets filed about [navigation 2](https://github.com/aliasrobotics/RVD/issues?q=is%3Aissue+is%3Aopen+label%3A%22robot+component%3A+navigation2%22) right? If that's the case, they've been filed as the result of some internal testing validating our security pipelines to enable (other than the ROS 2 "core") relevant ROS 2 packages. 

We tried navigation and manipulation but didn't process them any further (nor report maintainers as you point out, we don't think it's ready just yet). As of now, reporting is disabled on these stacks  and they need to be de-duplicated. We will focus our efforts first in ROS 2 "core" but I don't see why we can't cover them later on.

> What's the feedback or solution-fixing mechanism you're proposing?

We're pretty open about this (and happy to get inputs!). One way is to agree with maintainers to automatically flag tickets in the corresponding packages (e.g. navigation2) that point to each flaw reported. The new ticket-format at RVD should provide a Docker-image for "reproducing" the flaw which will faciliate mitigation/fixes (e.g. see `reproduction-image` tag in https://github.com/aliasrobotics/RVD/issues/509).

This is just a proposal I came up with now though, very open to other ideas.

SteveMacenski, 2019-11-21:
I'm just referring to what you've provided above, I see nav2 utils nav2_map_server. Frankly, after the discourse announcement I meant to follow up and learn more about what you were proposing but... you know how that goes with a stack of tickets on your desk. I'm not sure what you mean by "tried navigation.. but didn't process them any further". 

I guess what I'm asking is if you find issues with these things, why aren't there tickets being filed in the respective ecosystems to fix them (or PRs always welcome to fix for us :wink:)? Do I have to check your ticket list in RVD to know of issues? Are you just reporting on the package that has the issue, or any package that includes from the offending package?

Filing tickets would be nice, but following that link, I really don't want to add 66 tickets about 4 actual problems. Automated would be nice, but just filing 1 ticket for the offending package and not other packages that are just using it. If that list is also still untenable, maybe have 1 metaticket that you dump all the tickets into (and maybe link back to the full ticket on RVD). Ex. where those ticket links are for the RVD ticket 

```
Magic All-Issue Metaticket:

Hey, you have some problems [repo]
- [ ] Issue 1 #5311
- [ ] Issue 2 #54534
...
```

vmayoral, 2019-11-21:
Pretty good input! Thanks a lot :)

> I'm just referring to what you've provided above, I see nav2 utils nav2_map_server. Frankly, after the discourse announcement I meant to follow up and learn more about what you were proposing but... [...]

Got you! You're referring then to [RVD#167](https://github.com/aliasrobotics/RVD/issues/167) and [RVD#334](https://github.com/aliasrobotics/RVD/issues/334). These tickets were detected using the Google Sanitizers and have been addressed at [RVD#333 ](https://github.com/aliasrobotics/RVD/issues/333) which remains "open" because the [patch](https://github.com/ros2/rcl/pull/509) we sumitted is still being reviewed. The ticket was filed from the package that caused the memory leak however, reality is that in both of those tickets the actual problem wasn't in navigation2 package but in rcl. Long story-short this was "affecting" navigation2 but wasn't caused by any of the software components of it. Based on this, I'd argue that we should probably not report to navigation2 maintainers but *maybe* label the tickets appropriately indicating a relationship to the navigation2 package.

> I guess what I'm asking is if you find issues with these things, why aren't there tickets being filed in the respective ecosystems to fix them (or PRs always welcome to fix for us 😉)? Do I have to check your ticket list in RVD to know of issues? Are you just reporting on the package that has the issue, or any package that includes from the offending package?

We're still discussing internally how to approach this. As pointed out above we're flexible and accept requests/input. Personally, I like quite a lot your suggestion about the periodic meta-tickets for those flaws that "*originate*"  from the package. Those that don't (originate there) but may affect it, probably shouldn't get reported directly but instead, get compiled into a weekly report that summarizes flaws and their dependencies.

SteveMacenski, 2019-11-21:
> I'd argue that we should probably not report to navigation2 maintainers but maybe label the tickets appropriately indicating a relationship to the navigation2 package.

Yes. I don't want to have unactionable tickets being filed. I'm ruthless in closing out tickets in projects who's issues are in other projects. If an issue is in packageA, I don't want to hear about it in packageN (but sure, labeling it or a quick one-time @SteveMacenski ping is OK).

mjcarroll, 2019-11-21:
If there are any:
* Specific enhancements or features
* API breaking changes or migration notes
* Known issues in the Eloquent release
that you would like to be included on the [Eloquent Elusor Release Page](https://index.ros.org/doc/ros2/Releases/Release-Eloquent-Elusor/), please make pull requests against [ros2_documentation](https://github.com/ros2/ros2_documentation/blob/master/source/Releases/Release-Eloquent-Elusor.rst)

mjcarroll, 2019-11-22:
:confetti_ball: https://discourse.ros.org/t/ros-2-eloquent-elusor-released :confetti_ball:

closed 2019-11-22 by mjcarroll