ros2/ros2 #735: Python3.5 support lost in Dashing release

emersonknapp, 2019-06-25:
## Bug report
According to REP2000, the target Python version for ROS2 Dashing is 3.5

Variable annotation syntax (https://www.python.org/dev/peps/pep-0526/) was introduced to `rclpy` and `launch` between Crystal and Dashing (e.g. https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/node.py#L136, https://github.com/ros2/launch/blob/3d34d3b50e43e9236fcfc5804966eee67e4d4c77/launch/launch/launch_description_source.py#L52) - this syntax is not Python3.5 compatible.

Perhaps we need to update REP2000, or patch these python libraries to use "type comments" as introduced in https://www.python.org/dev/peps/pep-0484/

**Required Info:**

- Operating System:
  - Yocto Poky
- Installation type:
  - ARM32 tarball
- Version or commit hash:
  - https://ci.ros2.org/view/packaging/job/packaging_linux/1506/
- DDS implementation:
  - any
- Client library (if applicable):
  - rclpy

ivanpauno, 2019-06-25:
This is from `Dashing Diameta` dependency requirements table in http://www.ros.org/reps/rep-2000.html:

```
Minumum language requirements:
C++14
Python 3.5
```

Package | Ubuntu Bionic | MacOS** | Windows 10** | Debian Stretch
-- | -- | -- | -- | --
Python | 3.6.5 | 3.7.3 | 3.7.3 | 3.5.3

I think that the intention was stop supporting python 3.5, and update the minimum requirement to 3.6.

kyrofa, 2019-06-25:
Given that Debian isn't at 3.6, perhaps we should loop @gbiggs into this discussion as they're invested in it.

jacobperron, 2019-06-25:
We've discussed this at the ROS 2 meeting. Considering that Debian is a Tier 3 platform that we are not testing, and Debian Buster will be released soon (supporting Python 3.7), we will not actively work towards fixing the incompatibility. We will, however, accept fixes from the community to backport to Dashing.

kyrofa, 2019-06-25:
Yay for 3.6 then!

nuclearsandwich, 2019-08-30:
https://github.com/ros2/launch/pull/324 and https://github.com/ros2/rclpy/pull/414 resolve the issues in launch and rclpy. I'll have to run another bottom-up build to determine if there are any other regressions against 3.5 but we are in much better shape.

Getting a recurring job for Stretch is a bit more of a... stretch goal... at present as there's still a few things in the way wrt rosdeps and packages.

dirk-thomas, 2019-08-30:
> Getting a recurring job for Stretch is a bit more of a... stretch goal... at present as there's still a few things in the way wrt rosdeps and packages.

Could you elaborate what these few things are? Beside that setting up a Stretch CI job on build.ros2.org sounds straight forward.

nuclearsandwich, 2019-09-03:
> Beside that setting up a Stretch CI job on build.ros2.org sounds straight forward.

Setting up this Stretch CI job is straightforward _after_ we've set up binary packages for Debian Stretch since CI jobs require the ros-workspace package unconditionally. I'm not sure if that's a requirement that we can soften now or no but it would require additional changes in ros_buildfarm to support.

Adding the Stretch packages for ros-workspace would mean updating the distribution.yaml platform and re-blooming at least the repos ament_package, ament_cmake and ros_workspace but anyone else whose rosdep deps aren't available in stretch would get failures when blooming so I'm not sure that it's worth doing.

If we're going to have Debian Buster support in F-turtle alongside 20.04 then there's a case for starting the F release cycle with Buster availability so have the option of turning on these jobs and smoking out missing rosdeps but that's outside the scope here.

nuclearsandwich, 2019-09-03:
Using this Dockerfile from @sloretz checking for syntax issues is straightforward.

```
FROM debian:stretch

RUN export DEBIAN_FRONTEND=noninteractive \
 && apt-get update \
 && apt-get install -y \
    tzdata \
    locales \
    lsb-release \
    curl \
    gnupg \
 && apt-get clean \
 && sed -i 's/^# *\(en_US.UTF-8\)/\1/' /etc/locale.gen \
 && locale-gen  \
 && update-locale LANG=en_US.UTF-8

RUN apt-get update \
 && apt-get install -y \
    git wget \
    python3-pip python3-setuptools python3-venv \
 && apt-get clean

RUN cd ~ \
 && python3 -m venv ~/env3 \
 && . ~/env3/bin/activate \
 && pip3 install -U setuptools \
 && pip3 install flake8 vcstool \
 && wget https://raw.githubusercontent.com/ros2/ros2/dashing/ros2.repos \
 && vcs import --input ros2.repos . \
 && find -name AMENT_IGNORE | xargs -I{} sh -c 'rm -r $(dirname {})' \
 && python3 -m flake8 --select=E99
```

It will produce some syntax errors in googletest auxiliary scripts and anywhere else. Since we can't get true CI for Debian Stretch easily I am going to add running this check to the release process for Dashing.

nuclearsandwich, 2019-09-11:
As of Dashing Patch release 3 Python 3.5 support has been restored. I wasn't able to easily get a CI job set up for Python 3.5 on Stretch but I will be running the above test before triggering future syncs. However it's worth noting that the above only covers code in the ros2.repos file. Any compatibility issues outside of that scope should be ticketed directly on the upstream repository.

closed 2019-09-11 by nuclearsandwich