ros2/ros2 #736: Dashing Patch Release 2

nuclearsandwich, 2019-06-26:
This issue will track the broad status of the next Dashing patch release. The date for this patch release is not currently set and will be determined by a critical mass of desired changes ready to be released.

We're going to use a GitHub Project board for this patch release again : [Dashing Patch Release 2](https://github.com/orgs/ros2/projects/6)

As with prior releases, if you want to contribute fixes please open pull requests targeting the latest development branch and mention to the reviewer that you'd like the pull request to be considered for backporting into Dashing.

esteve, 2019-06-26:
@nuclearsandwich could https://github.com/ros/pluginlib/pull/154 and https://github.com/ros2/rosbag2/pull/113 be considered for this release? They basically enable crosscompiling rosbag2. Thanks.

nuclearsandwich, 2019-06-26:
> could [ros/pluginlib#154](https://github.com/ros/pluginlib/pull/154) and [ros2/rosbag2#113](https://github.com/ros2/rosbag2/pull/113) be considered for this release? They basically enable crosscompiling rosbag2. Thanks.

The pluginlib one is already up (rolled over from patch release 1) and I've added ros2/rosbag2#113 to the proposed column.

esteve, 2019-06-27:
@nuclearsandwich cool, thanks for the update.

crdelsey, 2019-08-01:
I'd like to see ros2/rclcpp#777  backported if not too late.

nuclearsandwich, 2019-08-01:
> I'd like to see [ros2/rclcpp#777](https://github.com/ros2/rclcpp/pull/777) backported if not too late.

The window for new releases into patch 2 is closed for all but critical fixes in order to allow time for the build farm to settle and give folks a chance to test the packages while they're available from the ROS 2 testing repository.

I've proposed ros2/rclcpp#777 to be backported for the next patch release (#753)

nuclearsandwich, 2019-08-06:
The release is out.

PR #754 (2019-08-06): Dashing Patch Release 2
Update the dashing-release repos file for patch release 2. The merged commit will also be tagged as `release-dashing-20190806`.

Closes #736
closed 2019-08-06 by nuclearsandwich