ros2/ros2 #738: Missing dependencies ros2 dashing windows 10

TheLaas, 2019-06-27:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - binaries
- Version or commit hash:
  - ros2-dashing-20190614-windows-debug-amd64.zip
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
1. Install Ros2 according to https://index.ros.org/doc/ros2/Installation/Dashing/Windows-Install-Binary/
2. In cmd run call <Path to ros>/local_setup.bat
3. run ros2 --help
```

#### Expected behavior
```
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...

ros2 is an extensible command-line tool for ROS 2.

optional arguments:
  -h, --help            show this help message and exit

Commands:
  action     Various action related sub-commands
  bag        Various rosbag related sub-commands
  component  Various component related sub-commands
  daemon     Various daemon related sub-commands
  launch     Run a launch file
  lifecycle  Various lifecycle related sub-commands
  msg        Various msg related sub-commands
  multicast  Various multicast related sub-commands
  node       Various node related sub-commands
  param      Various param related sub-commands
  pkg        Various package related sub-commands
  run        Run a package specific executable
  security   Various security related sub-commands
  service    Various service related sub-commands
  srv        Various srv related sub-commands
  test       Run a ROS2 launch test
  topic      Various topic related sub-commands

  Call `ros2 <command> -h` for more detailed usage.
```
#### Actual behavior
I get this information but it's preceeded about a bunch of errors:
```
Failed to load entry point 'info':

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the multiarray numpy extension module failed.  Most
likely you are trying to import a failed build of numpy.
Here is how to proceed:
- If you're working with a numpy git repository, try `git clean -xdf`
  (removes all files not under version control) and rebuild numpy.
- If you are simply trying to use the numpy version that you have installed:
  your installation is broken - please reinstall numpy.
- If you have already reinstalled and that did not fix the problem, then:
  1. Check that you are using the Python you expect (you're using c:\python37\python_d.exe),
     and that you have no directories in your PATH or PYTHONPATH that can
[... 202 lines cut ...]
Commands:
  action     Various action related sub-commands
  bag        Various rosbag related sub-commands
  component  Various component related sub-commands
  daemon     Various daemon related sub-commands
  launch     Run a launch file
  lifecycle  Various lifecycle related sub-commands
  msg        Various msg related sub-commands
  multicast  Various multicast related sub-commands
  node       Various node related sub-commands
  param      Various param related sub-commands
  pkg        Various package related sub-commands
  run        Run a package specific executable
  security   Various security related sub-commands
  service    Various service related sub-commands
  srv        Various srv related sub-commands
  test       Run a ROS2 launch test
  topic      Various topic related sub-commands

  Call `ros2 <command> -h` for more detailed usage.
```
#### Additional information
I've installed python with all additional debug module as suggested in https://github.com/ros2/ros2/issues/525 since I had that error as well.

I've tried to reinstall numpy package.

ivanpauno, 2019-06-27:
Have you followed these steps: https://index.ros.org/doc/ros2/Installation/Dashing/Windows-Development-Setup/#extra-stuff-for-debug-mode?

mjcarroll, 2019-06-27:
Yes, running debug on Windows will require at least `numpy` installed as the debug variant.  It's up to you how you want to manage this, but virtualenv can keep you from clobbering the already-installed `numpy` variant.

This is the command that is of most relevance from the document that Ivan posted:
```
> python_d -m pip install --force-reinstall https://github.com/ros2/ros2/releases/download/numpy-archives/numpy-1.16.2-cp37-cp37dm-win_amd64.whl
```

If you plan to use sros2 functionality, you will also need `lxml` installed in the same fashion.

TheLaas, 2019-07-01:
Thank you for your replies.

I ran both:
```
> python_d -m pip install --force-reinstall https://github.com/ros2/ros2/releases/download/numpy-archives/numpy-1.16.2-cp37-cp37dm-win_amd64.whl
> python_d -m pip install --force-reinstall https://github.com/ros2/ros2/releases/download/lxml-archives/lxml-4.3.2-cp37-cp37dm-win_amd64.whl
```
It took away the numpy errors, I however still get this error:
```
Failed to load entry point 'info': DLL load failed: The specified module could not be found.
The C extension 'C:\dev\ros2-dashing\Lib\site-packages\rosbag2_transport\_rosbag2_transport_py_d.pyd' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'play': DLL load failed: The specified module could not be found.
The C extension 'C:\dev\ros2-dashing\Lib\site-packages\rosbag2_transport\_rosbag2_transport_py_d.pyd' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'record': DLL load failed: The specified module could not be found.
The C extension 'C:\dev\ros2-dashing\Lib\site-packages\rosbag2_transport\_rosbag2_transport_py_d.pyd' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
```
and this error seems to still prevent me from starting any nodes. Any ideas?

Thank you.

EpicZa, 2019-07-08:
Also getting this issue on windows 10, binary install.

obucklin, 2019-09-17:
I am having the same issue. 

> Also getting this issue on windows 10, binary install.

rosbag2 transport errors are the same. Additionally I tried to install from source and am receiving the same error.
<
Anyone find a solution? Thanks!

hidmic, 2019-10-24:
@TheLaas @EpicZa @obucklin we've recently merged a fix for above's DLL loading failures, see https://github.com/ros2/rosbag2/pull/190. This fix will be included in the upcoming [Eloquent Elusor release](https://index.ros.org/doc/ros2/Releases/Release-Eloquent-Elusor/#timeline-before-the-release). You may build from source to verify this.

I'll close this issue now. Feel free to re-open as needed.

closed 2019-10-24 by hidmic