ros2/ros2 #739: Release Testing Automation

hidmic, 2019-06-27:
## Feature request
#### Feature description

As of today, release testing is a manual, time consuming process that demands the ROS2 team and willing external contributors a significant share of time during the weeks that precede a release. An attempt is made to cover a testing matrix that spans tutorials, demos, CLI across Tier ~1 platforms and RMW implementations, which can only get worse as the available feature or the supported platform set grow over time.

There's, however, a fair amount of tests that could be automated right away, and some redundancy within the matrix itself that could be reduced, increasing overall code quality and likely catching regressions earlier in the cycle.

#### Implementation considerations

[Here's](https://docs.google.com/spreadsheets/d/1ywAS6a0V73hAZegMZjvhpDpcY4yPrUQ_t5YMhOcgH0E/edit?usp=sharing) a sample of the release testing matrix. It has been color coded to separate what can be automated from what not, and what should be automated and what not. It may also be used as a TODO list and discussion site.

Most tests, if not all, require further specification to be coded. This task, though initially delegated to the tests author and reviewer, should ensure basic coverage e.g.:

- Validate correct output like stdout, messages and logs, both in terms of content and latency.
- Validate correct behavior when run in unusual ways e.g. repeatedly re-running it or running things out of (the usual) order.
- Validate graceful signal handling e.g. SIGINT and SIGTERM.

As of Dashing, [ros_testing](https://github.com/ros2/ros_testing), a launch based testing framework, can be leveraged when automating many of the process-level tests that can be found in the release testing matrix. It may also be worth increasing test coverage for the framework itself as we rely more and more on it.

hidmic, 2019-06-27:
@ivanpauno @cevans87 feel free to comment and link your respective PRs and issues to this meta ticket.

ivanpauno, 2019-06-28:
https://github.com/ros2/ros2cli/pull/287 is adding tests for increasing coverage of `ros2cli`, and avoid testing it manually.

ivanpauno, 2019-07-04:
Ok, now this is finally passing.

The only problem I see is that running the tests is really long, it took about 6 minutes to run them in my machine (4 rmw implementations: fastrtps, fastrtps dynamic, opensplice and connext).

Considering that we are going to add a bunch of new tests (https://github.com/ros2/ros2/issues/739), maybe it would be better to not run them in every build, but just in the nightlies. Or maybe, only before release as replacement of the manual testing.

We could add a new cmake variable for this, `BUILD_NIGHTLY_TESTS` or something like that. So we usually only ran a subset of tests.

@hidmic @jacobperron any thoughts?

hidmic, 2019-07-04:
> Considering that we are going to add a bunch of new tests (#739), maybe it would be better to not run them in every build, but just in the nightlies.

I fully agree. As the test base grows, we may need to limit which ones are run on every CI job.

ivanpauno, 2019-07-04:
> Ok, now this is finally passing.
>
> The only problem I see is that running the tests is really long, it took about 6 minutes to run them in my machine (4 rmw implementations: fastrtps, fastrtps [...]

My intention was to do that comment in https://github.com/ros2/ros2cli/pull/287, but I mixed the browser tabs.
I will left the comment here as it's a good place too.

jacobperron, 2019-07-08:
I agree that it would be nice to have a mechanism for limiting which tests are run. I think this is outside the scope of https://github.com/ros2/ros2cli/pull/287, but could be tracked in this ticket or a spin-off (maybe on https://github.com/ros2/ci). It would be nice to have an opt-in check box to enable all tests for a particular CI run.

hidmic, 2019-07-19:
Current status is:

- Adding end-to-end test coverage for all ros2cli commands (https://github.com/ros2/ros2cli/pull/304). **Rough ETA ~1.5 week FTE**, including reviews and excluding general QA discussions that may apply to our entire test base.

- Adding end-to-end test coverage for `demo_nodes_cpp`. TODO(@cevans87): link PR/branch.

hidmic, 2019-11-19:
As of Eloquent, a good deal of progress has been made. In particular, test coverage for ROS 2 CLI commands is now much more complete, though still partial and an undergoing effort e.g.: 

- [x] https://github.com/ros2/ros2cli/pull/304
- [x] https://github.com/ros2/ros2cli/pull/391
- [ ] https://github.com/ros2/ros2cli/pull/389
- [ ] https://github.com/ros2/ros2cli/pull/390

To accommodate these tests, the testing infrastructure itself was improved and extended e.g.:

- [x] https://github.com/ros2/launch/pull/279
- [x] https://github.com/ros2/launch/pull/312

Neither patch list is exhaustive. Release testing matrix pruning will be conducted once all CLI commands and verbs have associated tests, post-Eloquent.

mjcarroll, 2020-02-05:
I have added ros2/ros2cli#389 and ros2/ros2cli#390 to the foxy project board, so I'm going to close this out.

closed 2020-02-05 by mjcarroll