ros2/ros2 #740: No setup files to source after Dashing install

thvhauwe, 2019-07-01:
## Bug report

**Required Info:**

- Operating System:
```
NAME="Ubuntu"
VERSION="18.04.2 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.2 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```
- Installation type:
Debian package
- Version or commit hash:
Dashing Diademata
- DDS implementation:

- Client library (if applicable):

#### Steps to reproduce issue

``` sudo apt install ros-dashing-desktop  ``` -->
```

```

#### Expected behavior
Being able to find setup files in `/opt/ros/dashing` to source the ROS2 installation.
#### Actual behavior
All other files in bin, include, lib, opt, share seem to be installed correctly. No setup files found in `/opt/ros/dashing`
#### Additional information
none

clalancette, 2019-07-01:
Hm, this is definitely working for me (and I'm pretty sure that we would have heard about it before if this was truly a bug).  Those files come from the package `ros-dashing-ros-workspace`; can you run:

```
dpkg -l ros-dashing-ros-workspace
dpkg -L ros-dashing-ros-workspace
```

And post the results here?

thvhauwe, 2019-07-01:
The output was: 
```
$ dpkg -l ros-dashing-ros-workspace
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  ros-dashing-ro 0.7.1-1bioni amd64        Provides the prefix level environ
$ dpkg -L ros-dashing-ros-workspace
/.
/opt
/opt/ros
/opt/ros/dashing/_order_packages.py
/opt/ros/dashing/local_setup.bash
/opt/ros/dashing/local_setup.sh
/opt/ros/dashing/local_setup.zsh
/opt/ros/dashing/setup.bash
/opt/ros/dashing/setup.sh
/opt/ros/dashing/setup.zsh
/opt/ros/dashing/share/ament_index/resource_index/package_run_dependencies/ros_workspace
/opt/ros/dashing/share/ament_index/resource_index/packages/ros_workspace
/opt/ros/dashing/share/ament_index/resource_index/parent_prefix_path/ros_workspace
/opt/ros/dashing/share/ros_workspace
/opt/ros/dashing/share/ros_workspace/cmake
/opt/ros/dashing/share/ros_workspace/cmake/ros_workspaceConfig-version.cmake
/opt/ros/dashing/share/ros_workspace/cmake/ros_workspaceConfig.cmake
/opt/ros/dashing/share/ros_workspace/environment
/opt/ros/dashing/share/ros_workspace/environment/ament_prefix_path.sh
/opt/ros/dashing/share/ros_workspace/environment/library_path.sh
/opt/ros/dashing/share/ros_workspace/environment/path.sh
/opt/ros/dashing/share/ros_workspace/environment/pythonpath.sh
/opt/ros/dashing/share/ros_workspace/local_setup.bash
/opt/ros/dashing/share/ros_workspace/local_setup.sh
/opt/ros/dashing/share/ros_workspace/local_setup.zsh
/opt/ros/dashing/share/ros_workspace/package.xml
/usr
/usr/share
/usr/share/doc
/usr/share/doc/ros-dashing-ros-workspace
/usr/share/doc/ros-dashing-ros-workspace/changelog.Debian.gz
/usr/share/doc/ros-dashing-ros-workspace/copyright
```
I got the files by reinstalling the package you mentioned.

Solved by: `sudo apt-get install --reinstall ros-dashing-ros-workspace`

closed 2019-07-01 by thvhauwe