ros2/ros2 #741: Cannot build dashing from source with visual studio 2019

EpicZa, 2019-07-08:
WIndows 10
Source
Dashing
Fast & Connext

tinyxml2
gtestvendor
tinydir_vendor
poco_vendor
fastcdr
osrf_testing_tools

Are giving "RuntimeError: Unknown / unsupported VS version ''"

Visual Studio 2019 Community, x64 admin prompt.

Repulled the source for dashing will not build with 2019.

mjcarroll, 2019-07-08:
Have you followed the instructions to insure that you have the required CMake version?  There are some versions of CMake (including one in the prereleased VS2019) that didn't support generating VS2019 targets.

You can check your cmake version with `cmake --version` on the command line.

The output of `where cmake` would also be helpful for diagnosing.

EpicZa, 2019-07-08:
C:\ros2>cmake --version
cmake version 3.14.5

CMake suite maintained and supported by Kitware (kitware.com/cmake).

C:\ros2>where cmake
C:\Program Files\CMake\bin\cmake.exe

I deselected c++ cmake in the visual studio log installer, and just updated from 12 to 14. Still getting the error.

EpicZa, 2019-07-15:
> Have you followed the instructions to insure that you have the required CMake version? There are some versions of CMake (including one in the prereleased VS2019) that didn't support generating [...]

Which versions are supported / not supported, and where is this listed?

sloretz, 2019-07-15:
> Which versions are supported / not supported, and where is this listed

@EpicZa http://www.ros.org/reps/rep-2000.html#dashing-diademata-may-2019-may-2021

The minimum CMake version on Windows 10 is listed as 3.14.4

mjcarroll, 2019-10-17:
@EpicZa were you able to resolve this with a newer version of CMake?

dirk-thomas, 2019-10-17:
I will go ahead and close the ticket for now assuming the previous feedback resolved the problem. If not please feel free to comment on the closed ticket and it can be reopened.

closed 2019-10-17 by dirk-thomas

EpicZa, 2019-10-17:
@mjcarroll  I unfortunately no longer work on ROS stuff. 3.14.5 > 3.14.4 though, unless it needed an exact match? 

If I get a second this weekend, I'll test a version above 3.14.5.