ros2/ros2 #743: stderr: tf2

DanielLuu, 2019-07-12:
## Bug report

**Required Info:**

- Operating System:
  - macOS 10.14.5
- Installation type:
  - From source
- Version or commit hash:
  - 
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Followed build from source 
On running
`colcon build --symlink-install`

``` code that can be copy-pasted is preferred ``` -->
```
--- stderr: tf2
In file included from /Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/src/gtest-all.cc:41:
In file included from /Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest.cc:135:
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-internal-inl.h:63:1: error: C++ requires a type specifier for all declarations
GTEST_DISABLE_MSC_WARNINGS_PUSH_(4251 \
^
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-internal-inl.h:64:76: error: expected ';' after top level declarator
/* class A needs to have dll-interface to be used by clients of class B */)
                                                                           ^
                                                                           ;
In file included from /Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/src/gtest-all.cc:42:
In file included from /Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:47:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/fcntl.h:23:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/fcntl.h:351:18: error: field has incomplete type 'struct flock'
        struct flock    fl;             /* flock passed for file locking */
                        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/fcntl.h:351:9: note: forward declaration of 'flock'
        struct flock    fl;             /* flock passed for file locking */
               ^
In file included from /Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/src/gtest-all.cc:42:
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:95:46: error: use of undeclared identifier 'GTEST_DEFAULT_DEATH_TEST_STYLE'
static const char kDefaultDeathTestStyle[] = GTEST_DEFAULT_DEATH_TEST_STYLE;
                                             ^
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:291:24: error: member access into incomplete type 'internal::UnitTestImpl'
      GetUnitTestImpl()->internal_run_death_test_flag();
                       ^
/usr/local/include/gtest/internal/gtest-internal.h:98:7: note: forward declaration of 'testing::internal::UnitTestImpl'
class UnitTestImpl;                    // Opaque implementation of UnitTest
      ^
In file included from /Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/src/gtest-all.cc:42:
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:371:43: error: member access into incomplete type 'internal::UnitTestImpl'
  TestInfo* const info = GetUnitTestImpl()->current_test_info();
                                          ^
/usr/local/include/gtest/internal/gtest-internal.h:98:7: note: forward declaration of 'testing::internal::UnitTestImpl'
class UnitTestImpl;                    // Opaque implementation of UnitTest
      ^
In file included from /Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/src/gtest-all.cc:42:
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:381:24: error: no template named 'Matcher'
                       Matcher<const std::string&> matcher, const char* file,
                       ^
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:383:27: error: member access into incomplete type 'internal::UnitTestImpl'
  return GetUnitTestImpl()->death_test_factory()->Create(
                          ^
/usr/local/include/gtest/internal/gtest-internal.h:98:7: note: forward declaration of 'testing::internal::UnitTestImpl'
class UnitTestImpl;                    // Opaque implementation of UnitTest
      ^
In file included from /Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/src/gtest-all.cc:42:
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:400:42: error: no template named 'Matcher'
  DeathTestImpl(const char* a_statement, Matcher<const std::string&> matcher)
                                         ^
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:441:3: error: no template named 'Matcher'
  Matcher<const std::string&> matcher_;
  ^
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:1065:43: error: no template named 'Matcher'
  ForkingDeathTest(const char* statement, Matcher<const std::string&> matcher);
                                          ^
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:1080:36: error: no template named 'Matcher'
                                   Matcher<const std::string&> matcher)
                                   ^
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:1087:8: error: cannot initialize object parameter of type 'const testing::internal::DeathTestImpl' with an expression of type 'testing::internal::ForkingDeathTest'
  if (!spawned())
       ^~~~~~~
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:1090:3: error: cannot initialize object parameter of type 'testing::internal::DeathTestImpl' with an expression of type 'testing::internal::ForkingDeathTest'
  ReadAndInterpretStatusByte();
  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:1094:3: error: cannot initialize object parameter of type 'testing::internal::DeathTestImpl' with an expression of type 'testing::internal::ForkingDeathTest'
  set_status(status_value);
  ^~~~~~~~~~
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:1102:44: error: no template named 'Matcher'
  NoExecDeathTest(const char* a_statement, Matcher<const std::string&> matcher)
                                           ^
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:1131:3: error: cannot initialize object parameter of type 'testing::internal::ForkingDeathTest' with an expression of type 'testing::internal::NoExecDeathTest'
  set_child_pid(child_pid);
  ^~~~~~~~~~~~~
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:1134:5: error: cannot initialize object parameter of type 'testing::internal::DeathTestImpl' with an expression of type 'testing::internal::NoExecDeathTest'
    set_write_fd(pipe_fd[1]);
    ^~~~~~~~~~~~
/Users/danielluu/ros2_ws/install/gtest_vendor/src/gtest_vendor/./src/gtest-death-test.cc:1141:22: error: member access into incomplete type 'internal::UnitTestImpl'
    GetUnitTestImpl()->listeners()->SuppressEventForwarding();
                     ^
/usr/local/include/gtest/internal/gtest-internal.h:98:7: note: forward declaration of 'testing::internal::UnitTestImpl'
class UnitTestImpl;                    // Opaque implementation of UnitTest
      ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o] Error 1
make[1]: *** [gtest/CMakeFiles/gtest.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< tf2	[ Exited with code 2 ]
Aborted  <<< rcl
Aborted  <<< shape_msgs
Aborted  <<< nav_msgs
Aborted  <<< diagnostic_msgs
Aborted  <<< visualization_msgs
Aborted  <<< tf2_msgs
Aborted  <<< sensor_msgs
```

#### Expected behavior
All packages processed

#### Actual behavior
Erroring on tf2 would like to see if anyone has run into this

MengNan-Li, 2019-09-25:
I also meets the same problem.

Karsten1987, 2019-09-28:
I just ran into a similar problem.

The way I could solved it was by uninstalling `console_bridge` and solely rely on the `console_bridge_vendor` package. 
My believe is that the `console_bridge` from homebrew brings in a somewhat different version of gtest.

wjwwood, 2019-10-25:
@Karsten1987 since you opened a pull request related to this (https://github.com/ros/console_bridge/pull/63) can I ask you to resolve this issue when appropriate? (I assigned you)

If you don't think that make sense, please let me know.