ros2/ros2 #744: Porting TopicStatistics from ros1 and improvement ideas

tarunbelagodu, 2019-07-12:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request
Porting topic statistics from ros1 to ros2

#### Feature description
 In ros1, it was possible to subscribe to /statistics which is of type TopicStatistics
http://docs.ros.org/api/rosgraph_msgs/html/msg/TopicStatistics.html

I dont see it listed under ros2 rcl interfaces
https://github.com/ros2/rcl_interfaces/tree/master/rosgraph_msgs

#### Implementation considerations

1. I also see that the ros2 demo for topic monitoring has a different implementation based on qos offered by the DDS middleware
https://github.com/ros2/demos/blob/master/topic_monitor/topic_monitor/scripts/topic_monitor.py

This demo is in python. Would be good to have some cpp examples. 

2. Rather than publish stats on all topics (ok if it defaults to true), would be great to enable this selectively for certain messages so that it is lightweight for subscribers

3. For our needs at Voyage, we're exploring if it is viable to extend the Publisher/DataWriter APIs so that such "meta" information about topics can be made available.

tarunbelagodu, 2019-08-12:
Adding as comment, not sure if it should be added to the ticket description itself: 
Based on some recent work on our ros1 system, we were able to confirm that enabling statistics add a bunch of compute overhead and affects the performance of the stack especially when we have large number of pub/sub and large data. From our reading of the current ros1 code, it seems like the overhead is introduced on potentially both sides: publisher side to compute stats, subscriber side to query it. Would be immensely useful to consider adding capability to enable/disable this on per topic and on the subscriber side have it moved into a helper thread so that it does not slow down the callback handling the data

https://github.com/ros/ros_comm/blob/89ca9a7bf288b09d033a1ab6966ef8cfe39e1002/clients/roscpp/src/libros/subscription.cpp#L663

RobotJINI, 2019-12-10:
I'm looking into this issue, I am new to DDS but it appears there are data_fields left to the user in DDS that can probably be used to track statistics.  I'm writing a test program to investigate further.

RobotJINI, 2019-12-11:
Investigating further I do not think the DDS data_fields will work.

________________________________________________________________________________

Right now I am thinking about a topic_monitor node that subscribes to all or a subset of messages in ros2 then starts collecting and reporting the statistics data on them.

There are some issues with this method, you either;
 1. assume that since the topic_monitor received the data other nodes subscribed to the topic also got the data.
2. Implement code on the subscription side that if a monitoring parameter is enabled it sends an ack to the topic_monitor via <topic_name>_monitor_ack

Also on the publishing side you would be assuming the topic_monitor is actually receiving all the messages sent by the publisher which is also not necessarily true.

_______________________________________________________________________________

Another method might be to create monitored_subscriber and monitored_publisher classes to be used instead of normal pub/sub and could keep track of the stats independently, periodically this data could be fused in a topic_monitor node and then reported as statistics.  This would require changing code to monitor topics though.

RobotJINI, 2019-12-11:
subcription.hpp stats can be recorded in the <void handle_message> function when turned on

publisher.hpp stats can be recorded in the <void publish> function when turned on

________________________________________________________________________________

To auto republish the statistics when statistics is turned on would require adding a callback to the node class which creates a publisher for all enabled subscriptions and publications owned by that node and publishing out that information ASAP or at some other rate.  This way you could turn statistics and their publishing on and off via parameter setting at the node level.

_______________________________________________________________________________

OR we could also just record the statistics when enabled on the pub/sub classes and leave the possible republishing of the statistics to the user.

________________________________________________________________________________

Whether the user republishes or the node auto does it, there will need to be a separate aggregation node that combines info from publisher and subscriber to get.

# number of messages delivered during the window
int32 delivered_msgs
# numbers of messages dropped during the window
int32 dropped_msgs

dabonnie, 2020-06-18:
Is this still relevant given #901 and #917?

closed 2023-03-30 by clalancette