ros2/ros2 #750: Delete wiki

rotu, 2019-07-29:
When searching for "Install ROS2" (in Google, Bing, and DuckDuckGo), this ROS2 wiki comes up before the actual content on index.ros.org where it is now hosted. Perhaps this wiki should be deleted altogether to make that official documentation site more discoverable.

Google results:
[image]

First hit:
[image]

Second hit:
[image]

tfoote, 2019-08-07:
I agree it might be time to disable the wiki. We switched all the pages over to be the pointers to the new site in the hopes that people with old links would find their way to the new location and that google would too, since there's no content anymore. However they appear to give more pagerank to the github url than our own even now. 

Since we switched over in October it's now over 9 months later and 2 distributions later it would be better to get the improved search placement over keeping backwards compatibility.

nuclearsandwich, 2019-08-08:
I have disabled the wiki. If we get a deluge of complaints of broken links it will be possible to re-enable it with the contents preserved.

closed 2019-08-08 by nuclearsandwich