ros2/ros2 #752: Failed to build custom message

zxc455233, 2019-08-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - deb
- Version or commit hash:
  - 
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - 

#### Steps to reproduce issue
I tried to build the custom message in ros2_ws,
there are my steps:

```
$ source /opt/ros/dashing/setup.bash
$ cd ros2_ws
$ colcon build --symlink-install --packages-select <my_package>
```
Then, I got ERROR like this:
```
Starting >>> strategy
--- stderr: strategy                               
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/rosidl_typesupport_opensplice_cpp/rosidl_typesupport_opensplice_cpp", line 55, in <module>
    sys.exit(main())
  File "/opt/ros/dashing/lib/rosidl_typesupport_opensplice_cpp/rosidl_typesupport_opensplice_cpp", line 50, in main
    args.idl_pp,
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_typesupport_opensplice_cpp/__init__.py", line 82, in generate_dds_opensplice_cpp
    _modify(h_filename, msg_name, _replace_path_and_timestamp, idl_path=idl_path)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_typesupport_opensplice_cpp/__init__.py", line 96, in _modify
    modified = callback(lines, msg_name, idl_path=idl_path)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_typesupport_opensplice_cpp/__init__.py", line 115, in _replace_path_and_timestamp
    assert False, "Failed to find '// Generated: ' line"
AssertionError: Failed to find '// Generated: ' line
make[2]: *** [rosidl_typesupport_opensplice_cpp/strategy/msg/dds_opensplice/robot_state__type_support.cpp] Error 1
make[2]: *** Deleting file 'rosidl_typesupport_opensplice_cpp/strategy/msg/dds_opensplice/robot_state__type_support.cpp'
make[1]: *** [CMakeFiles/strategy__rosidl_typesupport_opensplice_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< strategy	[ Exited with code 2 ]

Summary: 0 packages finished [27.3s]
  1 package failed: strategy
  1 package had stderr output: strategy
```

#### Expected behavior
I  worked fine and built success before I try to rebuild ros2_ws...
After I deleted the build files(build/ install/ log/)
The error message appears repeatedly

#### Additional information
I tried to build the custom message and build ros1_bridge to bridge message between ROS1 and ROS2, but I stuck in this step

----
## Feature request

#### Feature description

#### Implementation considerations

dirk-thomas, 2019-08-01:
I guess you installed the latest version of OpenSplice (6.9.190705). That version slightly changed the generated code which makes the mentioned assertion fail.

We have fixed the problem already with ros2/rosidl_typesupport_opensplice#33 and also backported the change into Dashing and Crystal but it will take until the next sync for the fixes to land in the apt repos.

Until then you can workaround the problem my manually editing the file with the assertion and comment it out. (That change will be overwritten when installing the upcoming version of the `rosidl_typesupport_opensplice_cpp` package.)

zxc455233, 2019-08-02:
> I guess you installed the latest version of OpenSplice (6.9.190705). That version slightly changed the generated code which makes the mentioned assertion fail.
>
> We have fixed the problem already [...]

Thank you so much

I remove ```opensplice``` by ```apt```, then error messages is not appear again.

closed 2019-08-02 by zxc455233