ros2/ros2 #753: Dashing Patch Release 3

nuclearsandwich, 2019-08-01:
This issue will track the broad status of the next Dashing patch release. The date for this patch release is not currently set and will be determined by a critical mass of desired changes ready to be released.

We're going to use a GitHub Project board for this patch release again : [Dashing Patch Release 3](https://github.com/orgs/ros2/projects/9)

As with prior releases, if you want to contribute fixes please open pull requests targeting the latest development branch and mention to the reviewer that you'd like the pull request to be considered for backporting into Dashing.

clalancette, 2019-09-03:
Addition to the release notes:

- The default opensplice configuration was changed from `AllowMulticast=true` to `AllowMulticast=spdp`.  This means that Opensplice will now use multicast for discovery but unicast for data delivery, which brings it inline with how the other RTPS/DDS implementations operate.  This should improve data delivery on wireless networks when using Opensplice, at the cost of increasing network bandwidth on wired networks when there are many subscribers to the same topic.  This is a backport of https://github.com/ros2/rosidl_typesupport_opensplice/pull/34.

nuclearsandwich, 2019-09-11:
Packaging jobs

:penguin: [image]
:penguin: :muscle: :six::four: [image]
:penguin: :muscle: :business_suit_levitating: [image]
:apple: [image]
:wind_chime: [image]
:wind_chime: :no_entry_sign: :beetle: [image]

nuclearsandwich, 2019-09-11:
The release is [out](https://discourse.ros.org/t/new-packages-and-patch-release-for-ros-2-dashing-diademata-2019-09-10/10627).

closed 2019-09-11 by nuclearsandwich