ros2/ros2 #760: Python3-lark-parser not available on Raspbian Buster

jff8, 2019-08-15:
## Bug report

**Required Info:**

- Operating System:
  - Raspbian Buster on Raspberry Pi 3B and 4B
- Installation type:
  - Install from source 
- Version or commit hash:
  - Dashing
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
From https://index.ros.org/doc/ros2/Installation/Crystal/Linux-Development-Setup/ Instructions:

sudo apt update && sudo apt install -y  python3-lark-parser
E: Unable to locate package python3-lark-parser

Appears  http://packages.ros.org/ros2/ubuntu has not been setup for Raspbian or Ubuntu Buster

```

#### Expected behavior

#### Actual behavior

#### Additional information

<!-- If you are reporting a bug delete everything below
     If you are requesting a feature deleted everything above this line -->
----
## Feature request

#### Feature description
<!-- Description in a few sentences what the feature consists of and what problem it will solve -->

#### Implementation considerations
<!-- Relevant information on how the feature could be implemented and pros and cons of the different solutions -->

nuclearsandwich, 2019-08-21:
If you're building from source, you can install lark-parser from PyPI via pip. https://pypi.org/project/lark-parser/

python3-lark-parser is also in our Debian Stretch repositories. I can't guarantee the package will work on Raspbian but I expect it will.

I'm hoping to set up Buster repositories for ROS 2 this week, which will include adding python3-lark-parser. In the long term I'm hoping to get it moved upstream so we can stop packaging it ourselves.

nuclearsandwich, 2019-09-30:
python3-lark-parser is available in the ROS 2 repos for all supported architectures on Ubuntu Bionic and Debian Stretch and Buster.

closed 2019-09-30 by nuclearsandwich