ros2/ros2 #767: Error with rcutils when running colcon build

jff8, 2019-08-26:
## Bug report

**Required Info:**

- Operating System:
  - Raspbian buster on RPi4
- Installation type:
  - https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/
- Version or commit hash:
  - Eloquent and Dashing
- DDS implementation:
  - Default
- Client library (if applicable):
  -  N/A 

#### Steps to reproduce issue
Followed steps as per <https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/ to build ROS2 on Raspberry Pi 4

When running:
colcon build --symlink-install

Get error with rcutils:
Failed   <<< rcutils	[ Exited with code 1 ]

```

#### Expected behavior
Expect colcon to complete with no critical errors

#### Actual behavior
Fails with this error:

CMake Error at /home/pi/install/launch_testing_ament_cmake/share/launch_testing_ament_cmake/cmake/launch_testing_ament_cmake-extras.cmake:17 (find_package):
  By not providing "Findpython_cmake_module.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "python_cmake_module", but CMake did not find one.

  Could not find a package configuration file provided by
  "python_cmake_module" with any of the following names:

    python_cmake_moduleConfig.cmake
    python_cmake_module-config.cmake

  Add the installation prefix of "python_cmake_module" to CMAKE_PREFIX_PATH
  or set "python_cmake_module_DIR" to a directory containing one of the above
  files.  If "python_cmake_module" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  /home/pi/install/launch_testing_ament_cmake/share/launch_testing_ament_cmake/cmake/launch_testing_ament_cmakeConfig.cmake:38 (include)
  /home/pi/install/ament_lint_auto/share/ament_lint_auto/cmake/ament_lint_auto_find_test_dependencies.cmake:36 (find_package)
  CMakeLists.txt:120 (ament_lint_auto_find_test_dependencies)

#### Additional information
Cmake version 3.13.4 which installs by default on Raspbian buster.
Also has the same error with DASHING

#### Implementation considerations
<!-- Relevant information on how the feature could be implemented and pros and cons of the different solutions -->

Jesus805, 2019-12-08:
@dirk-thomas I would like to add that I'm also getting this exact issue on a RPi4 with a freshly installed Raspbian Buster.
This is the output of `colcon build --symlink-install`
```
Starting >>> shared_queues_vendor
--- stderr: rcutils
/usr/bin/ld: CMakeFiles/test_atomics_executable.dir/test/test_atomics.c.o: in fu                              nction `main':
test_atomics.c:(.text+0x11a8): undefined reference to `__atomic_store_8'
/usr/bin/ld: test_atomics.c:(.text+0x11c0): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x1268): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x12b8): undefined reference to `__atomic_exc                              hange_8'
/usr/bin/ld: test_atomics.c:(.text+0x12f8): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x13cc): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x13e8): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x1484): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x14c8): undefined reference to `__atomic_exc                              hange_8'
/usr/bin/ld: test_atomics.c:(.text+0x150c): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x1dd8): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x1df4): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x1ea0): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x1ef4): undefined reference to `__atomic_exc                              hange_8'
/usr/bin/ld: test_atomics.c:(.text+0x1f38): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x2008): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x2020): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x20b8): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x20f8): undefined reference to `__atomic_exc                              hange_8'
/usr/bin/ld: test_atomics.c:(.text+0x2138): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x27e8): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x2804): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x28b0): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x2904): undefined reference to `__atomic_exc                              hange_8'
/usr/bin/ld: test_atomics.c:(.text+0x2948): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x2a18): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x2a30): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x2ac8): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x2b08): undefined reference to `__atomic_exc                              hange_8'
/usr/bin/ld: test_atomics.c:(.text+0x2b48): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x3278): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x3294): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x3340): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x3394): undefined reference to `__atomic_exc                              hange_8'
/usr/bin/ld: test_atomics.c:(.text+0x33d8): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x34a8): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x34c0): undefined reference to `__atomic_loa                              d_8'
/usr/bin/ld: test_atomics.c:(.text+0x3558): undefined reference to `__atomic_sto                              re_8'
/usr/bin/ld: test_atomics.c:(.text+0x3598): undefined reference to `__atomic_exc                              hange_8'
/usr/bin/ld: test_atomics.c:(.text+0x35d8): undefined reference to `__atomic_loa                              d_8'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_atomics_executable.dir/build.make:85: test_atomics                              _executable] Error 1
make[1]: *** [CMakeFiles/Makefile2:424: CMakeFiles/test_atomics_executable.dir/a                              ll] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rcutils    [ Exited with code 2 ]
Aborted  <<< urdf
Aborted  <<< shared_queues_vendor
--- stderr: rviz_ogre_vendor
[... cut ...]
    url: https://github.com/ros2/tlsf.git
    version: 0.5.0
  src/ros2/unique_identifier_msgs:
    type: git
    url: https://github.com/ros2/unique_identifier_msgs.git
    version: 2.1.0
  src/ros2/urdf:
    type: git
    url: https://github.com/ros2/urdf.git
    version: 2.2.0
  src/ros2/urdfdom:
    type: git
    url: https://github.com/ros2/urdfdom.git
    version: 2.2.0
  src/ros2/yaml_cpp_vendor:
    type: git
    url: https://github.com/ros2/yaml_cpp_vendor.git
    version: 54feb9dfcf88c6925ee16aed4d5d7eda1eea946c
```

Jesus805, 2019-12-08:
It seems this issue has been reported on the rcutils repository 
https://github.com/ros2/rcutils/issues/171