ros2/ros2 #771: ROS2 DD "failed to create process"

qootec, 2019-09-03:
## Bug report

**Required Info:**

- Operating System: Windows 10
- Installation type: Binaries
- Version or commit hash: release DD patch 2
- DDS implementation: N/A (RTI)
- Client library (if applicable): N/A (rcpcpp)

#### Steps to reproduce issue
- Have Python already installed on the system in a path != c:\Python37
- Install following instructions for Windows
- Open CMD and try to run "ros2"

#### Expected behavior
- Should show just the help for ros2

#### Actual behavior
- Produces only the line "Failed to create process".

#### Additional information
As described in bug #525, the script  ros2-script.py is enforcing a specific location of the Python executable. This location should not be hardcoded since the location is system/installation dependent. Any user having a Python installation in a different path before starting to use ROS2 gets stuck on this basic error message "Failed to create process" for which the only solution is to re-install Python or to go and patch the ROS2 installation. Neither is desirable.

Is there a compelling reason to have the path enforced? Can't we just have
#! python
That looks for python on the path, no?

aekanman, 2020-01-23:
if you still have this issue, do you mind copying your PATH here. I have previously ran into this issue and after hours of debugging, it turned out CMake and one random Chocolatey path variable were missing from the PATH. Adding them manually fixed the issue.

hidmic, 2020-02-13:
@qootec I've followed your steps and I did not run into the issues you described. https://github.com/ros2/ros2/issues/525 has been long resolved. So I'm inclined to think that @aekanman is right and something's not right with your PATH. 

I'll thus close this issue. Feel free to comment (or re-open) afterwards if need be.

closed 2020-02-13 by hidmic