ros2/ros2 #772: ROS2 expects RTI on hard-coded installation path

qootec, 2019-09-04:
## Bug report

**Required Info:**

- Operating System: Windows 10
- Installation type: binaries
- Version or commit hash: release DD patch 2
- DDS implementation: RTI 5.3.1
- Client library (if applicable): N/A

#### Steps to reproduce issue
- Make sure the RTI is installed on a non-default path (e.g. c:\prog\rti...)
- Try to build any ROS2 project

#### Expected behavior
- Normal build, taking into account the NDDSHOME variable.

#### Actual behavior
- Build failure triggered by the following warning:
`CMake Warning at D:/local/Dev/ros2-windows/share/rmw_connext_shared_cpp/cmake/ament_cmake_export_include_directories-extras.cmake:11 (message):
  Package 'rmw_connext_shared_cpp' exports the include directory 'C:/Program
  Files/rti_connext_dds-5.3.1/include' which doesn't exist
`

#### Additional information
Cause is a hardcoded path in 
ament_cmake_export_include_directories-extras.cmake.
These paths are "expected" to be on a fixed location, but that is not the case for everyone.
I guess relying on the environment variable NDDSHOME is the most suitable solution.

qootec, 2019-09-04:
__ put on hold __

This could be a side effect of me not converting an old project (Crystal) to Dashing in a correct way.
The issue occurs afais only on that old Crystal project and not on e.g. the new Dashing examples.

Still the presence of hard-coded paths is not logical, but I'll first check my Crystal-Dashing migration...

qootec, 2019-09-05:
Conclusion:

- Crystal project still works fine under Crystal
- New project with Dashing and custom messages in .msg format:
  Hits the hard-coded path. Examples do work because generated messages are already in binaries.
  Worked around this by symbolic link to my 'real' installation location.
`cd "c:\Program Files" && 
mklink /d rti_connext_dds-5.3.1 c:\prog\rti_connext_dds-5.3.1`
- New project with Dashing and custom messages in .idl format:
   No clue how to get it working...

ivanpauno, 2019-09-06:
I don't see any hardcoded path in connext related code, can you add a link where there is one?

```
Cause is a hardcoded path in
ament_cmake_export_include_directories-extras.cmake.
These paths are "expected" to be on a fixed location, but that is not the case for everyone.
I guess relying on the environment variable NDDSHOME is the most suitable solution.
```
`ament_cmake_export_include_directories-extras.cmake` is a file generated in the build process, so the path wasn't hardcoded.

We're using `NDDSHOME` to find connext:
https://github.com/ros2/rosidl_typesupport_connext/blob/20407822a1978fb1080149e6e7f44e1c3d7b603f/connext_cmake_module/cmake/Modules/FindConnext.cmake#L180

Please, try deleting your `build` and `install` folders and try again.
Also, check if your `NDDSHOME` environment variable is set to the correct directory.

qootec, 2019-09-07:
I saw that the script connext.bat does this check and should overrule the path based on NDDSHOME.
What I see:
- Remove build & install: no difference in behaviour
- With my symbolic link mentioned above: warning appears when performing local_setup on ros2-windows: The NDDSHOME path is different from the path at buildtime, the latter will be used… etc
That makes me conclude that it sees the NDDSHOME.
- Without my symbolic link, there is no warning at all (there is always one for OpenSplice, but that is not relevant to me). That makes me conclude that it correctly finds NDDSHOME and that it points to the correct directory, since else the rti script from the resources subdirectory would not be found and the script would complain.

My project changed a bit by now, so the message is somewhat different from before, but essentially looks the same to me. Currently, when I then try to "colcon build" my project containing to-be-generated msg files, it complains about not finding stuff in c:\program files:

CMake Warning at D:/local/Dev/ros2-windows/share/rmw_connext_shared_cpp/cmake/ament_cmake_export_libraries-extras.cmake:62 (message):
  Package 'rmw_connext_shared_cpp' exports the library 'C:/Program
  Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/nddscppd.lib' which doesn't
  exist
Call Stack (most recent call first):
  D:/local/Dev/ros2-windows/share/rmw_connext_shared_cpp/cmake/rmw_connext_shared_cppConfig.cmake:38 (include)
  D:/local/Dev/ros2-windows/share/rmw_connext_cpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  D:/local/Dev/ros2-windows/share/rmw_connext_cpp/cmake/rmw_connext_cppConfig.cmake:38 (include)
  D:/local/Dev/ros2-windows/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:61 (find_package)
  D:/local/Dev/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_get_typesupports.cmake:26 (call_for_each_rmw_implementation)
  D:/local/Dev/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:27 (rosidl_generator_py_get_typesupports)
  D:/local/Dev/ros2-windows/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  D:/local/Dev/ros2-windows/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:21 (rosidl_generate_interfaces)

ivanpauno, 2019-09-09:
Did you install from source or from archive?

qootec, 2019-09-09:
From the archive ros2-dashing-20190806-windows-release-amd64.zip

ivanpauno, 2019-09-10:
Ok, so this is only a problem in the from archive installation, and it won't be a problem if you install from source (except you re-install connext in another place, in that case the whole workspace should be rebuilt to avoid the problem).

Some generated cmake files during the archive's build are filled with the path of the connext libraries that was found at build time, and it doesn't look at `NDDSHOME` environment variable again.
I have downloaded the windows fat archive and I confirm that's the case.
e.g. download [this archive](https://github.com/ros2/ros2/releases/download/release-crystal-20190820/ros2-crystal-20190820-windows-release-amd64.zip), from `PATH_TO_EXTRACTED_WS/share/rmw_connext_shared_cpp/cmake/ament_cmake_export_libraries-extras.cmake`:
```
set(_exported_libraries "rmw_connext_shared_cpp;optimized;C:/Program Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/nddsc.lib;optimized;C:/Program Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/nddscore.lib;optimized;C:/Program Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/nddscpp.lib;optimized;C:/Program Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/rticonnextmsgcpp.lib;debug;C:/Program Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/nddscd.lib;debug;C:/Program Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/nddscored.lib;debug;C:/Program Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/nddscppd.lib;debug;C:/Program Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/rticonnextmsgcppd.lib")
```

I'm not too sure on how to solve the problem, nor in how many places a change is needed.
@dirk-thomas any thoughts?