ros2/ros2 #776: Dashing Patch Release 4

nuclearsandwich, 2019-09-11:
This issue will track the broad status of the next Dashing patch release. The date for this patch release is not currently set and will be determined by a critical mass of desired changes ready to be released.

We're going to use a GitHub Project board for this patch release again: [Dashing Patch Release 4](https://github.com/orgs/ros2/projects/12)

As with prior releases, if you want to contribute fixes please open pull requests targeting the latest development branch and mention to the reviewer that you'd like the pull request to be considered for backporting into Dashing.

AAlon, 2019-09-16:
Hey, we see the following fixes as critical so it'd be great if you could them:
String parameter length limit in rcl: https://github.com/ros2/rcl/pull/487 
Fix race condition in rclcpp: https://github.com/ros2/rclcpp/pull/837
And if there are any other fixes that are related to https://github.com/ros-planning/navigation2/issues/1079.

When do you think the patch will go out? is next week feasible?
Thanks!

dirk-thomas, 2019-09-17:
> And if there are any other fixes that are related to ros-planning/navigation2#1079.

Maybe ros2/rcl#501 and ros2/rclcpp#854 ?

nuclearsandwich, 2019-09-17:
> And if there are any other fixes that are related to [ros-planning/navigation2#1079](https://github.com/ros-planning/navigation2/issues/1079).

@AAlon can you please be more precise when proposing PRs for backport. If there are specific changes you'd like to have backported it is far more helpful if they are enumerated rather than asking our release management team to rifle through a thread and pick out potentially related fixes.

> Hey, we see the following fixes as critical so it'd be great if you could include them:
> String parameter length limit in rcl: [ros2/rcl#487](https://github.com/ros2/rcl/pull/487)
> Fix race condition in rclcpp: [ros2/rclcpp#837](https://github.com/ros2/rclcpp/pull/837)

@AAlon as the sync windows is rapidly approaching, to make sure we parallelize as much as possible can your team review each of the proposed PRs for ABI and API compatibility with Dashing and open the backport PRs against the `dashing` branches? We've got people we can assign to do review and release once the backport PRs are open.

AAlon, 2019-09-17:
> > And if there are any other fixes that are related to [ros-planning/navigation2#1079](https://github.com/ros-planning/navigation2/issues/1079).
> 
> Maybe [ros2/rcl#501](https://github.com/ros2/rcl/pull/501) and [ros2/rclcpp#854](https://github.com/ros2/rclcpp/pull/854) ?

Thanks, let's include these too.

> @AAlon can you please be more precise when proposing PRs for backport. If there are specific changes you'd like to have backported it is far more helpful if they are enumerated rather than asking [...]

I wasn't asking for that :-) it was in case you/others already had context about those issues and know about a fix I missed, especially considering that those could be blockers for ROS2D usage in production systems - not just some random thread. 

> @AAlon as the sync windows is rapidly approaching, to make sure we parallelize as much as possible can your team review each of the proposed PRs for ABI and API compatibility with Dashing and open [...]

Yep! we'll be working on it asap.

eric1221bday, 2019-09-21:
Is it possible for https://github.com/ros2/rclpy/pull/389 to be included? It's quite a simple fix and allows for much more flexibility in parameter passing in launch files or otherwise.

nuclearsandwich, 2019-10-18:
Packaging jobs:

* Ubuntu Linux AMD64 [image]
* Ubuntu Linux :muscle::six::four: [image]
* Ubuntu Linux :muscle::business_suit_levitating: [image]
* CentOS Linux [image]
* macOS Sierra [image]
* Windows [image]
* Windows Debug [image]

dirk-thomas, 2019-10-18:
@nuclearsandwich Please do not use the nightly packaging jobs to create the archives for another distro. Instead use the non-nightly equivalent of these jobs.

We instruct users that they can get the latest archives for master from these jobs. I am pretty sure we ran into this very same problem last time and users reported unexpected behavior / content of the downloaded archive. And at that time we came to the same conclusion not to use the packaging jobs for creating archives for other distros.

nuclearsandwich, 2019-10-18:
> Please do not use the nightly packaging jobs to create the archives for another distro. Instead use the non-nightly equivalent of these jobs.

Sorry about that. It was an oversight on my part since these are launched manually.

nuclearsandwich, 2019-10-18:
This patch release is now out. https://discourse.ros.org/t/new-packages-and-patch-release-for-ros-2-dashing-2019-10-18/11086

PR #802 (2019-10-18): Dashing Patch Release 4
Update releases for Dashing patch release 4.

Closes #776.
closed 2019-10-18 by nuclearsandwich