ros2/ros2 #785: Support zero-copy in ROS2

Karsten1987, 2019-09-21:
top level GitHub ticket for connecting PRs related to the development for enabling zero-copy message transport in ROS2.

For me details, refer to the [design doc PR](https://github.com/ros2/design/pull/256)

Karsten1987, 2019-10-07:
An initial round of CI:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2019-10-11:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2019-10-18:
* Linux [image]  // unrelated test failures
* Linux-aarch64 [image]
* macOS [image]  // unrelated test failures
* Windows -- waiting for free executor

dirk-thomas, 2019-10-21:
@Karsten1987 Please create a corresponding PR for CycloneDDS asap since it is currently failing to build: e.g. [image]

closed 2020-02-05 by mjcarroll