ros2/ros2 #786: Include rmw_cyclonedds into ros2 repo for nightly jobs?

ruffsl, 2019-09-25:
With rmw_cyclonedds_cpp released into dashing, is there a particular reason for the the omission of https://github.com/ros2/rmw_cyclonedds from the ros2.repos file? It would be nice to have the rmw implementation easily available in the nightly archive job for testing and CI: https://github.com/osrf/docker_images/issues/324

dirk-thomas, 2019-09-25:
Before being added to the `ros2.repos` file an option would need to be added to our CI jobs on ci.ros2.org to enable/disable CycloneDDS. The same options already exist for the other RMW implementation we are building.

ruffsl, 2019-09-25:
Would one just open a PR mimicking the occurrences of fastrpts, or would there be more to it?
https://github.com/ros2/ci/search?q=fastrtps&unscoped_q=fastrtps

In comparison, it looks like opensplice neccetates more boilerplate.

dirk-thomas, 2019-09-25:
Each RMW impl. unfortunately needs their own little customization - but yes, the change necessary for CycloneDDS should be very similar.

dirk-thomas, 2019-09-26:
One more thing: it should also be added to [rmw_implementation](https://github.com/ros2/rmw_implementation/blob/4790322697cf290b0af76b2fc00d4955658103f5/rmw_implementation/package.xml#L14-L21).

ruffsl, 2019-09-26:
Should we also add it to the list of rosdep ignored packages in the install wiki? Or save that for release?

dirk-thomas, 2019-09-26:
> Should we also add it to the list of rosdep ignored packages in the install wiki?

Can you point to the exact instructions you are referring to.

ruffsl, 2019-09-26:
Oh, it's been updated to suggest apt installing the desktop meta package instead of globing over ros-: https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/#install-ros-2-packages

The install from source still lists some basic `--skip-keys` though: 
https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/#install-dependencies-using-rosdep

dirk-thomas, 2019-09-26:
I am not sure if it is necessary to add any of the CycloneDDS package names to the `--skip-keys`. I would be good to double check and only add them if really necessary.

ruffsl, 2019-09-26:
Was there a reason fast rtps was included in the `--skip-keys` given it should already exist in the workspace after vcs import from the ros2.repos file here, thus accounted for by `--ignore-src` arg. I could understand the skipping of opensplice or rti-connext-dds, to avoid installing external DDS debs.

dirk-thomas, 2019-11-04:
All referenced PRs have landed by now. Closing.

closed 2019-11-04 by dirk-thomas