ros2/ros2 #796: ROS1 - ROS2 migration

amranov, 2019-10-08:
HI, 

Thank you for this wonderful work.
My question seems litlebit basic, I am trying to migrate all ROS1 packages  i am using to control a mobile robot to ROS2 but i am litlebit confused. 
I already changed the launch files to python scripts, i would like to know what are the changes needed for migration (except for launch files) ?

mlanting, 2019-10-08:
This is probably a question to ask on https://answers.ros.org, but you can find a migration guide here that covers a lot of things you need to change: https://index.ros.org/doc/ros2/Contributing/Migration-Guide/

You're going to need to update your package.xml, CMakeLists.txt, as well as the source code.

clalancette, 2019-10-08:
Thanks @mlanting good answer.  I'll close this out as it belongs on answers.ros.org.

closed 2019-10-08 by clalancette