ros2/ros2 #798: Restrict DDS middleware network traffic to localhost

hidmic, 2019-10-08:
## Feature request

#### Feature description

Currently and by default, most DDS-based ROS middlewares lookup and choose a suitable network interface on the machine that's hosting them. It makes for easy configuration of multi-machine ROS systems, but it may also result in undesired cross-talk. This is specially true in CI setups, where multiple tests running in parallel may inadvertently interact with each other. 

Isolation via `ROS_DOMAIN_ID`, though possible, poses a maintenance burden as IDs must be carefully allocated and assigned by other means. Additionally, the set of available domain IDs is a rather limited one (see [RTI Connext documentation](https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/manuals/connext_dds/html_files/RTI_ConnextDDS_CoreLibraries_UsersManual/Content/UsersManual/ChoosingDomainID.htm), this relevant [eProsima Fast-RTPS ticket](https://github.com/eProsima/Fast-RTPS/issues/223) and the [RTPS standard](https://www.omg.org/spec/DDSI-RTPS/2.2/PDF)). 

Both burden and limitation are acceptable when a ROS system is intended to span several machines in a network, but when that's not the case it's an unnecessary source of problems. Providing an easy way for users to force DDS-based middlewares to use loopback would preclude all of these issues.

#### Implementation considerations

From a UX perspective, negative `ROS_DOMAIN_ID`s could provide the means to externally enable this restriction (credits to @dirk-thomas). `ROS_DOMAIN_ID`s sign would act as a flag: 

- if it's a positive number, a suitable network interface will be chosen (just like it is today); 
- if it's negative, loopback will be used instead.

Meanwhile, the actual domain ID would become the magnitude (i.e. the absolute value) of the `ROS_DOMAIN_ID` number. 

For instance, `ROS_DOMAIN_ID=-1` would read as _"Use 1 for a domain ID over loopback"_ while `ROS_DOMAIN_ID=1` would read as _"Use 1 for a domain ID over a suitable network interface"_.

allenh1, 2019-10-08:
> From a UX perspective, negative ROS_DOMAIN_IDs could provide the means to externally enable this restriction (credits to @dirk-thomas). ROS_DOMAIN_IDs sign would act as a flag:

So that certainly solves one problem, but I think we could solve more problems with a little bit more effort up front.

I'd suggest instead introducing a different variable we could route to the middleware, that would allow specific hosts to participate in DDS.

For example,

```
ROS_ALLOWED_HOSTS="localhost.local;robot_1.local"
```

would mean that the middleware would only talk to those hosts.

This way you could easily add a specific host and the daemon could periodically probe this variable.

This is already done via XML in fastRTPS (and probably more, I just can't speak for RTI or others), but the XML has to be in the run directory of the binary, which can be rather tedious (again, at least for fastRTPS).

clalancette, 2019-10-08:
Yeah, I have to agree with @allenh1 , something more generic for picking interfaces/networks would probably be more useful long-term.

BMarchi, 2019-10-08:
I'm seeing (but I didn't test this yet) that [connext](https://community.rti.com/static/documentation/connext-dds/5.3.1/doc/api/connext_dds/api_c/group__NDDS__Transport__UDPv4__Plugin.html) supports adding "properties" as a key,value pair to the QoS. To be more specific, you can add a white list of interfaces (property `dds.transport.UDPv4.builtin.parent.allow_interfaces`) for UDP and Shared Memory transport.

fujitatomoya, 2019-10-09:
> ROS_DOMAIN_ID=-1
domain identification is used to assign the port number, so i do not think this is not suitable. 

> ROS_ALLOWED_HOSTS="localhost.local:robot_1.local"
not sure this env variable interface is good,
but agree with the idea just like white list to specify the network interfaces which gives users more flexibility.

eboasson, 2019-10-09:
For the original problem, I think one should provide as simple a solution as possible, and the negative-domain id hack is certainly a clever way of doing that. I don't think it is wise to start adding environment variables with less-than-universal attributes and expect all the RMW implementations to parse that and make things happen. Even the notion of a network interface might not translate so well to different RMW implementations, although it will likely work fine for all DDS implementations.

But rather than using a negative id, I would suggest extending ``ROS_DOMAIN_ID`` to an integer + an optional tag, where only those nodes with the same tag should communicate. Then you can simply put the hostname in the tag to have communications limited to the host. E.g., something like `` export ROS_DOMAIN_ID=33:`hostname` `` and you can be assured that you will only ever talk to processes on the same machine.

Then you add the convention that a tag equal to the host name indicates that the *intent* is to limit communications to the local host and allow RMW implementations to use whatever tricks it needs to make that happen and/or to optimize communications.

This has the added advantage of mapping nicely to something that I've pushed for in [DDSI-RTPS 2.3](https://www.omg.org/spec/DDSI-RTPS/2.3/PDF) (see section 8.5.5.1) without being too DDS specific. I know that feature is not universally implemented yet — not even in Cyclone DDS 🤔 I'll put it in really soon 🙂 — but the "host name" convention works around that by allowing the RMW layer to configure the middleware appropriately.

From the comments above I take it that only implementing the "host name" convention will not be a problem on FastRTPS and Connext. And for Cyclone DDS and OpenSplice it is also straightforward.

ivanpauno, 2019-10-09:
I think the general goal of the PR is just to add:
> Providing an easy way for users to force DDS-based middlewares to use loopback ...

Though having a way to specify what hosts should be used sounds nice, I think it may be more difficult to implement than just forcing the middleware to use loopback.
I would rather limit the first iteration to add a way to enforce "local isolation", than add a full list of allowed network interfaces. In the case that doing the later is easy, I would go for it :).

dirk-thomas, 2019-10-09:
As long as the way the user configures the setting doesn't change in between that is ok. Therefore I think we should settle on how a user would configure both of the cases (localhost-only as well as set of specific hosts).

dirk-thomas, 2019-10-09:
After an offline conversation we agreed that mixing the information into `ROS_DOMAIN_ID` isn't appealing. Using negative numbers limits discoverability as well as possibilities to evolve it in the future. Having a separate environment variable for allowing  / restricting the network communication is therefore preferred.

Since we are not sure what exact feature set we could offer across all RMW impl (in terms of limiting the communication to specific hosts, network interfaces, broadcast addresses) we decided to go with a simple boolean flag.

In ROS 1 the env vars `ROS_HOSTNAME` and `ROS_IP` are used to describe how a host can be reached from other hosts. Therefore those names should be avoided for this boolean flag.

One proposal is to name the variable: `ROS_LOCALHOST_ONLY`.

Keeping in mind that we might want to make the localhost-only option the default value that name comes with some inverse logic. The potential new default would be the same as setting the env var to `1|true`. Users who want to communicate with other hosts would need to set it to `0|false`.

To avoid that kind of inversion maybe another possible name could be: `ROS_NETWORK`. The semantic would be the exact opposite: setting it to `1` would communicate with other hosts, `0` would only talk to localhost.

We are looking for any kind of feedback / opinion with the hope that we can get this new option implemented in time for Eloquent.

fujitatomoya, 2019-10-10:
I am not good at naming for this kind of stuff, but i prefer `ROS_NETWORK` since i believe that we likely to use network interfaces with ROS2 as distributed system.

allenh1, 2019-10-10:
> To avoid that kind of inversion maybe another possible name could be:
>
> `ROS_NETWORK`.

To be frank, I'm really not a fan of this idea. By default, I really do like the DDS discovery (it's extremely convenient).

But, in applications where it is desired to restrict things, I would prefer to be specific. While it may not be universal for all middlewares, it could be implemented by a few, and, should one where the future is not implemented be in use, that middleware could be responsible for announcing that it does not support the specific feature the user is asking for.

I'm just afraid that by using a binary value for a variable for something like `ROS_NETWORK` we will have to deprecate that name in the future. If we really want to do something like that, we should name it something along the lines of `ROS_MULTICAST_DISCOVERY`, or `ENABLE_MULTICAST_DISCOVERY`.

dirk-thomas, 2019-10-10:
> By default, I really do like the DDS discovery (it's extremely convenient).

The reason why we consider to change the default to be localhost-only is user feedback we have received in the past. It happens fairly frequently that users run into the case of unintentionally cross talking to instances on other machines in the same network. Those cases are hard to notice / debug since they are usually not deterministic (depending on if nodes are running on other machines).

The opposite - that no network communication is happening by default - will result in a clear, deterministic, easy to document default behavior. We think that this will serve the majority of the user base better.

> I'm just afraid that by using a binary value for a variable for something like ROS_NETWORK we will have to deprecate that name in the future.

We are aware of the desire to make this configuration more flexible. But since neither exact scope nor the possible configuration options are unclear at the moment we want to not hold on until that happens (very likely not for Eloquent) but move forward with the features we are certain about. If in the future another env var is introduced which provide more flexibility it can either coexist with the proposed on or that one can be deprecated and later removed if obsolete.

> If we really want to do something like that, we should name it something along the lines of `ROS_MULTICAST_DISCOVERY`, or `ENABLE_MULTICAST_DISCOVERY`.

Those terms - multicast as well as discovery - imply a specific way of implementation in the RMW layer Since there are RMW implementations which don't use these concepts (e.g. non-DDS based implementations) we want to avoid such implementation specific term here.

allenh1, 2019-10-10:
> The reason why we consider to change the default to be localhost-only is user feedback

Alright. That's certainly a good reason.

> Those terms - multicast as well as discovery - imply a specific way of implementation in the RMW layer Since there are RMW implementations which don't use these concepts

Also a very good point.

@dirk-thomas thanks for the clarification. 👍

gavanderhoorn, 2019-10-10:
@dirk-thomas wrote:
> another possible name could be:
> 
> `ROS_NETWORK`.
> 
> The semantic would be the exact opposite: setting it to 1 would communicate with other hosts, 0 would only talk to localhost.

@allenh1's point about being more descriptive is something I believe I would agree with.

To me (non-native speaker), an env var named `ROS_NETWORK` seems to have something to do with *which* network ROS will use, or which network *is* the ROS network (whatever that may mean). If that is the intention, I would not expect such a variable then to only contain a `1` (or `true`), but a name, or some other "network identifier" (whatever that may be).

Of course; everything is solvable by providing enough documentation, but comparing `ROS_NETWORK` to `ROS_MASTER_URI`, `ROS_HOSTNAME` and `ROS_IP` I have the feeling the latter three are more obviously named (ignoring the 10+ year we've been using these names of course). But at least it's clear that `ROS_MASTER_URI` should contain a URI and it's the "master's URI".

Referring to multicast or discovery would seem to be too DDS specific.

Perhaps `ROS_ALLOWED_HOSTS` (as suggested in https://github.com/ros2/ros2/issues/798#issuecomment-539692607) is not that bad. It more clearly conveys what it's supposed to configure. If we're not happy with it containing a list, and complicating things by having the RMWs parsing it then perhaps it could be allowed to only contain a single value `localhost`. If it's present and contains `localhost` -> restrict to `localhost`. If it contains something else -> no restrictions. Not present -> no restrictions.

That would leave room for incremental enhancement in the future by allowing other values instead of just `localhost`.

@dirk-thomas wrote:
> One proposal is to name the variable:
> 
> `ROS_LOCALHOST_ONLY`.

If `1|true` means all ROS communication is restricted to `localhost`, this would be the clearest name so far, and it seems to be also align with the desired binary behaviour.

mlanting, 2019-10-10:
How about `ROS_ENABLE_NETWORK` instead of just `ROS_NETWORK` to make its boolean nature more clear?

BMarchi, 2019-10-11:
I agree with @gavanderhoorn and @jacobperron, but instead of `String != localhost then ignore it` I would rather throw (not if the string is empty). This way we avoid what @jacobperron said:

>  there is a case where we introduce a new valid value and an RMW isn't updated to handle it properly (i.e. treating the new value as "no restrictions" might be unexpected by users)

Users will know that writing something different than localhost or empty will make every rmw implementation throw. I'll start implementing this in the possible rmw implementations. If anyone thinks this does not make any sense, it's easy to change the env variable to a boolean instead.

clalancette, 2019-10-11:
> The reason why we consider to change the default to be localhost-only is user feedback we have received in the past. It happens fairly frequently that users run into the case of unintentionally [...]

I can see arguments both ways on this, but if we *do* make it localhost-only by default, we should be aware that it is a huge change to downstream users.  With that in mind, I'd expect a large warning at startup that this will only communicate over localhost, and a pointer to the documentation on how to enable multi-machine communication.

gavanderhoorn, 2019-10-11:
> I can't stop thinking this is more of a `development` feature rather than an release enhancement.

It's actually a very big difference in how ROS 1 and ROS 2 behave.

ROS 1 is localhost by default (well, it's actually not, but without special configuration of 'outsiders', it is). ROS 2 is now *every reachable host* by default.

That has consequences for development, testing, verification and deployment. On both the concrete (ie: implementation) as well as the conceptual (ie: design) level.

It may be apparent to experienced DDS users that without special configuration everything listens to everything, but from a ROS UX perspective, it's actually already a "huge change to downstream users".

clalancette, 2019-10-11:
> It's actually a very big difference in how ROS 1 and ROS 2 behave.
> 
> ROS 1 is localhost by default (well, it's actually not, but without special configuration of 'outsiders', it is). ROS 2 is now _every reachable host_ by default.

That's a good point.  But it means we are now between a rock and a hard place; existing ROS 2 users have already worked through this and expect it to communicate over the network by default, while users coming from ROS 1 expect the "localhost-only" behavior.  Either way we go will confuse someone.

Which brings me back to the point that if we make this change, we need to be very loud about it.  I also think that this isn't something that we should slip into Eloquent at the last minute; we should probably have a pre-announcement on Discourse and our other channels, and then change it for a future distribution instead.

allenh1, 2019-10-11:
> this isn't something that we should slip into Eloquent at the last minute; we should probably have a pre-announcement on Discourse and our other channels, and then change it for a future distribution instead.

I absolutely agree with that.

BMarchi, 2019-10-11:
I made a mock for the implementation of this feature for FastRTPS. 

- [rmw](https://github.com/ros2/rmw/pull/190)
- [fastrtps](https://github.com/ros2/rmw_fastrtps/pull/331)

I would like your feedback and suggestions!

dirk-thomas, 2019-10-11:
> this isn't something that we should slip into Eloquent at the last minute

That is why we have a very generous time windows ahead of the release to freeze changes like this as well as release notes.

clalancette, 2019-10-11:
> That is why we have a very generous time windows ahead of the release to freeze changes like this as well as release notes.

In my opinion, the ~5 weeks until release isn't nearly enough time.  We tick-tock on API changes; why wouldn't we do something similar for behavioral changes?

dirk-thomas, 2019-10-11:
> We tick-tock on API changes; why wouldn't we do something similar for behavioral changes?

I am really curious for your proposal on how to tick-tock a boolean choice...

gavanderhoorn, 2019-10-11:
> > We tick-tock on API changes; why wouldn't we do something similar for behavioral changes?
> 
> I am really curious for your proposal on how to tick-tock a boolean choice...

Announce-but-not-actually-enable, then enable in later release?

clalancette, 2019-10-11:
> I am really curious for your proposal on how to tick-tock a boolean choice...

We put this feature into Eloquent, preserving the current behavior.  Tests/users can set it as needed, but it would work out-of-the-box like it does in Dashing.  Right after/along with the Eloquent release, we post it to Discourse and other channels, saying we are going to change the default to localhost.  That gives users ~6 months of notice.  And in F-turtle, we change the default to localhost-only.

dirk-thomas, 2019-10-11:
> Announce-but-not-actually-enable, then enable in later release?

There is no difference to the effect. Before the tock it continues to work for people and after the tock it suddenly breaks for them - one month vs. six months announcement ahead doesn't change anything.

gavanderhoorn, 2019-10-11:
> There is no difference to the effect. Before the tock it continues to work for people and after the tock it suddenly breaks for them - one month vs. six months announcement ahead doesn't change anything.

No, but users who want to enable it can.

Those that don't would still just see the current behaviour.

You essentially buy yourself a few months extra time of being "loud about it".

clalancette, 2019-10-11:
> There is no difference to the effect. Before the tock it continues to work for people and after the tock it suddenly breaks for them - one month vs. six months announcement ahead doesn't change anything.

It absolutely changes things; it gives people way more advance notice that things are going to break.  This is the same theory behind tick-tocking; you deprecate with a warning, and then you remove.  At some point if people don't read notices or upgrade, sure, things will break for them.  But you've given them plenty of advance notice (and a whole development cycle) that this is going to happen.

dirk-thomas, 2019-10-11:
Currently there are two options discussed in this thread: `ROS_ALLOWED_HOSTS` and `ROS_ENABLE_NETWORK`.

I mentioned earlier that we are unsure if it is feasible to support selecting multiple hostnames to restrict the communication too. Can anyone provide some references that this is possible across serveral RMW impl? If not, I don't think we should define an environment variable `ROS_ALLOWED_HOSTS` if we might not be able to deliver on the promise to support listing multiple hostnames in the near future.

I understand that we would want to have such kind of feature but I would like to see evidence for this before defining this.

gavanderhoorn, 2019-10-11:
@clalancette: you do agree that this is going to surprise / break things for users regardless right?

dirk-thomas, 2019-10-11:
> you deprecate with a warning

I doubt that we want to print a warning in Eloquent so the extra "tick" cycle doesn't increase visibility.

clalancette, 2019-10-11:
> @clalancette: you do agree that this is going to surprise / break things for users regardless right?

Oh yeah, of course.  Nobody is going to read the notice, and everyone will be surprised when it happens in F-Turtle.  But at least you can then say "We gave you lots of notice"; while people will still grumble, they'll understand.

On the other hand, if you slip it in in the last week before the freeze, everyone will get very upset about it, since it is a major behavior change right before a release.

jacobperron, 2019-10-11:
> I understand that we would want to have such kind of feature but I would like to see evidence for this before defining this.

This is fair. I vote for `ROS_ENABLE_NETWORK` (as a boolean flag) so this feature can land sooner, and we can consider introducing a `ROS_ALLOWED_HOSTS` in the future. Although, if we know that we can select at least one specific network (I don't know), we could leave our options open and use `ROS_NETWORK` that accepts `localhost` or nothing. Then it can be expanded to include other hosts in the future (not necessarily a list of hosts).

Regarding the question of tick-tocking, I don't feel strongly if we wait to flip the switch for F-turtle or not. I think it's fine to change behavior for Eloquent; certainly if we get it in before feature freeze and are loud about it (putting it at the to of [the release notes](https://index.ros.org/doc/ros2/Releases/Release-Eloquent-Elusor/#changes-since-the-dashing-release), Discourse posts, etc).

ivanpauno, 2019-10-11:
I think that adding a boolean environment variable is the best way to go for the moment, as it's not clear if support for a list of hosts could be implemented in all the `rmw` implementations.

Also, I find `ROS_LOCALHOST_ONLY` much clearer than `ROS_ENABLE_NETWORK`.

BMarchi, 2019-10-16:
I kept working on this and I opened 5 PRS to enable this feature for connext and fastrtps:

- [rmw](https://github.com/ros2/rmw/pull/190)
- [rmw_fastrtps](https://github.com/ros2/rmw_fastrtps/pull/331)
- [rmw_connext](https://github.com/ros2/rmw_connext/pull/373)
- [rmw_implementation](https://github.com/ros2/rmw_implementation/pull/75)
- [rcl](https://github.com/ros2/rcl/pull/520)

I would like your opinion/suggestions.

dirk-thomas, 2019-10-18:
@BMarchi Please run CI for the existing PRs as well as create at least stub PRs for OpenSplice and CycloneDDS (those only need the additional argument in the function signature and not implement the actual logic if it is not easy to do at the same time).

jdlangs, 2019-10-18:
Reading through this thread for the first, I'm very surprised at the reactions that this is a major change that requires a long deprecation period over multiple releases. It seems to me to be the ideal sort of breaking change: even if you completely ignore the release notes and try to do business as usual, you'll discover instantly that this has changed and the work you have to do to regain the old behavior is near-trivial.

BMarchi, 2019-10-18:
- [cyclone](https://github.com/ros2/rmw_cyclonedds/pull/50)
- [opensplice](https://github.com/ros2/rmw_opensplice/pull/288)

PRs that add the boolean parameter to cyclone and opensplice without implementing the feature.

dirk-thomas, 2019-10-18:
With the feature in flight what is a good way to restrict pretty much all our tests to use `ROS_LOCALHOST_ONLY=1`? If it becomes a default for running tests how can a user opt-out of this when network communication is actually desired for a test?

hidmic, 2019-10-18:
Well, it not being the default is certainly inconvenient. Maybe an environment hook for `rmw`? Users can then override that environment in `CMake` or directly in `pytest` or `gtest` tests.

dirk-thomas, 2019-10-18:
> Well, it not being the default is certainly inconvenient.

This question is unrelated to the decision on the default value. The tests should still set it to localhost-only in case the user has set it differently.

> Maybe an environment hook for rmw?

An environment hook is not specific to the tests and that is the question here.

`ros_environment` will set the env var in the future to whatever default we choose.

hidmic, 2019-10-18:
> This question is unrelated to the decision on the default value. The tests should still set it to localhost-only in case the user has set it differently.

Fair enough.

> An environment hook is not specific to the tests and that is the question here.

I know, but I don't see any other way to do so for both CMake-based *and* pure Python packages without requiring explicit user intervention. We could adapt CMake macros, setup pytest hooks or even somehow make `colcon test` force an environment, but all these options are brittle IMHO. 

> The tests should still set it to localhost-only in case the user has set it differently.

I personally would argue that people won't develop and/or run tests on their robots. Thus, having tests forcing `ROS_LOCALHOST_ONLY=1` on a `ROS_LOCALHOST_ONLY=0` environment is a rather unlikely scenario. If any roboticist on this thread thinks otherwise, please do tell.

dirk-thomas, 2019-10-18:
Following scenario: we set `ROS_LOCALHOST_ONLY=1` at some point (using an env hook) and a user needs to use network communication and overrides it with `ROS_LOCALHOST_ONLY=0` (e.g. after sourcing a workspace). Does that mean they are not supposed to run the test (without resetting the env var to `1`) or are they supposed to expect that some tests might fail? Both don't seem like desired options to me. Therefore I would prefer if tests explicitly force localhost-only. I don't have a good proposal how to do it that is why I raised the question here for input.

hidmic, 2019-10-18:
> Following scenario: we set ROS_LOCALHOST_ONLY=1 at some point (using an env hook) and a user needs to use network communication and overrides it with ROS_LOCALHOST_ONLY=0 (e.g. after sourcing a [...]

I'd say it adds a bit of overhead. The same overhead it implies to put or lift that restriction if we start automatically forcing a default regardless of the user environment. And not having that restriction does not necessarily imply tests will fail -- they may or may not find other things in the network, but that's about it.

In that sense, we could make tests assert (or warn unless) `ROS_LOCALHOST_ONLY` has a specific value. But that's very specific to each test.

ruffsl, 2019-10-19:
Just discovered this issue and I'm a bit taken back at the idea of disabling network connectivity for ROS2 by default. That seems like one hell-of-a nerf, given that ROS1 always defaulted to enabled networking, and the existing precedent that ROS2 has defaulted to use the middleware vendor's discovery config since initial release. 

> ROS 1 is localhost by default (well, it's actually not, but without special configuration of 'outsiders', it is). ROS 2 is now every reachable host by default.

Even though ROS1 only used one ros-core, it's the fact that the computation graph was accessible through the network from a stock setup that was expected and was the normative use case. E.g. running ROS on robots, then connecting to the robot over the network for control, introspection, or integration with other robots or peripheral sensors/devices.

> I can't stop thinking this is more of a development feature rather than an release enhancement.

I feel a bit this way as well, but admit I'm torn.

Flipping the normative case to localhost-only would be a large departure from how ROS2 has worked over the last +3 years. Of course ROS_DOMAIN_ID was introduced to expose a common DDS parameter, but I'm not sure we have a clear story on how all the source of configurations should resolve, e.g. what is the defined behavior if a user is already customizing which subset networking interfaces DDS should broadcast over via vendor config. If ROS2 adds another knob that only accounts for the loopback, which configuration takes precedence? This seems like a setting that should be deferred to the middleware vendor.

> the set of available domain IDs is a rather limited 

In regards to the limited range of domain numbers, DDS is already adding the concept of domain tags; this could be set to a uuid by the user if need be. This allows the data bus to remain accessible from the network while avoiding crosstalk over common domain ids:

> A domain tag is a logical sub-division within a domain. It is defined as a string. DomainParticipants associated to different domain tags will not discover each other even if they are in the same domain ID. You can think of a Domain Tag as the DDS equivalent of a network VLAN.
> https://community.rti.com/static/documentation/connext-dds/5.3.0/doc/manuals/cloud_discovery_service/core_concepts.html#domain-tags

If a user has an interest to avoid full duplex communication with external network participants, then perhaps they'd be better served by using a middleware more lightweight than DDS, given the IPC requirements would be so much narrower in scope. I don't see why you'd choose to have DDS/RTPS without networking by default instead of using a middleware more appropriate for that usecase.

Using ROS without networking would seem to be the more advanced/niche use case, where users have a configured development environment or continuous integration infrastructure, already curated and tuned with shell scripts, configs. But If I'm teaching a class of students using ROS2 on shared embedded devices, prototyping with a fresh install, I'd prefer the connectivity just worked out of the box, rather than having to remind users to check the networking environment variables for every ros sensor across lab, the ros node on the embedded robot, before launching any ros GUIs on their laptops when they go home.

To be clear, I'm not opposed to this optional setting, I'd just be in favor of not disabling the simple out of the box connectivity behavior, and thinking carefully on how we plan for all the various configs, envs, and global overrides to coexist in a consistent experience for users running ros primarily with robots, not just developers with advance use cases on the same LAN.

dirk-thomas, 2019-10-19:
With the feature implemented I will go ahead and close this ticket. The remaining decision when to enable the flag by default will be made in the upcoming weekly meeting and either ros/ros_environment#19 is merged now or after Eloquent has been branched.

closed 2019-10-19 by dirk-thomas

ruffsl, 2019-10-19:
> > this isn't something that we should slip into Eloquent at the last minute; we should probably have a pre-announcement on Discourse and our other channels, and then change it for a future distribution instead.
> 
> I absolutely agree with that.

@dirk-thomas , was the community ever notified and asked for feedback on this yet? 10 days from opening to finalizing this ticket on a major change to a default user experience seems rushed.

Sidenote; defaulting to local-loopback-only would also break existing ros2 setups using software defined networks, given that ros2 nodes in separate containers on the same machine do not share a common loopback interface by default (as there can be numerous technical restrictions for doing so).

ruffsl, 2019-10-21:
It seems that for many implementations of this feature, the ipv4 address has been hard-coded:

* https://github.com/ros2/rmw_connext/pull/373/files#diff-2469e9c52ef0878455668e9233e96b0aR67
* https://github.com/ros2/rmw_fastrtps/pull/331/files#diff-ddefaaddecf566327aac9fce20040c84R246

For user systems that ship with `net.ipv6.bindv6only=1` set, or containers on ipv6 software defined networks (either out of necessity or security) , ros2 would subsequently fail to bind to the local loopback out of the box. I don't think ROS2 should be making assumptions/restrictions on ipv4 availability.

https://lwn.net/Articles/688462/

dirk-thomas, 2019-10-21:
> For user systems that ship with `net.ipv6.bindv6only=1` set, or containers on ipv6 software defined networks (either out of necessity or security) , ros2 would subsequently fail to bind to the local loopback out of the box.

Please consider to contribute pull request for that use case.

ruffsl, 2019-10-21:
I still don't agree ROS2 is the layer of abstraction we should be making assumptions or if-checks on what type of network interface underpins the data bus. My suggestion would be to hold off on hardcoding ipv4 or if-checks for ipv6 until we can properly call vindor library functions that already includes such logic.

Would you want a PR to revert the hardcodings to avoid breaking eloquent for ipv6 only setups?

dirk-thomas, 2019-10-21:
> Would you want a PR to revert the hardcodings to avoid breaking eloquent for ipv6 only setups?

I don't think reverting the change is desired. The feature is planned for Eloquent - just removing it now isn't a good option. Any pull request improving the logic (e.g. properly call vendor library functions that already includes such logic) would be highly appreciated though.

RFRIEDM, 2020-12-16:
> I don't think reverting the change is desired. The feature is planned for Eloquent - just removing it now isn't a good option. Any pull request improving the logic (e.g. properly call vendor library [...]

I was using Fast-DDS and trying to figure out how to use a certain interface and came across this ticket when looking at how ROS2 specifies an interface to conduct communications on. While Fast-DDS supports [whitelisting](https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/whitelist.html) interfaces, this environment variable method implemented here only solves half the problem. What if I have three interfaces, `lo`, `eth1` and `eth2` and just want to use DDS on `eth2`. 

In my own socket programming, I have used the IPv4 address to bind sockets to interfaces, and for IPv6, it uses the interface name or interface integer instead as the zone id ([reference](https://docs.oracle.com/cd/E19455-01/806-1017/6jab5di2m/index.html#sockets-47146)).

tylerjw, 2022-05-20:
As stated in the Discourse issue I opened. Could a temporary solution (until DDS discovery is normally non-destructive to networks) be enabling this by default?