ros2/ros2 #812: ROS 2 Performance Test Infrastructure

cottsay, 2019-10-22:
Leverage existing performance testing tools to create infrastructure to compare the performance between DDS implementations, track performance trends over time, aid in root-causing performance regressions, and prevent them from happening in the first place.

At the moment, the only suite targeted to run is the [`performance_test`](https://github.com/ApexAI/performance_test) package written by Apex.AI. To that end, there is some outstanding work:
* Support ROS 2 Eloquent (ApexAI/performance_test#89)
* Support multiple participants for CycloneDDS (ApexAI/performance_test#90)

- [ ] Create job in `ros_buildfarm` to run the test suite
       The job must:
    - [x] Build the performance test code
    - [ ] Run the jobs for some number of DDS implementations ([staging repo](https://github.com/cottsay/buildfarm_perf_tests) - not a final design)
    - [ ] Aggregate the results and display them in Jenkins
        - [ ] Facilitate analysis per-build ros-infrastructure/ros_buildfarm#680
        - [ ] Facilitate analysis over time
- [ ] Run tests on AWS Graviton workers (ARM64)

Some other desirable work might be to hook into an external data analysis tool. The tool should ideally be available freely like the rest of the ROS 2 infrastructure is.

ahcorde, 2019-12-03:
I have created this two PRs to include the work taht @cottysay has done:

 - https://github.com/ros2/ros_buildfarm_config/pull/51
 - https://github.com/ros-infrastructure/ros_buildfarm/pull/689

dirk-thomas, 2019-12-17:
Items from past offline conversations:
* [ ] document what exactly is being tested
* [ ] document how to reproduce it locally

Before sharing the results / pointer to the docs on Discourse we should gather feedback from all vendors (to provide "neutral" statistics).

ahcorde, 2020-01-07:
@dirk-thomas  Documentation of how to reproduce it locally is in the [Readme.md](https://github.com/ros2/buildfarm_perf_tests)

ahcorde, 2020-01-07:
We are testing two different things. One one hand we are measuremtn the performance of the DDS vendors and on the other hand we are evaluating the additional overhead caused by a single pub-sub topic or one process spinning and detec potential leaks related to theses activities.

## Test 1 - Performance Test  (APEX.AI)

In this test we are testing the  Performance Test provided by APEX.AI. Right now we have [our own fork](https://github.com/ros2/performance_test) because there are two pending pull request in the official gitlab repository.

In this test we are measurement:
 - Average round-trip time
 - CPU usage (provided by APEX.AI tool)
 - Sent/Received packets per second
 - Total lost packets
 - Max resident set size

We are generating two plots per measurement
 - [One per per-build](http://build.ros2.org/view/Eci/job/Eci__nightly-performance_ubuntu_bionic_amd64/lastBuild/)
 - [Other over time](http://build.ros2.org/view/Eci/job/Eci__nightly-performance_ubuntu_bionic_amd64/plot/)

| Publisher/Subscriber     | rmw_fastrtps_cpp         | rmw_opensplice_cpp       | rmw_cyclonedds_cpp       | rmw_fastrtps_dynamic_cpp | rmw_connext_cpp          |
|--------------------------|--------------------------|--------------------------|--------------------------|--------------------------|--------------------------|
| rmw_fastrtps_cpp         | :heavy_check_mark:       | :heavy_multiplication_x: | :heavy_multiplication_x: | :heavy_multiplication_x: | :heavy_multiplication_x: |
| rmw_opensplice_cpp       | :heavy_multiplication_x: | :heavy_check_mark:       | :heavy_multiplication_x: | :heavy_multiplication_x: | :heavy_multiplication_x: |
| rmw_cyclonedds_cpp       | :heavy_multiplication_x: | :heavy_multiplication_x: | :heavy_check_mark:       | :heavy_multiplication_x: | :heavy_multiplication_x: |
| rmw_fastrtps_dynamic_cpp | :heavy_multiplication_x: | :heavy_multiplication_x: | :heavy_multiplication_x: | :heavy_check_mark:       | :heavy_multiplication_x: |
| rmw_connext_cpp          | :heavy_multiplication_x: | :heavy_multiplication_x: | :heavy_multiplication_x: | :heavy_multiplication_x: | :heavy_check_mark:       |

## Test 2 - Simple pub/sub 

In this case we are testing one publisher and one subscriber **in different processes** sending a 1kArray at 5Hz. This will allow us to 
 evaluate additional overhead caused by a single pub-sub topic and detect leaks related to this
activity.

We measurement for both publisher and subscriber:

 - Average round trip 
 - CPU usage ( readed from the filesystem )
 - Total lost packets
 - Received/Sent packets
 - Physical memory (Mb)
 - Resident anonymous memory (Mb)
 - Virtual memory (Mb)

Again we plot measurement:
 - [One per per-build](http://3.83.10.11/job/Dci__nightly-performance-overhead-multi_ubuntu_bionic_amd64/lastBuild/)
 - [Other over time](http://3.83.10.11/job/Dci__nightly-performance-overhead-multi_ubuntu_bionic_amd64/plot/)

| Publisher/Subscriber     | rmw_fastrtps_cpp   | rmw_opensplice_cpp | rmw_cyclonedds_cpp | rmw_fastrtps_dynamic_cpp | rmw_connext_cpp    |
|--------------------------|--------------------|--------------------|--------------------|--------------------------|--------------------|
| rmw_fastrtps_cpp         | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark:       | :heavy_check_mark: |
| rmw_opensplice_cpp       | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark:       | :heavy_check_mark: |
| rmw_cyclonedds_cpp       | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark:       | :heavy_check_mark: |
| rmw_fastrtps_dynamic_cpp | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark:       | :heavy_check_mark: |
| rmw_connext_cpp          | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark:       | :heavy_check_mark: |

## Test 3 - Node spinning

This test creates one process and spin for 1 minute to evaluate ROS 2 overhead and detect obvious leaks.

We measurement:

 - CPU usage ( readed from file system )
 - Physical memory
 - Resident anonymous memory
 - Virtual memory

Again we plot measurement:
 - [One per per-build](http://3.83.10.11/job/Dci__nightly-performance-overhead_ubuntu_bionic_amd64/lastBuild/)
 - [Other over time](http://3.83.10.11/job/Dci__nightly-performance-overhead_ubuntu_bionic_amd64/plot/Node%20Spinnig%20Results/)

| DDS                      | Process 1 |
|--------------------------|-----------|
| rmw_fastrtps_cpp         | :heavy_check_mark:   |
| rmw_opensplice_cpp       |  :heavy_check_mark:   |
| rmw_cyclonedds_cpp       | :heavy_check_mark:   |
| rmw_fastrtps_dynamic_cpp |  :heavy_check_mark:   |
| rmw_connext_cpp          |   :heavy_check_mark:   |

ahcorde, 2020-01-07:
# Notes

Test 2 and Test 3 are not upstream yet.

Where I should locate this documentation?

ahcorde, 2020-01-17:
https://github.com/ros2/buildfarm_perf_tests/pull/11

chapulina, 2020-02-01:
👍

closed 2020-02-01 by chapulina