ros2/ros2 #813: Runtime error with poco 1.9.4 on Dashing Patch 4

minggangw, 2019-10-22:
## Bug report

**Required Info:**

- Operating System:
  - macOS high sierra with xcode 9.4.1
- Installation type:
  - Binary of Dashing patch 4 (https://github.com/ros2/ros2/releases/tag/release-dashing-20191018)

#### Steps to reproduce issue
Run a program which needs librosidl_typesupport_c.dylib

#### Expected behavior
Run successfully
#### Actual behavior
```Bash
Error: dlopen(/Users/distiller/RobotWebTools/rclnodejs/build/Release/rclnodejs.node, 1): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/distiller/ros2_install/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
```
see [details](https://circleci.com/gh/minggangw/rclnodejs-1/912#tests/containers/0)
#### Additional information
This happens after the upgrade of poco to [1.9.4](https://formulae.brew.sh/formula/poco), I test with 1.9.3 and the error goes away, please see [here](https://circleci.com/gh/minggangw/rclnodejs-1/920?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link)

This also happen on the latest packaging at https://ci.ros2.org/view/packaging/job/packaging_osx/

nuclearsandwich, 2019-10-31:
High Sierra isn't a supported platform for the binary releases. Dashing has only ever officially supported macOS Sierra and as that platform has been dropped by Apple and Homebrew, Dashing patch 4's binaries are the last we'll release for that platform.

Building Dashing from source on recent versions of macOS hasn't been tested but as long as we can maintain compatibility with Sierra we can review PRs backporting support for newer platforms to Dashing.

> This also happen on the latest packaging at [ci.ros2.org/view/packaging/job/packaging_osx](https://ci.ros2.org/view/packaging/job/packaging_osx/)

We generally upgrade our macOS and Windows hosts as closely before a patch release as we can to make sure that binaries for the release have a decent shelf life. I'll take a look at what updates are available in homebrew and see if there are few enough we can do an extra round of upgrades without too much disruption.

wjwwood, 2019-11-21:
Can this be closed?

closed 2019-11-21 by wjwwood