ros2/ros2 #816: By default in ros2 the one in charge of sending data in fastrtps is an asynchronous thread. Do you know which is the

golizheh, 2019-10-30:
By default in ros2 the one in charge of sending data in fastrtps is an asynchronous thread. Do you know which is the 
depth of the history?  If history's depth is 1 and in the moment of writing a new sample the asynchronous thread didn't 
send the previous one then there is no space in history to store the new sample and the write function fails.

We are pending on implement a timeout for this cases, a maximum blocking time where write function waits until there is 
space on the history or fails.
↪ 2018-07-19 jue 20:17, Shane Loretz <notifications@github.com>:
>I'm unable to reproduce this on bionic.
>
>It comes from [rmw_publish.cpp:53](https://github.com/ros2/rmw_fastrtps/blob/7d3ff15e40999d878fb67104987b3d654d48db17/rmw_fastrtps_cpp/src/rmw_publish.cpp#L53) which happens when [eprosima::fastrtps::PublisherImpl::create_new_change_with_params()](https://github.com/eProsima/Fast-RTPS/blob/54f563a697e7de5e3c924a99fa1b2ce3877f69e7/src/cpp/publisher/PublisherImpl.cpp#L83) returns false.
>
>@richiware may I ask if you have any ideas about why a this call might have failed? If it can be reproduced, what info would help determine the cause?
>
>-- 
>You are receiving this because you were mentioned.
>Reply to this email directly or view it on GitHub:
>https://github.com/ros2/ros2/issues/539#issuecomment-406368451

_Originally posted by @richiware in https://github.com/ros2/ros2/issues/539#issuecomment-406529592_

allenh1, 2019-10-31:
@golizheh did you mean to close this issue? Seems you cleared out all the info

nuclearsandwich, 2019-10-31:
Closing for now. We can re-open it if there's something to report.

closed 2019-10-31 by nuclearsandwich