ros2/ros2 #819: Generate code from msg failed

jhx2002, 2019-11-04:
This test.msg is OK: 
```
std_msgs/Header    header
geometry_msgs/Pose[] poses
#int32[] idxx
```

But following test.msg failed, the only difference is one line, so what's the problem?
```
std_msgs/Header    header
geometry_msgs/Pose[] poses
int32[] idxx
```
Here is error code:
```
AttributeError: 'Token' object has no attribute 'children'
make[2]: *** [rosidl_typesupport_fastrtps_cpp/mymsg/msg/dds_fastrtps/trajectory__type_support.cpp] Error 1
make[1]: *** [CMakeFiles/mymsg__rosidl_typesupport_fastrtps_cpp.dir/all] Error 2
'Token' object has no attribute 'children' /home/huaxing/proj/ros2_nodes/mymsg/build/mymsg/rosidl_adapter/mymsg/msg/Trajectory.idl
Error processing idl file: /home/huaxing/proj/ros2_nodes/mymsg/build/mymsg/rosidl_adapter/mymsg/msg/Trajectory.idl
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/rosidl_typesupport_introspection_cpp/rosidl_typesupport_introspection_cpp", line 23, in <module>
    sys.exit(main())
  File "/opt/ros/dashing/lib/rosidl_typesupport_introspection_cpp/rosidl_typesupport_introspection_cpp", line 19, in main
    return generate_cpp(args.generator_arguments_file)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_typesupport_introspection_cpp/__init__.py", line 24, in generate_cpp
    generate_files(generator_arguments_file, mapping)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 90, in generate_files
    raise(e)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 70, in generate_files
    idl_file = parse_idl_file(locator)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 64, in parse_idl_file
    content = parse_idl_string(string, png_file=png_file)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 73, in parse_idl_string
    content = extract_content_from_ast(tree)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 144, in extract_content_from_ast
    add_message_members(msg, struct_defs[0])
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 365, in add_message_members
    abstract_type = get_abstract_type_from_type_spec(type_spec)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 406, in get_abstract_type_from_type_spec
    return get_abstract_type(child)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 439, in get_abstract_type
    basetype = get_abstract_type_from_type_spec(type_spec)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 406, in get_abstract_type_from_type_spec
    return get_abstract_type(child)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_parser/parser.py", line 415, in get_abstract_type
    while len(child.children) == 1:
AttributeError: 'Token' object has no attribute 'children'
make[2]: *** [rosidl_typesupport_introspection_cpp/mymsg/msg/trajectory__rosidl_typesupport_introspection_cpp.hpp] Error 1
make[1]: *** [CMakeFiles/mymsg__rosidl_typesupport_introspection_cpp.dir/all] Error 2
make: *** [all] Error 2

```

dirk-thomas, 2019-11-04:
What version of the `python3-lark-parser` package do you have installed?

jhx2002, 2019-11-05:
/usr/local/lib/python3.6/dist-packages/lark_parser-0.6.5-py3.6.egg-info
@dirk-thomas

dirk-thomas, 2019-11-05:
The current version of the Debian package is `0.7.2-1osrf~` so you should make sure you have an up-to-date system and try again.

jhx2002, 2019-11-05:
Same error after install lark_parser 0.7.6. @dirk-thomas

dirk-thomas, 2019-11-05:
> Same error after install lark_parser 0.7.6.

There is no such version available from Debian packages. Please uninstall any custom versions you installed (probably via pip) and only use the latest Debian package.

The provided `.msg` file parses / compiles just fine for me. If it still fails for you please provide a complete example (all involved files) with reproducible steps.

jhx2002, 2019-11-08:
use this release  also failed:  @dirk-thomas 
https://github.com/lark-parser/lark/archive/0.7.2.tar.gz

dirk-thomas, 2019-11-08:
Please see my previous comment:

> only use the latest Debian package.

> If it still fails for you please provide a complete example (all involved files) with reproducible steps.

jhx2002, 2019-11-08:
Do you mean use update lark_parser like this: @dirk-thomas 
'apt install lark_parser'

jhx2002, 2019-11-11:
I'v fixed it with``` pip3 install --upgrade lark_parser ```

closed 2019-11-11 by jhx2002

dirk-thomas, 2019-11-11:
> I'v fixed it with `pip3 install --upgrade lark_parser`

On a system where Debian packages you should only use those and avoid installing the same packages with `pip` since they will overlay the Debian package and not get updated automatically. So if we ever bump the required version of this package your environment will fail again.