ros2/ros2 #821: ROS 2 Microsoft - python version and path problem

ahcorde, 2019-11-15:
# Bug report

**Required Info:**

- Operating System:
   - Windows 10
- Installation type:
  -  binaries>
- Version or commit hash:
  - ros2-dashing-20191018-windows-release-amd64
- DDS implementation:
  - rmw_fastrtps_cpp
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

I'm trying to install ROS2 on windows following the instructions provided here https://index.ros.org/doc/ros2/Installation/Dashing/Windows-Install-Binary/#troubleshooting

The first issue that I found is that I need to install python 3.7 manually because `choco` is installing python38 by default

```
choco install python --version=3.7.2
```

All the python scripts are using version 3.7 I found this line in the header

```
#!c:\python37\python.exe
....
```

Another problem happens when I try to launch the python talker

```
C:\dev\ros2>ros2 run demo_nodes_py talker
None
Traceback (most recent call last):
  File "C:\dev\ros2\ros2-windows\lib\demo_nodes_py\talker-script.py", line 11, in <module>
    load_entry_point('demo-nodes-py==0.7.9', 'console_scripts', 'talker')()
  File "C:\dev\ros2\ros2-windows\Lib\site-packages\demo_nodes_py\topics\talker.py", line 39, in main
    rclpy.init(args=args)
  File "C:\dev\ros2\ros2-windows\Lib\site-packages\rclpy\__init__.py", line 75, in init
    return rclpy_implementation.rclpy_init(args if args is not None else sys.argv, context.handle)
RuntimeError: Failed to initialize init_options: failed to load shared library of rmw implementation. Exception: Cannot load library: C:\dev\ros2\ros2-windows\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:137, at C:\J\workspace\packaging_windows\ws\src\ros2\rcl\rcl\src\rcl\init_options.c:55
````

For some reason is trying to find rmw_fastrtps_cpp here

```
C:\dev\ros2\ros2-windows\bin/rmw_fastrtps_cpp.dl
```

But I think the backslash is breaking the path

mjcarroll, 2019-11-15:
FYI we just added this to the eloquent instructions: https://github.com/ros2/ros2_documentation/pull/418

ahcorde, 2019-11-15:
Regarding the rmw path? Any ideas?

ahcorde, 2019-11-18:
Adding more information:

```
Traceback (most recent call last):
  File "C:\dev\ros2_dashing\install\lib\demo_nodes_py\talker-script.py", line 11, in <module>
    load_entry_point('demo-nodes-py==0.7.9', 'console_scripts', 'talker')()
  File "C:\dev\ros2_dashing\install\Lib\site-packages\demo_nodes_py\topics\talker.py", line 39, in main
    rclpy.init(args=args)
  File "C:\dev\ros2_dashing\install\Lib\site-packages\rclpy\__init__.py", line 73, in init
    return rclpy_implementation.rclpy_init(args if args is not None else sys.argv, context.handle)
RuntimeError: Failed to initialize init_options: failed to load shared library of rmw implementation. Exception: Cannot load library: C:\dev\ros2_dashing\install\bin/rmw_fastrtps_cpp.dll, at C:\dev\ros2_dashing\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:137, at C:\dev\ros2_dashing\src\ros2\rcl\rcl\src\rcl\init_options.c:55
```

it's something related with python.exe

[image]

Translation: **openssl_sk_new_reserve could not be located in the dynamic link library**

ahcorde, 2019-11-18:
Fixed the python version must be `3.7.5`

closed 2019-11-18 by ahcorde

mjcarroll, 2019-11-18:
This may also be that the wrong version of OpenSSL is installed.  We recommend 1.0.2 in the instructions, but that error seems to point to 1.1.0?

ahcorde, 2019-11-18:
I'm running the examples with OpenSSL 1.1.1d. It looks good for now