ros2/ros2 #822: Adding support for `ON_REQUESTED_INCOMPATIBLE_QOS` and `ON_OFFERED_INCOMPATIBLE_QOS` event callbacks in ROS2.

jaisontj, 2019-11-16:
#### Feature description
Currently, it is possible for ROS2 users to create publishers and subscriptions to a topic with incompatible QoS policies; for example: subscription to a topic with a deadline smaller than the deadline for a publisher to the topic. This results in the subscription not receiving any data even though the publisher is publishing to the topic. These scenarios are difficult to debug and hence the user needs to be notified/warned when they create publishers/subscriptions with incompatible QoS Policies. 

Thankfully, the DDS spec has callbacks in place which help with this, namely `on_requested_incompatible_qos` and `on_offered_incompatible_qos`.

#### Implementation considerations
The implementation can follow the same procedure as the existing event callbacks (`on_offered_deadline_missed`, `on_requested_deadline_missed`, `on_liveliness_lost` and `on_liveliness_changed`).

Listing out briefly the work required in each package:

- [ ] Changes in `ros2/rmw`
  * Add the RMW_EVENT_REQUESTED_INCOMPATIBLE_QOS and RMW_EVENT_OFFERED_INCOMPATIBLE_QOS values to the enum `rmw_event_type_t` 
  * Define two new structures to hold data regarding the status of these events: `rmw_requested_incompatible_qos_status_t` and `rmw_offered_incompatible_qos_status_t`. 

Something like this:

```c
struct  rmw_requested_incompatible_qos_status_t
{
   int32_t total_count;
   int32_t total_count_change;
   int32_t last_policy_id;
} ;
```
***Note*: `QosPolicyCountSeq policies` is something that I have skipped and would like to add as an enhancement once this is completed and merged. This is because we need to figure out how memory allocation for the `policies` array would work. For now these would alone provide a non zero enhancement to the system as users will be notified when creating incompatible qos policies between publishers and subscribers**

- [ ] `ros2/rcl`
  * Add `RCL_PUBLISHER_OFFERED_INCOMPATIBLE_QOS` and `RCL_SUBSCRIPTION_REQUESTED_INCOMPATIBLE_QOS` to enums `rcl_publisher_event_type_t` and `rcl_subscription_event_type_t` respectively.
  * Handle event types of `RCL_PUBLISHER_OFFERED_INCOMPATIBLE_QOS` and `RCL_SUBSCRIPTION_REQUESTED_INCOMPATIBLE_QOS` inside of `rmw_publisher_event_init` and `rmw_subscriber_event_init`
  * Unit tests similar to [these](https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_events.cpp#L346)

- [ ] `ros2/rclpy`
  * Add `RCL_PUBLISHER_OFFERED_INCOMPATIBLE_QOS` and `RCL_SUBSCRIPTION_REQUESTED_INCOMPATIBLE_QOS` to enums `QoSPublisherEventType` and `QoSSubscriptionEventType`
  * Define two new payloads that mirror `rmw_requested_incompatible_qos_status_t` and `rmw_offered_incompatible_qos_status_t` that were defined in the `rmw` package. These are defined as follows:

```python
QoSRequestedIncompatibleQoSInfo = NamedTuple(
 'QoSRequestedIncompatibleQoSInfo', [
 ('total_count', 'int'),
 ('total_count_change', 'int'),
 ('last_policy_id', 'int'),
 ])
 
 QoSOfferedIncompatibleQoSInfo = NamedTuple(
 'QoSOfferedIncompatibleQoSInfo', [
 ('total_count', 'int'),
 ('total_count_change', 'int'),
 ('last_policy_id', 'int'),
 ])
```

  * Enhance the `SubscriptionEventCallbacks` and `PublisherEventCallbacks` classes to support the new  `incompatible_qos`  callback.
    * define new member `incompatible_qos`
    * handle it in the `create_event_handlers` function
  * In `_rclpy_qos_event.c`
    * Add two entries of type `rmw_requested_incompatible_qos_status_t` and `rmw_offered_incompatible_qos_status_t` to the union `_qos_event_callback_data` 
    * Define and implement two functions which convert `rmw_requested_incompatible_qos_status_t` and `rmw_offered_incompatible_qos_status_t` to `PyObject` and return these  in the `_get_qos_event_data_filler_function_for` function
  * Unit tests similar to the ones [here](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_qos_event.py)

- [ ] `ros2/rclcpp`
  * Modify the `PublisherEventCallbacks` and `SubscriptionEventCallbacks` structs to include a new `incompatible_qos_callback`. These callback members should be of type `std::function<void (rmw_offered_incompatible_qos_status_t &)>` and `std::function<void (rmw_requested_incompatible_qos_status_t &)>` respectively.
  * Handle the above defined callbacks inside the constructor for `Publisher` and `Subscription`

- [ ] `ros2/demos`
  * Add an example showcasing this feature in the `quality_of_service_demo` package; using both `rclpy` and `rclcpp`

jaisontj, 2019-11-16:
List of PRs to be tracked towards this: 
- [ ] [ros2/rmw#193](https://github.com/ros2/rmw/pull/193/files)
- [ ] [ros2/rcl#535](https://github.com/ros2/rcl/pull/535)
- [ ] [ros2/rclpy#459](https://github.com/ros2/rclpy/pull/459)
- [ ] [ros2/rclcpp#924](https://github.com/ros2/rclcpp/pull/924/files)
- [ ] [ros2/demos#416] (https://github.com/ros2/demos/pull/416)
- [ ] [ros2/rmw_opensplice#294](https://github.com/ros2/rmw_opensplice/pull/294/files)

jaisontj, 2019-11-27:
@ivanpauno Could you please take a look at this?

jaisontj, 2019-11-27:
Extra feature: Adding incompatible event callbacks to `ros2 topic echo` and `ros2 topic pub`
PR: [ros2/ros2cli:410](https://github.com/ros2/ros2cli/pull/410)

wjwwood, 2020-01-25:
I gave an initial review of the connect pull requests. I think we need to do a few things related to `rmw_fastrtps_cpp` (our default currently) not being able to implement this yet:

- I'm hesitant to merge it without support in fast-rtps, and at least there should be support in our other tier 1, `rmw_connext_cpp`.
- I would like to see more detail on what happens when an rmw implementation does not support this, but the user configures it to be on. How does it fail? Does it fail? Is that an acceptable user experience, etc...

I'm inclined to be flexible on these points, so we can get it in if possible, because for the implementations where it does work it will be very useful.

mm318, 2020-03-12:
> * I would like to see more detail on what happens when an rmw implementation does not support this, but the user configures it to be on. How does it fail? Does it fail? Is that an acceptable user experience, etc...

@wjwwood, if the user registers incompatible QoS event callbacks and the RMW implementation being used does not support it, how about we log a warning and the user's callbacks will never be triggered? If the user does not register any incompatible QoS event callbacks, we do not log/issue such a warning.

wjwwood, 2020-03-12:
I don't feel like a warning is sufficient myself. If the user wanted a callback for this, it should fail programmatically with an exception or return code. If the user chooses to do so, they may ignore this error and log a warning themselves. But allowing it to essentially pass silently when the user asks to be called when this event occurs (warnings from libraries are often overlooked) I think is a bad idea.

Also, I think unless the user defines their own callback it would be nice to have a default callback which logs a warning when this event occurs. It would be ok, in this default case, to ignore such an error as "unimplemented" I think. Though I'd like to see a notice or warning that this is the case, something like (once) "[WARN] the rmw implementation being used does not support ON_REQUESTED_INCOMPATIBLE_QOS events, you will not be notified when publishers and subscribers on the same topic do not have matching QoS".

mm318, 2020-03-12:
Okay, that's reasonable. Between throwing an exception and returning an error code, this is how I imagine them:
- Exception: A throw occurs at time of attempting to register a callback in `rclcpp`/`rclpy`. `rclcpp`/`rclpy` will query the `rmw` implementation if such an event callback is supported. This behavior can be made universal for all event types, not just for incompatible QoS mismatch events. This will require adding new `rcl`/`rmw` APIs.
- Error code: An `rmw` implementation that doesn't support this event will fire on startup one occurrence of this event, but the `last_policy_kind` field will contain a value that indicates an error, as opposed to a value that indicates a type of incompatible QoS policy. It is up to the user how to handle this error code.

I think having a default callback is a good idea as well.

wjwwood, 2020-03-12:
Registration of this callback happens at creation of the pub/sub right? In that case I think it would be best to just have the creation fail. The caller can check the reason and retry without the callback if they so chose. That would avoid the need for "is supported" like API's.

mm318, 2020-03-12:
Correct, the registration of the callbacks happen at the creation of the publisher/subscription. "Is supported"-like APIs would still be needed even if we're returning an error code instead of an exception on creation of the publisher/subscription, or else those callbacks would just simply never fire.

Would returning an error code work? If the creation of a publisher/subscription fails, it will return a NULL `shared_ptr`. Where will the caller check the reason of failure?

wjwwood, 2020-03-12:
Oh I see, nvm, you're right that the support is not checked below rmw at creation of the pub/sub.

Instead, we could check if you tried to add those to the wait set and error at that point. So you'd get no exception if you created a pub/sub with the callback registered, but when you tried to spin on it, then it would fail due to not supporting that kind of event in the wait set.

mm318, 2020-03-12:
I think there is still a way. When `create_publisher()`/`create_subscription()` from `rclcpp` is called, it will create `rmw_publisher_t`/`rmw_subscription_t` as well as some `rmw_event_t`s. We can check at the time of creating the `rmw_event_t`s if the event type is supported or not, and bubble up the `rmw_ret_t` error to `rclcpp`'s `create_publisher()`/`create_subscription()`.

mm318, 2020-03-12:
> Instead, we could check if you tried to add those to the wait set and error at that point. So you'd get no exception if you created a pub/sub with the callback registered, but when you tried to spin on it, then it would fail due to not supporting that kind of event in the wait set.

That sounds like a bad experience. How would the user code respond at the point? It would need to recreate a bunch of publishers and subscriptions, as we don't support removing callbacks from already created publishers/subscriptions. Even if we did, I think that would make the user's code quite messy (ie. create publishers/subscriptions, do a trial spin, remove some callbacks, start spinning for real).

wjwwood, 2020-03-12:
> That sounds like a bad experience.

I agree, I was just offering that we could check there too. Sounds like checking when we create the event types is the right way to do it.

mm318, 2020-03-12:
So we'll add a new `is_event_type_supported` `rmw_implementation` API? We won't need to bubble this up to `rcl`; `rmw` just needs to communicate with the underlying `rmw_implementation` and fail at `rmw_publisher_event_init()`/`rmw_subscription_event_init()`.

ivanpauno, 2020-03-12:
> I agree, I was just offering that we could check there too. Sounds like checking when we create the event types is the right way to do it.

:+1:

> So we'll add a new is_event_type_supported rmw_implementation API? We won't need to bubble this up to rcl; rmw just needs to communicate with the underlying rmw_implementation and fail at rmw_publisher_event_init()/rmw_subscription_event_init().

Can't `rmw_publisher_event_init` just fail when the passed event type isn't supported?

mm318, 2020-03-12:
> Can't `rmw_publisher_event_init` just fail when the passed event type isn't supported?

That's what I am suggesting we do, but `rmw` will need to query the `rmw_implementation` via a new API to see if the passed event type is supported or not.

ivanpauno, 2020-03-12:
> That's what I am suggesting we do, but rmw will need to query the rmw_implementation via a new API to see if the passed event type is supported or not.

Oh, I didn't see that that function was implemented in `rmw` itself. Your proposal SGTM!

mm318, 2020-03-13:
I think we still haven't settled on how `rclcpp`/`rclpy` `create_publisher()` should respond to a failing `rmw_publisher_event_init()` call. These are some example ways that I can think of:
- `rclcpp`'s/`rclpy`'s `create_publisher()` will throw an exception
- `rclcpp`'s/`rclpy`'s `create_publisher()` will return `nullptr`/`None` and set an error code somewhere (is this currently being done or possible in `rclcpp`/`rclpy`?)
- `rclcpp`'s/`rclpy`'s `create_publisher()` will return a valid publisher that does not have the requested callback registered, and set an error code somewhere

What do you think, @wjwwood and @ivanpauno?

ivanpauno, 2020-03-13:
> rclcpp's/rclpy's create_publisher() will throw an exception

That's the best option IMO.

mm318, 2020-03-16:
Is it recommendable for a function in `rmw` to call a function in `rmw_implementation`? I am currently running into many instances of undefined reference errors when trying to build unit tests.

wjwwood, 2020-03-17:
Any function that is implemented by the rmw implementations needs to also be added in `rmw_implementation`.

mm318, 2020-03-17:
My understanding is that the problem is more complicated than that.

The situation is `rmw_publisher_event_init()` is implemented in `rmw`, and it calls `rmw_event_type_is_supported()`, which is implemented in the `rmw_implementation`.

So since `rmw_fastrtps_shared_cpp`/`rmw_connext_shared_cpp` have dependency on `rmw`, their unit tests will try to link against `librmw.so` but they will now be missing `rmw_event_type_is_supported()` unless they're linked against `rmw_implementation` as well.

Another weirder situation is `rmw`'s own unit tests will need to link against `librmw.so` and will also be missing `rmw_implementation`. But `rmw` can't be have a dependency on `rmw_implementation`, because it will end up being circular, and `colcon` is actually refuses to build in this situation.

I am just going to move `rmw_publisher_event_init()` and `rmw_subscription_event_init()` to `rmw_implementation`.

wjwwood, 2020-03-17:
Functions in `rmw` cannot call functions implemented in the rmw implementation.

If you have to call the "is supported" function from the event init functions, then you have to either:

- move the event init functions into the middleware implementations, or
- move the event init functions into the middleware implementations and drop the "is supported" functions all together.

mm318, 2020-03-17:
> Functions in `rmw` cannot call functions implemented in the rmw implementation.

Yeah, that's what I found out.

> * move the event init functions into the middleware implementations and drop the "is supported" functions all together.

I have indeed implemented it this way. The following are the set of pull requests related to this feature:
- https://github.com/ros2/rmw/pull/202
- https://github.com/ros2/rmw_implementation/pull/88/files
- https://github.com/ros2/rmw_fastrtps/pull/354
- https://github.com/ros2/rmw_connext/pull/397
- https://github.com/ros2/rmw_cyclonedds/pull/115
- https://github.com/ros2/rcl/pull/600
- https://github.com/ros2/rclcpp/pull/1026

(`rclpy` did not need any changes as far as I can tell.)

ivanpauno, 2020-03-17:
@mm318 @wjwwood The proposed solution is ok, though it involves some duplicated code.
Another way to solve the dependency inversion problem is to avoid showing an error in `rmw`, and instead do it in the next layer (`rcl`). i.e.: `rmw_event_type_is_supported` is directly called from `rcl_*_event_init`.

mm318, 2020-03-17:
If there are no strong objections, I'd like to keep it the way it is. We're now at least able to properly check the `implementation_identifier` of the `rmw_publisher_t`/`rmw_subscription_t` that was passed in (we used to only check it against `nullptr`).

wjwwood, 2020-03-17:
I don’t care strongly either though I have a slight preference for keeping it as is. It leaves the door open for other optimizations for the middleware impl and it ensures we don’t get in an inconsistent state event wise (if someone were to use rmw directly but not check if it is supported first, for example).

ivanpauno, 2020-03-17:
@mm318 It would be great to open a PR in `rclpy` too.

mm318, 2020-03-17:
> @mm318 It would be great to open a PR in `rclpy` too.

Looking at the code in `rclpy`, I didn't think any changes were needed as a different error message will just be gotten from `rcl_get_error_string()`.

@ivanpauno, perhaps do you think [this](https://github.com/ros2/rclpy/blob/78689f57a4511210cb7bd43d42b4f356453c5327/rclpy/src/rclpy/_rclpy_qos_event.c#L35) should be a made into a `PyExc_TypeError` instead when `ret == RCL_RET_UNSUPPORTED`?

ivanpauno, 2020-03-17:
> @ivanpauno, perhaps do you think this should be a made into a PyExc_TypeError instead when ret == RCL_RET_UNSUPPORTED?

I think that a custom exception (e.g.: `EventTypeUnsupported`) would be the best, though it's not required. You can left the code as-is, and a custom exception can be added in the future if really needed.

ivanpauno, 2020-03-17:
This one still should be finished with the others: https://github.com/ros2/rclpy/pull/459.
right?

mm318, 2020-03-17:
Right. There are two sets of pull requests now.

We should merge in this set first:
- https://github.com/ros2/rmw/pull/202
- https://github.com/ros2/rmw_implementation/pull/88
- https://github.com/ros2/rmw_fastrtps/pull/354
- https://github.com/ros2/rmw_connext/pull/397
- https://github.com/ros2/rmw_cyclonedds/pull/115
- https://github.com/ros2/rcl/pull/600
- https://github.com/ros2/rclcpp/pull/1026

Then look at this set of pull requests (`rmw` implementation repos will be added later to this set):
- https://github.com/ros2/rmw/pull/193
- https://github.com/ros2/rcl/pull/535
- https://github.com/ros2/rclcpp/pull/924
- https://github.com/ros2/rclpy/pull/459
- https://github.com/ros2/ros2cli/pull/410
- https://github.com/ros2/demos/pull/416

ivanpauno, 2020-03-19:
> Right. There are two sets of pull requests now.

First set have been merged!

mm318, 2020-03-26:
> Also, I think unless the user defines their own callback it would be nice to have a default callback which logs a warning when this event occurs. It would be ok, in this default case, to ignore such [...]

@wjwwood can we implement this as a follow-up?

wjwwood, 2020-03-26:
Yeah, sure.

mm318, 2020-04-01:
I have opened issues https://github.com/ros2/rclcpp/issues/1048 and https://github.com/ros2/rclpy/issues/533 for tracking the addition of default incompatible QoS callbacks to `rclcpp` and `rclpy` respectively.

mm318, 2020-04-15:
All implementation work (except for `rmw_fastrtps_cpp`, which is gated on support by FastRTPS) is now complete. This issue may be closed. (Remaining `rmw_fastrtps_cpp` work is still being tracked at https://github.com/ros2/rmw_fastrtps/issues/356.)

ivanpauno, 2020-04-15:
Sounds good, thanks for your work @jaisontj @mm318 !

closed 2020-04-15 by ivanpauno