ros2/ros2 #824: CycloneDDS - Connext comm issues

hidmic, 2019-11-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 AMD64
- Installation type:
  - From source
- Version or commit hash:
  - Current [ros2.repos](https://github.com/ros2/ros2/blob/a326ca46d03c9db647dad5583b92a0ed52ca466f/ros2.repos)

#### Steps to reproduce issue

1. On one terminal run a Connext publisher:

```sh
RMW_IMPLEMENTATION=rmw_connext_cpp ros2 topic pub /connext/echo std_msgs/msg/String
```

2. On another terminal run a CycloneDDS publisher:

```sh
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 topic pub /cyclone/echo std_msgs/msg/String
```

#### Expected behavior
Each publisher behaves normally.

#### Actual behavior
CycloneDDS publisher often logs something like the following:

```console
1574109326.702695 [0]     recvUC: malformed packet received from vendor 1.1 state parse:acknack <52545053 02010101 0309a8c0 99090000 01000000 0e010c00 a4f61001 78e1dfcd @0x24 06011800 c7040000 c2040000 00000000 00000000 00000000 02000000> (note: maybe partially bswap'd) {{6,1,24},4c7,4c2,0,0}
```

#### Additional information
I'd think it's a cross-vendor discovery issue, but I can't confirm.

hidmic, 2019-11-18:
cc @eboasson

eboasson, 2019-11-18:
Thanks for letting me know. Connext being sloppy again …

I'll white-list it, just like I've white-listed other known violations where there is a reasonable interpretation. Given the suspicion that it is a discovery issue, I am guessing it is a so-called "pre-emptive ACKNACK" message, something that is rarely truly needed (only to trigger a protocol restart after re-establishing a connection in some cases), but that pretty much every implementation sends also on initially establishing a connection.

This just happens to be my first encounter with Connext sending out this particular malformed message. Interestingly, I've encountered exactly this form from Fast-RTPS before, so for Fast-RTPS it is white-listed already. It annoys me that they don't use a well-formed one for it …

And because of that annoyance you're getting the gory details (which takes me far longer to write than the white-listing itself takes 😁):

It decodes as:
* 6: ACKNACK
* 1: little-endian
* 24: octets to next submessage header starting from the next byte
* 4c7: reader entity id ("subscriptions reader" discovery reader)
* 4c2: writer entity id ("publications writer" discovery writer)
* 0: base sequence number (it ACKs all sequence numbers < 0)
* 0: NACK bitmap size
* (and 2: ACKNACK "count", a sequence number of the ACKNACK message itself)

The problem is that well-formedness requirements include:
* the sequence number be ≥1 (section 8.3.5.5);
* the size of the bitmap be ≥1 on the UDP transport for DDSI versions < 2.3 (section 9.4.2.6; and this message claims to be protocol version 2.1, from the second word in the dump) .

The latter restriction has been ignored pretty much by all implementations in the messages they generate because Connext had (has?) the unfortunate habit of mishandling ACKNACK messages that acknowledge everything published so far that are well-formed in DDSI < 2.3.

Those (necessarily) have a NACK bitmap extending beyond the last sequence number, but with the bits corresponding to those still-to-be-written all set to 0, indicating they are not part of the set of sequence numbers for which a retransmit is requested. But it triggers/triggered a response from Connext that resulted in an endless loop …

eboasson, 2019-11-19:
@hidmic the warnings should be gone now, but I can't verify it for myself because I don't have access to Connext. I would appreciate it if you could check it.

hidmic, 2019-11-21:
Confirmed it now works as expected! I'll close this ticket.

closed 2019-11-21 by hidmic

mjcarroll, 2019-11-21:
@hidmic Can you also verify this when I get the debs out?

hidmic, 2019-11-21:
Confirmed!