ros2/ros2 #830: ROS 2 Foxy Fitzroy

jacobperron, 2019-11-23:
This meta ticket aggregates the tasks targeted for the next ROS 2 release, Foxy Fitzroy.

The release date is May 23, 2020. 

Tasks will be updated as they progress.

---

### Planned for Inclusion

TBD

---

### Stretch Goals

*We think that these items are important but don't currently have the bandwidth to commit to them; if you can help out, let us know!*

TBD

kyrofa, 2019-12-10:
Canonical's contributions for Foxy:

* Security enhancements
    * [ ] Finalize [NoDL design](https://github.com/ros2/design/pull/266)
    * [ ] Design for NoDL integration in ROS 2 security
    * [ ] Tooling for installing/parsing designed NoDL
* New security support
    * [ ] Logging plugin for Fast RTPS
    * [ ] Support for Cyclone DDS security plugins in Cyclone RMW

We are also submitting various security fixes, bug fixes and quality improvements across the stack as a result of scans and code analysis. Will update this list over time.

joespeed, 2019-12-16:
ADLINK's contributions for Foxy:
- [ ] work with Canonical on SROS 2 support for Eclipse Cyclone DDS security plugins and permissions files
- [ ] testing & optimization of DDS Security v1.1 eclipse-cyclonedds/cyclonedds#132 [security branch](https://github.com/eclipse-cyclonedds/cyclonedds/tree/security)
- [ ] Eclipse iceoryx zero-copy built into Eclipse Cyclone DDS for small & fast ROS middleware in collaboration with Bosch
- [ ] testing rmw_cyclonedds serializer contribution by @rotu @roverrobotics with 2.1x faster PointCloud512k, 5.6x faster Struct32k
- [x] implement @irobot use case to find 1 robot among ~1,000 on network via domainTag
We are also submitting various fixes and quality improvements. Will update this list over time.

JaimeMartin, 2019-12-19:
eProsima's contribution for Foxy (to be edited and detailed as we progress):

Recap of What is inside 1.9.3:
- [x] DDS Security (All mandatory plugins)
- [x] DDS Dynamic Types
- [x] Integration Service
- [x] Real-Time Behaviour (Static allocations & non-blocking calls)
- [x] Server Based Discovery
- [x] ROS2 Extended QoS (deadline, livespan, liveliness)
- [x] WAN Support (TCP Transport)
- [x] Intraprocess - Sync and Async publishing

Fast RTPS 1.9.4 (Patch release for eloquent, scheduled mid Jan) 
- [x] Static Memory Allocation - Some last details. Funded and validated by APEX.
- [x] Memory usage improvements
- [x] CPU usage improvements
- [x] Several Bug Fixes

Fast RTPS 1.10 (Patch release for eloquent, Scheduled end of Feb):
- [ ] Shared Memory Transport (Funded and validated by SONY and ROSIN)
- [ ] Better DDS compliance
- [ ] Modern DDS C++ API - Partial

Fast DDS 2.0 (Foxy Release):
- [ ] Modern DDS C++ API - Minimum Profile
- [ ] New DDS Security Plugin: Logging Plugin (Canonical & eProsima joint effort)

Micro-ROS (Planning now) - Coordinating the effort of eProsima, Bosch, PIAP, and Fiware Foundation
- [ ] Foxy Support
- [ ] User configuration API.
- [ ] Parameters support
- [ ] Graph support
- [ ] SOSS micro-ROS SH
- [ ] Events Support.

Integration Service / SOSS
- [x] New Release 
- [x] CI
- [x] Better docs, examples
- [ ] Improved Dynamic Types Support.
- [ ] Transformation library - For extended transformations
- [ ] SH ROS1 Dynamic Types Support.

Karsten1987, 2020-01-16:
Bosch's contribution to Foxy:

**rosbag2**
- [x] make `rosbag2` a metapackage for easy install
- [ ] make `rosbag2` [part of ros-core](https://github.com/ros2/variants/pull/24)
- [ ] record/replay topics on [different QoS](https://github.com/ros2/rosbag2/issues/125)
- [ ] investigate [performance](https://github.com/ros2/rosbag2/issues/123)

**diagnostics**
- [ ] release diagnostics in Foxy
- [ ] contribute [diagnostics aggregator](https://github.com/ros/diagnostics/pull/118)

**iceoryx**
- [ ] finalize iceoryx and rmw_iceoryx to run on ci.ros2.org
- [ ] join efforts to integrate cyclone_dds and iceoryx

dejanpan, 2020-01-16:
Apex.AI contributions to Foxy:

1. Bring ROS 2 and DDS API to parity. 
    1. [ ] take/read of loaned messages 
    1. [ ] waitset in rclcpp 
1. Performance
   1. [ ] user message being copied to/from DDS type and ROS type
   1. [ ] Inefficiency in the waitset implementation
   1. [ ] take(n) => happening above
1. rosbag2  testing
    1. [ ] define metrics
   1. [ ] implement infrastructure
   1. [ ] decouple rclcpp, DDS and file storage
   1. [ ] improve
1. certification of rclcpp and dependencies
    1. Apex.AI is doing an aggressive certification of rclcpp and its dependencies. This includes doing a very thorough static code analysis and line, function, branch and MC/DC code coverage
   1. [ ] bugs will be upstreamed
1. open source containers
    1. [ ] string
    1. [ ] vector
    1. [ ] map
    1. [ ] list
    1. [ ] set
1. [Implement Autonomous Valet Parking demo](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/ASWG-Autonomous-Valet-Parking-hackathon-and-demonstration)
    1. [ ] the deliverable is an ope source stack of algorithms for autonomous driving: https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto

SteveMacenski, 2020-01-17:
Samsung

- [ ] Working with Samsung Russia to implement Navigation costmap filters
  - [ ] Setting speed zones
  - [ ] Setting keep out zones
  - [ ] Setting preferred lanes
- [ ] Improving SLAM Toolbox code quality, documentation, and tutorials for ROS2 as the new default SLAM
- [ ] Working on supporting multi-level and multi-map map distribution in Navigation2
- [ ] Working with collaborators to externally test and validate (fix) Navigation2 stack in efforts to increase long-term reliability

gbiggs, 2020-02-05:
Tier IV's contributions to Foxy:

1. Enhancement of rosidl package for applying real-time system
    - [ ] Apply custom memory allocator in rosidl layer
1. Perform memory audit
    - [ ] Fix remaining [ASan report](https://discourse.ros.org/t/introducing-ros2-sanitizer-report-and-analysis/9287) such as rcl
    - [ ] Add more core packages to sanitizer CI
1. Benchmark/analyses from user's perspective
    - [ ] Benchmark/analyses with model case

ROBOTIS-Will, 2020-02-20:
ROBOTIS

1. [TurtleBot3](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/)
    - [ ] Update TurtleBot3 packages
    - [ ] Update Manipulation packages
    - [ ] Revise Autonomous Driving for ROS 2 Foxy
    - [ ] Support TurtleBot3 binary installation
2. [ros2arduino](https://github.com/ROBOTIS-GIT/ros2arduino)
    - [ ] Update to support ROS 2 Foxy

lokesku, 2020-02-20:
LG

- [ ] OpenEmbedded build support for Foxy on ros/meta-ros

- [ ] Support latest yocto versions for OpenEmbedded ROS2 build

seanyen, 2020-02-20:
Microsoft

* [ ] ROS 2 Windows support for GitHub actions under https://github.com/ros-tooling.

hyang5, 2020-02-26:
Intel

Realsense camera sensor support
- [ ] support of camera model T265 (with tracking) and D435i (with IMU)
- [ ] upgrade to support latest librealsense
- [ ] support reconfigure parameter at runtime

thomas-moulard, 2020-03-04:
# AWS RoboMaker

## Core
 * [x] [Expend RMW API to list publishers/subscribers QoS options](https://github.com/ros2/rmw/pull/186)
 *  [x] CLI: [Enhanced ros2 topic info to display node name, node namespace, topic type and qos profile of the publishers and subscribers](https://github.com/ros2/ros2cli/pull/385)
    <details>
    <summary>New, improved ROS 2 CLI output</summary>

    ```
    $ ros2 topic info rosout
    
    Topic              : rosout
    
    Publisher count    : 2
    
    Node Name: _NODE_NAME_UNKNOWN_
    Node Namespace: _NODE_NAMESPACE_UNKNOWN_
    Topic Type: rcl_interfaces::msg::dds_::Log_
    QoS Profile:
      Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
      Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
      Lifespan: 2147483651294967295 nanoseconds
      Deadline: 2147483651294967295 nanoseconds
      Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
      Liveliness Lease Duration: 2147483651294967295 nanoseconds
    
    Node Name: listener
    Node Namespace: /
    Topic Type: rcl_interfaces::msg::dds_::Log_
    QoS Profile:
      Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
      Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
      Lifespan: 2147483651294967295 nanoseconds
      Deadline: 2147483651294967295 nanoseconds
      Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
      Liveliness Lease Duration: 2147483651294967295 nanoseconds
    
    Subscription count : 0
    ```
    </details>
 * [x] CLI: warn users when one, or more nodes have the same name when invoking [ros2node info](https://github.com/ros2/ros2cli/pull/463), and [ros2node list](https://github.com/ros2/ros2cli/pull/462)
 * [x] ARM HF Tier 2 Support (debug, and fix compilation issues for ARM HF builds)
 * [x] [AWS ROS 2 Oncall](https://github.com/ros-tooling/aws-oncall)

## Tooling
 * [x] Creating, leading, and running the Tooling WG
 * [x] [Cross Compilation Tool](https://github.com/ros-tooling/cross_compile)
   * Compile ROS 1, and ROS 2 workspaces for ARM-HF, and ARM 64.
System Metrics Collector (link)
 * [x] [Lightweight ROS 2 node to aggregate and publish system metrics](https://github.com/ros-tooling/system_metrics_collector)
 * [x] [rosbag2 compression, and data chunking support](https://github.com/ros2/rosbag2)
   * Compress rosbag2 data on the file using zstd.
   * Store rosbag2 data in multiple files to analyze and transfer large data collects faster.
 * [x] GitHub actions for ROS 2 [CI](https://github.com/ros-tooling/action-ros2-ci), and [linting](https://github.com/ros-tooling/action-ros2-lint)
 * [ ] [Warning users when a single launch files contain multiple nodes with similar names](https://github.com/ros2/launch_ros/pull/127)

   <details>
     <summary>Example output</summary>

     ```
     ...
    [INFO] [launch]: Default logging verbosity is set to INFO
    [WARNING] [talker-3]: /talker is not a unique node name
    [WARNING] [listener-4]: /listener is not a unique node name
    [INFO] [talker-1]: process started with pid [18029]
    [INFO] [listener-2]: process started with pid [18030]
    [INFO] [talker-3]: process started with pid [18031]
    [INFO] [listener-4]: process started with pid [18032]
    ... 
    ```
   </details>
 * [ ] [Design, and implementation of a library to compute and publish node topic statistics](https://github.com/ros-tooling/system_metrics_collector)
    * Similar to topic statistics in ROS 1, with slightly different metrics to account for the changes in the communication layer

## Other Efforts

 * [x] [Founding member of the `ros2_control` working group](https://discourse.ros.org/t/announcing-the-ros2-control-working-group/13066/4)
 * [x] [`colcon-bundle` CI migration to GitHub Action, and stabilization](https://github.com/colcon/colcon-bundle)
 * [ ] [REP-2001: minimal C++ ROS 2 variant](https://github.com/ros-infrastructure/rep/pull/231)

## In partnership with Open Robotics

* [ ] [REP-2004: ROS 2 Core Quality Level Definition](https://github.com/ros-infrastructure/rep/pull/218)
  * Community effort to define a set of requirements for production packages.
* [ ] ROS 2 Performance jobs
  * Nightly job running benchmarks on the top of ROS 2 to quickly identify regressions.
* [ ] Cloud-based ROS 2 CI Buildfarm Windows Workers
  * Speed-up ROS 2 CI, and enable it to scale by replacing physical machines by AWS EC2 instances.
* [ ] ROS 2 Rolling Release
  * ROS 2 packages will automatically be released to newer releases, reducing the time required to maintain a ROS 2 package, and ensuring that all existing packages will be available in future ROS 2 distributions!

iluetkeb, 2020-03-13:
Bosch will also contribute timestamp support in the rcl_wait API, as discussed in https://github.com/ros2/design/issues/259. Currently ongoing work is in https://github.com/ros2/rcl/pull/591 and https://github.com/ros2/rmw/pull/200

For full effect, this will require support by the DDS vendors as well.

vmayoral, 2020-03-18:
Here's a bit of what [Alias](https://aliasrobotics.com/) is doing, and plans to finalize for Foxy: 

### ROS 2 Standard Library (see [WIP](https://github.com/ros-infrastructure/rep/pull/)):
- [x] ROS 2 package-subset code testing and reporting ([link](https://github.com/aliasrobotics/RVD/issues?q=is%3Aissue+is%3Aopen+label%3A%22robot+component%3A+ROS2%22+sort%3Acreated-asc))
- [ ] **sros2**
  - triage issues
  - code testing
  - penetration testing
- [ ] **ros2/rmw** (ros2/rmw_implementation, rmw_fastrtps, etc.)
  - code testing
- [ ] **ros2/rcl**
  - code testing
- [ ] **Fast-RTPS**
  - code testing
     - (may add other additional work, pending on funding)
     - (may add other DDS vendors depending on available funding)
- [ ] **ros-planning/navigation2**
  - code testing

### Other efforts:
- [x] maintain the [Robot Vulnerability Database](https://github.com/aliasrobotics/RVD)
  - [ ] Push it towards [inclusion in the Security WG](https://github.com/ros-security/community/issues/6)
- [x] Maintain the Robot Vulnerability Scoring System (RVSS) and push it as a severity scoring mechanism for ROS 2 flaws
  - [ ] Push it towards [inclusion in the Security WG](https://github.com/ros-security/community/issues/5)
- [x] Help steer the [quality/security discussions](
https://github.com/ros-infrastructure/rep/pull/218#discussion_r370583550)
- [x] Help steer the [discussions on the ROS Vulnerability Disclosure Policy](https://docs.google.com/document/d/1qJFlenKvHh5mKVbY-Hc5RK-k_FAJFpiB-rzIfcFqoC0/edit?userstoinvite=victor%40aliasrobotics.com)
- [x] Help steer [another row of “formal” feedback](https://github.com/mikaelarguedas/rep/pull/1#discussion_r393230990) from the Security WG to REP-2004
- [ ] contribute with good devops practices in ROS
- [ ] consider automatic flaw reports, directly in batches (see discussion [here](https://github.com/ros2/ros2/issues/734#issuecomment-557259319)).

closed 2020-06-11 by jacobperron