ros2/ros2 #831: ros2_dashing/ros2-osx/lib/librosidl_typesupport_c.dylib   Reason: image not found

supagas, 2019-11-25:
## Bug report

**Required Info:**

- Operating System:
  - macOS Mojave 10.14.6
- Installation type:
  - 
- Version or commit hash:
  - ros2-dashing-20191018-macos.amd64
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
- / 

#### Steps to reproduce issue
1. . ~/ros2_dashing/ros2-osx/setup.bash
2. ros2 run demo_nodes_cpp talker
after this the error occures

#### Expected behavior
- that the example works/run
#### Actual behavior
- Failed to load entry point 'load': dlopen(/Users/XXX/ros2_dashing/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/XXX/ros2_dashing/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
#### Additional information
tried to trouble shoot with die SIP - tried with enabled and disabled, still same error message
everything is installed via brew
----
## Feature request
none
#### Feature description
none
#### Implementation considerations
none

minggangw, 2019-11-26:
I once met the same error and has reported an issue, please see https://github.com/ros2/ros2/issues/813. The wrong version of poco caused this, because the dashing was built with a different version of poco you are using.

For your case, you have to install [1.9.3](https://bintray.com/package/files/homebrew/bottles/poco) (the latest version is 1.9.4 on brew). This problem is not reproducible on [Eloquent Elusor](https://github.com/ros2/ros2/releases/tag/release-eloquent-20191122), so I suggest you upgrade the ROS2 to Eloquent :smile:

supagas, 2019-11-26:
Thanks for your response ! Downgrade to 1.9.3 worked for me :)
Which not reproducible you mean it works with 1.9.4?

jacobperron, 2019-11-27:
To clarify, if you are using [Dashing](https://index.ros.org/doc/ros2/Releases/Release-Dashing-Diademata/) then it sounds like you should downgrade to poco 1.9.3. If you are using [Eloquent](https://index.ros.org/doc/ros2/Releases/Release-Eloquent-Elusor/), then poco 1.9.4 should be okay.

It sounds like the issue is resolved. If you continue to have issues, feel free to comment and we can reopen this ticket.

closed 2019-11-27 by jacobperron