ros2/ros2 #842: vcs import src < ros2.repos

Furansujin, 2019-12-13:
when I make this command line: 
vcs import src < ros2.repos
I have this error : 
Input data is not valid yaml format: unacceptable character #xdcef: special characters are not allowed
  in "<stdin>", position 0

nuclearsandwich, 2019-12-13:
From the information available my first thought is that your ros2.repos file is either not intact or has errors.

How did you retrieve that repos file?

fujitatomoya, 2019-12-16:
i do not have any problems with https://github.com/ros2/ros2/commit/3b420a7cf4a39c0127eba0b88962bb7f104adbb1

nuclearsandwich, 2019-12-26:
@Furansujin I am going to close this issue until we can get more information. We can re-open it if you're still having trouble and can provide more details.

closed 2019-12-26 by nuclearsandwich

Eddie-717314-Huang, 2022-01-17:
I have same issue: vcs import src < ros2.repos 
stty: stdin isn't a terminal
Video Contact Sheet *NIX v1.13.4, (c) 2007-2019 Toni Corvera
Unable to locate DejaVu Sans font. Falling back to helvetica.
File "import" doesn't exist
File "src" doesn't exist
stty: stdin isn't a terminal

need some help here, thanks

clalancette, 2022-01-17:
It looks like you have another program installed with the name "vcs", which is conflicting with what we expect.  I'll suggest removing that or renaming it, and then following the ROS installation instructions again.

In the future, please ask questions like this on https://answers.ros.org, which is our central Question-and-Answer site.