ros2/ros2 #844: Cannot load rmw_fastrtps_cpp.dll

dzywater, 2019-12-25:
ROS2 Version
ros2-dashing-20191213-windows-amd64.zip

OS details
Edition: Windows 10 Enterprise
Version: 1909
OS Build: 18363.476

[ERROR] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_cpp'), with error message 'failed to load shared library of rmw implementation. Exception: Cannot load library: C:\ros2\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\ci_packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:137', exiting with 1.

dzywater, 2019-12-25:
I tried opensplice, it ran successfully when cmd: ros2 run demo_nodes_cpp talker.

1.Download PXXX-VortexOpenSplice-6.9.190925OSS-HDE-x86_64.win-vs2019-installer.zip
2.unzip into C:\openslice
3.let OSPL_HOME=C:\opensplice
4.let RMW_IMPLEMENTATION=rmw_opensplice_cpp

richiware, 2019-12-26:
I was able to reproduce your issue. Windows binaries package comes with FastRTPS compiled with security (OpenSSL). But the OpenSSL libraries are not included in the package. [Here](https://slproweb.com/download/Win64OpenSSL_Light-1_0_2u.exe) you have the light installer of OpenSSL. After the installation and pointing PATH environment variable to include OpenSSL libraries, it should start working.

dzywater, 2019-12-26:
@richiware 
I tested the Win64OpenSSL_Light-1_0_2u.exe above. It works. I found I installed Win64OpenSSL-1_1_0L.exe before. So the issue is from the wrong openssl version. I also downloaded Win64OpenSSL-1_0_2u.exe. It works. The ros2 document says "Download an OpenSSL installer from this page. Scroll to the bottom of the page and download Win64 OpenSSL v1.0.2. Don’t download the Win32 or Light versions." But the Win64OpenSSL_Light-1_0_2u.exe seems OK.

Whatever. Thanks.

adriennewinter, 2022-04-28:
I was struggling with this issue too and re-installing OpenSSL from the link provided in the ROS installation guide and re-doing the Path variable worked for me.