ros2/ros2 #845: foonathan_memory_vendor is causing colcon build to fail

sudo-panda, 2019-12-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - master with install instructions for foxy
- DDS implementation:
  - Fast RTPS

#### Steps to reproduce issue
Follow the steps in [Building ROS 2 on Linux](https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/#linux-latest) but instead of 
```wget https://raw.githubusercontent.com/ros2/ros2/foxy/ros2.repos``` 
in ***Get ROS2 code*** step use:
```wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos```
(I used this because foxy branch hasn't been created yet)

#### Expected behavior
colcon builds successfully

#### Actual behavior
Getting error in building footnathan_memory_vendor. The stderr output is provided [here](https://gist.github.com/sudo-panda/c851dae55650c7b56722f2550a4c19fb) to prevent this description from getting too long.

The first error that appears is that `‘abort’ is not a member of ‘std’` but there might be other problems too.

richiware, 2020-01-07:
I was not able to reproduce your issue. I've used a `ubuntu:latest` docker image following your instructions. Also, I've checked using default GCC and default Clang, and `foonathan_memory_vendor` is built successfully. Can you provide all stdout_stderr log of `foonathan_memory_vendor` package?

sudo-panda, 2020-01-07:
I have uploaded it as a gist you can find it [here](https://gist.github.com/sudo-panda/ab92e2a3ddae98b998512803de3f21ec).

richiware, 2020-01-08:
I need the log traces when CMake configures the project, when CMake searchs the compiler to be used and those things. I don't see those log traces in your gist.

sudo-panda, 2020-01-10:
The error disappeared after cleaning `log`, `install` and `build` folders and running `colcon build` again

closed 2020-01-10 by sudo-panda

cardboardcode, 2020-07-15:
```
sudo rm -r build/ log/ install/
colcon build
```

Can confirm that this works for me as well. Weird... but it works. :/

Bug-terminator, 2021-07-24:
> I need the log traces when CMake configures the project, when CMake searchs the compiler to be used and those things. I don't see those log traces in your gist.

Hi, I encounterd the same bug, here is the log:
```shell
--- stderr: pybind11_vendor
* Closing connection 0
* Closing connection 1
---
Finished <<< pybind11_vendor [7.93s]
--- stderr: foonathan_memory_vendor
Cloning into 'foo_mem-ext'...
HEAD is now at c619113 Whitespace clean up (#70)
Submodule 'cmake/comp' (https://github.com/foonathan/compatibility.git) registered for path 'cmake/comp'
Cloning into '/Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext/cmake/comp'...
CMake Error at cmake/comp/comp_base.cmake:288 (message):
  error downloading feature file cpp11_lang/alignas.cmake: "Couldn't connect
  to server".  Check spelling of feature.

    Trying ::...

  TCP_NODELAY set

  Connection failed

  connect to :: port 443 failed: Connection refused

    Trying 0.0.0.0...

  TCP_NODELAY set

  Connection failed

  connect to 0.0.0.0 port 443 failed: Connection refused

  Failed to connect to raw.githubusercontent.com port 443: Connection refused

  Closing connection 0

Call Stack (most recent call first):
  cmake/comp/comp_base.cmake:347 (_comp_fetch_feature)
  cmake/comp/comp_base.cmake:369 (_comp_handle_feature)
  cmake/compatibility.cmake:13 (comp_target_features)
  CMakeLists.txt:50 (include)

make[2]: *** [foo_mem-ext-prefix/src/foo_mem-ext-stamp/foo_mem-ext-configure] Error 1
make[1]: *** [CMakeFiles/foo_mem-ext.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< foonathan_memory_vendor [11.0s, exited with code 2]
```
What's the problem?

christophebedard, 2021-07-24:
It's failing to download a file from GitHub. It's a network/proxy problem and/or the IP is blocked.

leeyunhome, 2022-06-16:
Hello, @richiware @christophebedard  @sudo-panda @Bug-terminator 

Thank you for your good reply.

I ran into the following error while building ros2 from source on nvidia jetson nano (ubuntu18.04).
It seems similar to this issue, can you tell me how to solve it?

`Starting >>> ament_cmake_xmllint                                    
Finished <<< ament_cmake_xmllint [5.27s]      
Starting >>> foonathan_memory_vendor                               
--- stderr: foonathan_memory_vendor           
Cloning into 'foo_mem-ext'...
Note: checking out 'v0.7-1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 19ab075 Release 0.7-1
CMake Error at CMakeLists.txt:7 (cmake_minimum_required):
  CMake 3.11 or higher is required.  You are running version 3.10.2

make[2]: *** [foo_mem-ext-prefix/src/foo_mem-ext-stamp/foo_mem-ext-configure] Error 1
make[1]: *** [CMakeFiles/foo_mem-ext.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< foonathan_memory_vendor [25.8s, exited with code 2]
Aborted  <<< uncrustify_vendor [1min 21s]                                      
                                         
Summary: 38 packages finished [2min 50s]
  1 package failed: foonathan_memory_vendor
  1 package aborted: uncrustify_vendor
  1 package had stderr output: foonathan_memory_vendor
  252 packages not processed
`

Thank you.

EduPonz, 2022-06-16:
Hi @leeyunhome ,

I'm guessing that you are trying to build ROS 2 Humble or Rolling on Ubuntu 18.04. Be aware that the last ROS 2 distribution that officially supported 18.04 was Eloquent. From there on, the supported Ubuntu has been 20.04, up until Humble, which supports 22.04 (see [REP 2000](https://ros.org/reps/rep-2000.html#platforms-by-distribution)).

That being said, as the error points out, upgrading CMake should be enough to get over this particular bump (which does not mean that you will not encounter others along the way).

zhangyi1357, 2024-11-08:
> It's failing to download a file from GitHub. It's a network/proxy problem and/or the IP is blocked. Delete your build/log/install directories and try again.
> 
> See also [foonathan/memory#52](https://github.com/foonathan/memory/issues/52)

Same in 2024

zhangyi1357, 2024-11-08:
> ```
> sudo rm -r build/ log/ install/
> colcon build
> ```
> 
> Can confirm that this works for me as well. Weird... but it works. :/

Same in 2024.