ros2/ros2 #849: openssl error

purewater0901, 2020-01-05:
## Bug report

**Required Info:**

- Operating System:
  -  Mac OS Catalina 10.15.2(maybe latest)
- Installation type:
  -  binaries
- Version or commit hash:
  -  ROS 2 Eloquent Elusor from ros2.repos
- DDS implementation:
  -  Default
- Client library (if applicable):
  -  rclcpp

#### Steps to reproduce issue

install ros2 on the procedure in wiki
```
ros2 run demo_nodes_cpp talker
```

#### Expected behavior
```
[INFO] [talker]: Publishing: 'Hello World: 1'
[INFO] [talker]: Publishing: 'Hello World: 2'
[INFO] [talker]: Publishing: 'Hello World: 3'
```

#### Actual behavior
A lot of error log that indicate I should install openssl 1.0.

#### Additional information
Homebrew already did not support openssl 1.0, I think. There are only openssl@a1.1. 
I think we should use openssl@1.1.

dirk-thomas, 2020-01-16:
`Catalina` is not a supported platform for ROS 2 Eloquent (see https://www.ros.org/reps/rep-2000.html#eloquent-elusor-november-2019-november-2020). In order to try using ROS 2 on Catalina you can try to build ROS 2 from source (not guarantees that it will work though).

closed 2020-01-16 by dirk-thomas

trifasico, 2020-04-16:
Hi, i have a Mac with Catalina, have you any of you sucsesfully installed Ros2 on Catalina? if yes, could you help me? i have been trying but i still could make it work