ros2/ros2 #850: "colcon test" giving warning

iamrajee, 2020-01-09:
WARNING: Testing via this command is deprecated and will be removed in a future version. Users looking for a generic test entry point independent of test runner are encouraged to use tox.
#ros2 #dashingDiademata

dirk-thomas, 2020-01-09:
Please provide more context what your workspace contains, for what package you see the warning, what platform you are on, etc.

iamrajee, 2020-01-10:
**I get stderr for below packages:** 

examples_rclpy_minimal_action_server
examples_rclpy_minimal_action_client
examples_rclpy_minimal_subscriber
examples_rclpy_minimal_service
examples_rclpy_minimal_publisher
examples_rclpy_minimal_client
examples_rclpy_executors

**Workspace is /ros2_example_ws which is create by following [this tutorial](https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/)
Thus /ros2_example_ws/src contain folders as**
build  examples   install  log

**I'm using Ubuntu 18.04**

Do you need any other information?

dirk-thomas, 2020-01-10:
I can't reproduce the error messages yet. Can you please provide the following information for one of the packages:
* the information of the package as seen by colcon:
  `colcon info --packages-select examples_rclpy_executors`
* the actually invoked commands to test the package:
  `cat log/latest_test/examples_rclpy_executors/command.log`
* for completeness the full output from the tested package:
  `cat log/latest_test/examples_rclpy_executors/stdout_stderr.log`

Arnatious, 2020-01-29:
@dirk-thomas Can confirm this error appeared for me when working on nodl_python

```
ubuntu@ros2-live:~/workspace$ colcon info --packages-select nodl_python
path: /home/ubuntu/workspace/src/nodl/nodl_python
  type: ros.ament_python
  name: nodl_python
  dependencies:
    build: ament_index_python
    run: ament_index_python
    test: ament_flake8 ament_lint_auto ament_lint_common ament_mypy python3-pytest python3-pytest-mock
  metadata:
    get_python_setup_options: <function RosPackageIdentification.identify.<locals>.getter at 0x7ff8fd39dd90>
    version: 0.0.
```

```
ubuntu@ros2-live:~/workspace$ cat log/latest_test/nodl_python/command.log 
[0.355s] Invoking command in '/home/ubuntu/workspace/src/nodl/nodl_python': AMENT_PREFIX_PATH=/home/ubuntu/workspace/install/nodl_python:${AMENT_PREFIX_PATH} PYTEST_ADDOPTS=--tb=short --junit-xml=/home/ubuntu/workspace/build/nodl_python/pytest.xml --junit-prefix=nodl_python -o cache_dir=/home/ubuntu/workspace/build/nodl_python/.pytest_cache PYTHONPATH=/home/ubuntu/workspace/install/nodl_python/lib/python3.6/site-packages:${PYTHONPATH} /usr/bin/python3 setup.py pytest egg_info --egg-base /home/ubuntu/workspace/build/nodl_python
[1.906s] Invoked command in '/home/ubuntu/workspace/src/nodl/nodl_python' returned '1': AMENT_PREFIX_PATH=/home/ubuntu/workspace/install/nodl_python:${AMENT_PREFIX_PATH} PYTEST_ADDOPTS=--tb=short --junit-xml=/home/ubuntu/workspace/build/nodl_python/pytest.xml --junit-prefix=nodl_python -o cache_dir=/home/ubuntu/workspace/build/nodl_python/.pytest_cache PYTHONPATH=/home/ubuntu/workspace/install/nodl_python/lib/python3.6/site-packages:${PYTHONPATH} /usr/bin/python3 setup.py pytest egg_info --egg-base /home/ubuntu/workspace/build/nodl_python
```

```
ubuntu@ros2-live:~/workspace$ cat log/latest_test/nodl_python/stdout_stderr.log 
[0.646s] running pytest
[0.801s] WARNING: Testing via this command is deprecated and will be removed in a future version. Users looking for a generic test entry point independent of test runner are encouraged to use tox.
[0.855s] running egg_info
[0.855s] writing /home/ubuntu/workspace/build/nodl_python/nodl_python.egg-info/PKG-INFO
[0.855s] writing dependency_links to /home/ubuntu/workspace/build/nodl_python/nodl_python.egg-info/dependency_links.txt
[0.856s] writing requirements to /home/ubuntu/workspace/build/nodl_python/nodl_python.egg-info/requires.txt
[0.856s] writing top-level names to /home/ubuntu/workspace/build/nodl_python/nodl_python.egg-info/top_level.txt
[0.857s] reading manifest file '/home/ubuntu/workspace/build/nodl_python/nodl_python.egg-info/SOURCES.txt'
[0.858s] writing manifest file '/home/ubuntu/workspace/build/nodl_python/nodl_python.egg-info/SOURCES.txt'
[0.860s] running build_ext
[1.126s] ============================= test session starts ==============================
[1.127s] platform linux -- Python 3.6.9, pytest-5.3.4, py-1.8.1, pluggy-0.13.1
[1.132s] cachedir: /home/ubuntu/workspace/build/nodl_python/.pytest_cache
[1.132s] rootdir: /home/ubuntu/workspace/src/nodl/nodl_python, inifile: pytest.ini
[1.132s] plugins: ament-xmllint-0.8.1, ament-pep257-0.8.1, ament-mypy-0.8.1, ament-flake8-0.8.1, ament-copyright-0.8.1, ament-lint-0.8.1, mock-2.0.0, cov-2.5.1
[1.150s] collecting ... 
[1.150s] collected 1 item                                                               
[1.151s] 
[1.792s] test/test_mypy.py F                                                      [100%]
[1.792s] 
[1.793s] =================================== FAILURES ===================================
[1.793s] __________________________________ test_mypy ___________________________________
[1.793s] test/test_mypy.py:11: in test_mypy
[1.793s]     assert rc == 0, 'Found code style errors / warnings'
[1.793s] E   AssertionError: Found code style errors / warnings
[1.793s] E   assert 1 == 0
[1.793s] ----------------------------- Captured stdout call -----------------------------
[1.794s] 
[1.794s] 6 files checked
[1.794s] 19 errors
[1.794s] test/wip_test_nodl_parse.py: note: In function "valid_nodl":
[1.794s] test/wip_test_nodl_parse.py:11:24: error: Argument 1 to "parse" has incompatible type "Path"; expected "Union[str, bytes, int, IO[Any]]"
[1.794s] nodl/nodl_parse.py: note: In function "parse_action":
[1.794s] nodl/nodl_parse.py:74:12: error: Too many positional arguments for "NoDLAction"
[1.794s] nodl/nodl_parse.py: note: In function "parse_parameter":
[1.794s] nodl/nodl_parse.py:79:12: error: Too many positional arguments for "NoDLParameter"
[1.794s] nodl/nodl_parse.py: note: In function "parse_service":
[1.794s] nodl/nodl_parse.py:90:12: error: Too many positional arguments for "NoDLService"
[1.794s] nodl/nodl_parse.py:90:51: error: Argument 3 to "NoDLService" has incompatible type "Optional[bool]"; expected "bool"
[1.795s] nodl/nodl_parse.py:90:59: error: Argument 4 to "NoDLService" has incompatible type "Optional[bool]"; expected "bool"
[1.795s] nodl/nodl_parse.py: note: In function "parse_topic":
[1.795s] nodl/nodl_parse.py:101:12: error: Too many positional arguments for "NoDLTopic"
[1.795s] nodl/nodl_parse.py:101:49: error: Argument 3 to "NoDLTopic" has incompatible type "Optional[bool]"; expected "bool"
[1.795s] nodl/nodl_parse.py:101:60: error: Argument 4 to "NoDLTopic" has incompatible type "Optional[bool]"; expected "bool"
[1.795s] nodl/nodl_parse.py: note: In function "parse_nodes":
[1.795s] nodl/nodl_parse.py:124:12: error: Incompatible return value type (got "List[Element]", expected "List[NoDLNode]")
[... cut ...]
[1.795s] nodl/nodl_parse.py:137:31: error: Incompatible types in assignment (expression has type "Union[str, Element]", variable has type "Element")
[1.795s] Found 12 errors in 2 files (checked 6 source files)
[1.796s] 
[1.796s] 
[1.796s] Checked files:
[1.796s] 
[1.796s] * setup.py
[1.796s] * nodl/__init__.py
[1.796s] * nodl/nodl_parse.py
[1.796s] * nodl/nodl_types.py
[1.796s] * test/test_mypy.py
[1.796s] * test/wip_test_nodl_parse.py
[1.796s] --- generated xml file: /home/ubuntu/workspace/build/nodl_python/pytest.xml ----
[1.797s] ============================== 1 failed in 0.67s ===============================
```

Arnatious, 2020-01-29:
Looking at https://github.com/pzahemszky/pZudoku/issues/97 it appears the issue is with running setup.py pytest

Reproduced by running
```
ubuntu@ros2-live:~/workspace/src/nodl/nodl_python$ python3 setup.py pytest
running pytest
WARNING: Testing via this command is deprecated and will be removed in a future version. Users looking for a generic test entry point independent of test runner are encouraged to use tox.
running egg_info
creating nodl_python.egg-info
writing nodl_python.egg-info/PKG-INFO
writing dependency_links to nodl_python.egg-info/dependency_links.txt
writing requirements to nodl_python.egg-info/requires.txt
writing top-level names to nodl_python.egg-info/top_level.txt
writing manifest file 'nodl_python.egg-info/SOURCES.txt'
reading manifest file 'nodl_python.egg-info/SOURCES.txt'
writing manifest file 'nodl_python.egg-info/SOURCES.txt'
running build_ext
...
...
```

Arnatious, 2020-01-29:
Note: appears with pytest 5.3.4 installed from pypi

dirk-thomas, 2020-01-29:
The warning was introduced in `setuptools` 41.5.0 (see pypa/setuptools#1878). See colcon/colcon-core#296.

dirk-thomas, 2020-01-29:
Closing in favor of colcon/colcon-core#296.

closed 2020-01-29 by dirk-thomas