ros2/ros2 #852: No module named 'rclpy._rclpy' on running "ros2 run rqt_console rqt_console"

iamrajee, 2020-01-10:
**On running command "ros2 run rqt_console rqt_console" I got below error message:**

Traceback (most recent call last):
  File "/opt/ros/dashing/lib/rqt_console/rqt_console", line 8, in <module>
    sys.exit(main.main(sys.argv, standalone='rqt_console.console.Console'))
  File "/opt/ros/dashing/lib/python3.6/site-packages/rqt_gui/main.py", line 55, in main
    argv = rclpy.utilities.remove_ros_args(args=argv)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/utilities.py", line 41, in remove_ros_args
    from rclpy.impl.implementation_singleton import rclpy_implementation
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/impl/implementation_singleton.py", line 31, in <module>
    rclpy_implementation = _import('._rclpy')
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/impl/__init__.py", line 21, in _import
    return importlib.import_module(name, package='rclpy')
  File "/home/rajendra/miniconda3/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'rclpy._rclpy'
~/ros2_ws_moveit(master ?:1 ✗) ros2 run rqt_console rqt_console
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/rqt_console/rqt_console", line 8, in <module>
    sys.exit(main.main(sys.argv, standalone='rqt_console.console.Console'))
  File "/opt/ros/dashing/lib/python3.6/site-packages/rqt_gui/main.py", line 55, in main
    argv = rclpy.utilities.remove_ros_args(args=argv)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/utilities.py", line 41, in remove_ros_args
    from rclpy.impl.implementation_singleton import rclpy_implementation
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/impl/implementation_singleton.py", line 31, in <module>
    rclpy_implementation = _import('._rclpy')
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/impl/__init__.py", line 21, in _import
    return importlib.import_module(name, package='rclpy')
  File "/home/rajendra/miniconda3/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'rclpy._rclpy'

===== **Other Details** =====

**SYSTEM** = Ubuntu 18.04
**LD_LIBRARY_PATH**=/opt/ros/dashing/opt/yaml_cpp_vendor/lib:/opt/ros/dashing/opt/rviz_ogre_vendor/lib:/usr/lib:/opt/rti.com/rti_connext_dds-5.3.1/lib/x64Linux3gcc5.4.0:/opt/ros/dashing/lib
**AMENT_PREFIX_PATH**=/opt/ros/dashing
**OSPL_URI**=file:///opt/ros/dashing/share/opensplice_cmake_module/config/ros_ospl.xml
**PWD**=/home/rajendra/ros2_ws_moveit
**ROS_DOMAIN_ID**=27
**ROS_VERSION**=2
**ROS_PYTHON_VERSION**=3
**PYTHONPATH**=/opt/ros/dashing/lib/python3.6/site-packages
**ROS_DISTRO**=dashing

dirk-thomas, 2020-01-10:
The traceback shows you are using `conda` as well as a custom Python 3.7 version. The ROS Debian packages are built against the default Python version which is 3.6 on Ubuntu Bionic.

Please try the same without using `conda` or a custom Python version.

tomeeeS, 2020-04-29:
my Bionic says it has 2.7.7 of python.
could I create a conda environment for ROS with python 3.6 and use that? would that work?

dirk-thomas, 2020-04-29:
Please see my above comment and use the Python 3 version shopped with Bionic.

Conda is not officially supported. You can try it but might run into issues which you should be comfortable to debug in your own.

Since there was no response on the original feedback I will close this ticket for now.

closed 2020-04-29 by dirk-thomas