ros2/ros2 #855: $COLCON_PREFIX_PATH not showing installation path

suab321321, 2020-01-17:
`source /opt/ros/dashing/setup.bash` is not loading the `env` varible `COLCON_PREFIX_PATH` with its installation path

**Required Info:**

- Operating System:
  Ubuntu 18.04 
- Installation type:
  source
- Version or commit hash:
  `dashing`
- DDS implementation:
  `default`

dirk-thomas, 2020-01-17:
The Debian packages are not built with `colcon` and therefore they don't have that environment variable set.

Why are you expecting it / what are you needing it to be set for?

suab321321, 2020-01-18:
when building rclnodejs the env variable COLCON_PREFIX_PATH needs to be setup for its full installation using npm install.
In crystal version sourcing the setup.bash loaded the env variable COLCON_PREFIX_PATH but in dashing version it is not happening.

dirk-thomas, 2020-01-18:
`rclnodejs` should simply not rely on the presence of the `COLCON_PREFIX_PATH` environment variable. In the case of Debian packages `colcon` is not involved hence this variable doesn't exist. Instead the package could check the `AMENT_PREFIX_PATH` which is defined in both cases - from binary as well as from source.

Since this is not in problem in ROS 2 but only in the `rclnodejs` package I will go ahead and close this ticket. Please fill a ticket in that repository instead (and cross reference this ticket for future readers to find the new follow up ticket).

closed 2020-01-18 by dirk-thomas

suab321321, 2020-01-18:
@dirk-thomas but sir rclnodejs uses COLCON_PREFIX_PATH but in crystal version COLCON_PREFIX_PATH was getting loaded successfully.o
One version is able to do it and other is not.Is this not problem in dashing version?

dirk-thomas, 2020-01-18:
While it was set in Crystal it was never meant to be used by package and as of Dashing it is not available anymore.

suab321321, 2020-01-18:
okay @dirk-thomas :)

minggangw, 2020-01-19:
Hi @suab321321 the rclnodejs has adoptedd `AMENT_PREFIX_PATH` since v0.10.2 (currently v0.11.1) please see https://github.com/RobotWebTools/rclnodejs/commit/e40a6631f030af64324e1a6e6d2f00c1de7cdc1c, so would you please verify it with the latest version?

suab321321, 2020-01-19:
@minggangw develop branch of rclnodejs is using COLCON_PREFIX_PATH for installing it.
see this
[image]

minggangw, 2020-01-19:
It seems that you are running rclnodejs@0.10.0

suab321321, 2020-01-19:
I m runnig develop branch of rclnodejs wasnt that suppose to be latest @minggangw ?

minggangw, 2020-01-19:
Please reference https://github.com/RobotWebTools/rclnodejs#match-with-ros-20-stable-releases

suab321321, 2020-01-19:
@minggangw yes its working now AMENT_PREFIX_PATH is working for 0.10.3.Thank you for your help.