ros2/ros2 #856: replace custom get env logic in various places with rcutils_get_env()

dirk-thomas, 2020-01-17:


Suyash458, 2020-01-30:
@dirk-thomas I'd like to pick this up. Is it still available?

dirk-thomas, 2020-02-03:
@Suyash458 Sure, please feel free to create PRs for packages which currently have their own custom `getenv()` logic.

mbuijs, 2020-09-27:
Using a fresh ros2 master checkout, I searched for `getenv` and `get_env` to see if there are any occurences left to change to rcutils or rcpputils, but I could not find any. Maybe this ticket can be closed?

fujitatomoya, 2020-09-28:
@dirk-thomas 

could you take a look at https://github.com/ros2/rcpputils/pull/99?

and i think this does not require the following `test code`, my understanding is correct? if not, I can replace the following part into `rcutils_get_env`.

- https://github.com/ros2/performance_test_fixture/blob/4790a64ddd38f36f6907dcb85f4be85ae1a78a2d/src/performance_test_fixture.cpp#L28
- https://github.com/ros2/rviz/blob/3f6108cbf071956bd57d8df946df97bb4bbd2203/rviz_visual_testing_framework/src/internal/visual_test.cpp#L197-L201

dirk-thomas, 2020-09-28:
> Using a fresh ros2 master checkout, I searched for getenv and get_env to see if there are any occurences left to change to rcutils or rcpputils, but I could not find any.

Please search again. I get multiple matches for `getenv(`:

```
ament/ament_index/ament_index_cpp/src/get_search_paths.cpp:
  40:   ament_prefix_path = getenv(env_var);

ros-perception/image_common/camera_info_manager/tests/unit_test.cpp:
  625:   char * home_env = getenv("HOME");

ros2/performance_test_fixture/src/performance_test_fixture.cpp:
  28:   const char * performance_test_fixture_enable_trace = getenv(

ros2/rcpputils/include/rcpputils/filesystem_helper.hpp:
  454:   const char * temp_path = getenv("TMPDIR");
```

Not sure about the `std::getenv(` matches:

```
osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/memory_tools/verbosity.cpp:
  33:   const char * value = std::getenv("MEMORY_TOOLS_VERBOSITY");

osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/memory_tools/vendor/bombela/backward-cpp/backward.hpp:
  3522:     const char* prefixes_str = std::getenv("BACKWARD_CXX_SOURCE_PREFIXES");

osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/test_runner/get_environment_variable.hpp:
  38:   env_value = std::getenv(env_var_name.c_str());

ros2/rviz/rviz_visual_testing_framework/src/internal/visual_test.cpp:
  197:   if (!std::getenv("GenerateReferenceImages")) {
  201:   std::string generate_references = std::getenv("GenerateReferenceImages");
```

fujitatomoya, 2020-09-29:
@dirk-thomas 

okay, so we just simply replace all of the `getenv` into `rcutils_get_env` including test code. i'll fix them.

fujitatomoya, 2020-10-08:
following is the summary, and I would like to confirm that we can create `rcutils` dependency for each one of them? (I think that would be less portability.) I am not so convinced to do that.

| Package | rcutils dependent | Note |
|:---|:---|:---|
| osrf_testing_tools_cpp | No | **We do really want to create dependency on rcutils just to replace `std:getenv` into `rcutils_get_env`?** |
| rcpputils | Yes | https://github.com/ros2/rcpputils/pull/99 |
| urdfdom | No | test code only, same with `osrf_testing_tools_cpp` but `stdlib.h` |
| rviz_visual_testing_framework | No | test code only, same with `osrf_testing_tools_cpp` |
| performance_test_fixture | No | test code only, same with `osrf_testing_tools_cpp` |
| ament_index_cpp | No | same with `osrf_testing_tools_cpp` |
| gtest_vendor | No | same with `osrf_testing_tools_cpp` but `stdlib.h` |

clalancette, 2020-10-09:
@fujitatomoya Thanks for doing that overview!

> osrf_testing_tools_cpp

No, we don't want to add an rcutils dependency there; it's supposed to be independent of ROS 2.

> urdfdom

This is a pure cmake package, and doesn't really have any ROS 2 dependencies.  So I think we don't want to add one there either.

> rviz_visual_testing_framework

This one already has a dependency on rcutils via its dependencies, so I see no problem switching this one to use `rcutils_get_env`.

> performance_test_fixture

This one doesn't really depend on any ROS 2 stuff, so we should not add a dependency to rcutils either.

> ament_index_cpp

Same.

> gtest_vendor

Same.

So I think that leaves us with ros2/rcpputils#99, and a change to the rviz_visual_testing_framework to use `rcutils_get_env`.  The rest we should ignore.

fujitatomoya, 2020-10-10:
@clalancette 

thanks for the clarification, i will work on rviz_visual_testing_framework then.

fujitatomoya, 2020-10-10:
@clalancette @dirk-thomas 

the following is summary on this issue,

| Package | rcutils dependent | Note |
|:---|:---|:---|
| osrf_testing_tools_cpp | No | skip, see https://github.com/ros2/ros2/issues/856#issuecomment-706343834 |
| rcpputils | Yes | https://github.com/ros2/rcpputils/pull/99 |
| urdfdom | No | skip, see https://github.com/ros2/ros2/issues/856#issuecomment-706343834 |
| rviz_visual_testing_framework | Yes | https://github.com/ros2/rviz/pull/609 |
| performance_test_fixture | No | skip, see https://github.com/ros2/ros2/issues/856#issuecomment-706343834 |
| ament_index_cpp | No | skip, see https://github.com/ros2/ros2/issues/856#issuecomment-706343834 |
| gtest_vendor | No | skip, see https://github.com/ros2/ros2/issues/856#issuecomment-706343834 |

closed 2020-10-13 by clalancette