ros2/ros2 #857: Port topic_tools to ROS2

emersonknapp, 2020-01-22:
## Feature request

Port `topic_tools` to ROS2 http://wiki.ros.org/topic_tools

#### Feature description
The `topic_tools` package provided a variety of useful utilities for manipulating topics:
* mux: multiplex between multiple topics.
* relay: republish data on one topic to another.
* relay_field: allow to republish data in a different message type
* drop: relay a topic, dropping X out of every Y message.
* throttle: relay a topic, but limit republishing to a maximum bandwidth or rate.
* transform: transform a topic (or topic field) using a Python expression and publishes the result on another topic.

These are likely to be quite useful to ros2 users as well.

#### Implementation considerations
Most of the functionality is likely straightforward, except for adaptive QoS, which can use the new `get_topic_info` RMW APIs being currently merged (same as the problem for rosbag2). However, the nodes could take in input and output QoS explicitly, as a early workaround if necessary.

mjcarroll, 2020-01-22:
These would also serve well as components in addition to standalone nodes, and was a motivating case for components.

mjcarroll, 2020-01-22:
Is this something that you are interested in doing @emersonknapp ?

wjwwood, 2020-01-22:
Actually, I think it's not that straightforward. You need to work with types that are not know at compile time.

`rosbag2` does something like this, but there are limitations, see:

https://github.com/ros2/rosbag2/blob/c34601acba77d08c22211d3eb062b8ba867fef71/rosbag2_transport/src/rosbag2_transport/generic_subscription.hpp#L28-L34

It only works if the type support for the type and matching rmw implementation is built locally and accessible to the current process (via the ament index I think). I think it opens the type support library with dlopen to access the rmw message type support structure it needs.

The "shape shifter" in ROS 1 (which is what topic tools used) worked differently, allowing you to forward all the settings and type information from the publish to the subscriber along with the serialized data without ever having to interpret it. I don't think this pattern will work for us, given the middleware layer. I think it might be possible to do something like this with DDS X-types but I'm not sure it will be the best solution.

I've thought about making each node offer a service (or maybe a latched topic) which lets remote nodes access the message definition for any type that node is using, as a string (maybe .msg or .idl, or maybe as something like JSON). That way you could always ask the remote node what it thinks the type definition is and you could dynamically interpret it (sort of like what plot juggler uses in ROS 1, see: https://wiki.ros.org/ros_type_introspection). This also allows you to do clever things when the remote publisher is using a different version of the same message type (i.e. the name matches but the checksum for the definition is different).

emersonknapp, 2020-01-22:
Thanks for the notes, @wjwwood, that is really helpful context. @mjcarroll I'm running it by our team, we are focusing on perceived feature gaps in migrating people to ROS2 so this fits the charter, but I'm not sure how high we will want to prioritize it yet.

nyxaria, 2020-08-17:
Is there any progress on this? I am currently trying to create a generic subscriber in ROS2 - should I go down the route that rosbag2 went?

emersonknapp, 2020-08-18:
I think that the GenericSubscription and GenericPublisher from rosbag2 would be good utilities to expose generally in, say, rclcpp. That could then be used for a topic_tools port. I don't think it has the drawbacks @wjwwood mentioned because it only deals with `SerializedMessage`s and never has to introspect the contents. This should work well for every topic_tool listed except `transform`

nyxaria, 2020-08-25:
@emersonknapp is there a way to go from a SerializedMessage to a JSON which has key-value pairs of the data fields inside it?

emersonknapp, 2020-08-26:
> @emersonknapp is there a way to go from a SerializedMessage to a JSON which has key-value pairs of the data fields inside it?

Not that I know of, at least in C++. There is now a way to deserialize the message (into a `void *`, which would need casting). In Python, from my understanding, we can do introspection of the fields and their types at runtime via the `get_fields_and_field_types` classmethod for msg types.

dejanpan, 2020-11-05:
@emersonknapp @wjwwood @mjcarroll we would actually need topic_tools in this ROS 2 project https://github.com/tier4/Pilot.Auto/issues/82 and we have a fantastic developer in @nnmm to do the port. Would you guys be able to assist him for deep technical questions?

dejanpan, 2020-11-05:
Also - would one of you be able to answer this question from https://github.com/tier4/Pilot.Auto/issues/82#issuecomment-722475817:
> where should the code live? 

Should we just start with the standalone package `topic_tools` in apexai github organization and you meanwhile find the place for it in https://github.com/ros2/?

wjwwood, 2020-11-05:
> where should the code live?

> > Should we just start with the standalone package topic_tools in apexai github organization and you meanwhile find the place for it in https://github.com/ros2/?

We can always move it later. Sometimes we make it a branch on the ROS 1 repository. It's really up to the maintainers. In this case since it was part of `ros_comm` before, I'd say make a new repository and we'll move it to the `ros2` org. Just point me to it, and I will move it.

dejanpan, 2020-11-05:
@nnmm  ^^^

emersonknapp, 2020-11-05:
Glad to see that there's bandwidth to work on this! I haven't been able to get any resources from our side. I'd be happy to help with code reviews or design, feel free to ping me.

nnmm, 2020-11-09:
@wjwwood Could you or someone else clarify why "It only works if the type support for the type and matching rmw implementation is built locally and accessible to the current process (via the ament index I think)." – that is in contrast to packages installed with rosdep/apt, correct? However, I checked a few of the `ros-foxy-*-msg` packages in the repo, and they all come with type support libs and ament index, so I don't understand why they wouldn't work.

cc @dejanpan who might also know.

nnmm, 2020-11-09:
A few further comments/questions:
* I've implemented a `relay` prototype based on GenericSubscription/GenericPublisher, assuming that is the best way forward. Although I'd naively think that it might be possible to not use any type information at all and do it all on a lower level – theoretically no (de)serialization is needed, just forwarding a binary blob, right?
* I'd remove the GenericSubscription/GenericPublisher parts from `rosbag2transport` and make it depend on the new package if that sounds good. I'd assume putting GenericSubscription/GenericPublisher into `rclcpp` is not desirable since it introduces a dependency on `ament_index_cpp`.
* Should there be a Python API for GenericSubscription/GenericPublisher?

@emersonknapp Perhaps you can offer some feedback here? I'll also gladly take you up on your offer on reviewing the code.

wjwwood, 2020-11-10:
> @wjwwood Could you or someone else clarify why "It only works if the type support for the type and matching rmw implementation is built locally and accessible to the current process (via the ament [...]

It's not in contrast. When I said "is built locally" I actually meant the shared libraries for that message type are available locally, either from apt-get or built from source or some other binary distribution. Sorry I wasn't more precise.

> I've implemented a relay prototype based on GenericSubscription/GenericPublisher, assuming that is the best way forward. Although I'd naively think that it might be possible to not use any type [...]

In theory yes, but you'd have to use some more complex DDS API's like X-Type's dynamic data API. We don't currently rely on that and there's no regular DDS API that gives you access to the message data without any type support what so ever. ROS 1 worked this way, you could subscribe to a topic without having any type information, neither as a Template argument nor as a string.

That's currently not possible in ROS 2 atm.

> I'd remove the GenericSubscription/GenericPublisher parts from rosbag2transport and make it depend on the new package if that sounds good. I'd assume putting GenericSubscription/GenericPublisher into rclcpp is not desirable since it introduces a dependency on ament_index_cpp.

A separate package is definitely ok, but it might make sense to place it in `rclcpp` as well. @ivanpauno @mabelzhang (the other ros2/rclcpp maintainers) what do you guys think?

> Should there be a Python API for GenericSubscription/GenericPublisher?

Yes, that would be nice to have.

> What is meant by "adaptive QoS" in the issue description?

I actually don't know this one.

nnmm, 2020-11-10:
Thank you @wjwwood, that mostly clears up what I need to do.

ivanpauno, 2020-11-10:
> A separate package is definitely ok, but it might make sense to place it in rclcpp as well. @ivanpauno @mabelzhang (the other ros2/rclcpp maintainers) what do you guys think?

:+1:, I think it makes sense to add them to rclcpp.

> Although I'd naively think that it might be possible to not use any type information at all and do it all on a lower level – theoretically no (de)serialization is needed, just forwarding a binary blob, right?

As @wjwwood commented, the only blocker is the lack of DDS API to do it.
Theoretically, you could create a subscription to a topic/type without the type support if you only want to access serialized data, but DDS doesn't support that AFAIU (you will always need a "type identifier" to access the correct topic, but you don't need to know how to serialize/deserialize that type if you're not going to do so).

nnmm, 2020-11-10:
@ivanpauno Just to clarify, you mean putting it in a new package in the rclcpp repo, next to rclcpp, rclcpp_action, rclcpp_components, and rclcpp_lifecycle? Or rclcpp proper?

nnmm, 2020-11-10:
We'll also need to put [typesupport_helpers](https://github.com/ros2/rosbag2/blob/foxy/rosbag2_cpp/src/rosbag2_cpp/typesupport_helpers.cpp) somewhere – maybe that's better suited for rcpputils?

ivanpauno, 2020-11-10:
> @ivanpauno Just to clarify, you mean putting it in a new package in the rclcpp repo, next to rclcpp, rclcpp_action, rclcpp_components, and rclcpp_lifecycle? Or rclcpp proper?

I think that it can be rclcpp or a new package in that repo, I don't mind much.

> We'll also need to put typesupport_helpers somewhere – maybe that's better suited for rcpputils?

`rcpputils` is for non-ros related utils, and typesupport is ros related.
I would add those utils in the same package were the generic publisher/subscription are going to be added (i.e. rclcpp or the new package in rclcpp repo, no need of a separate package for the typesupport helpers).

nnmm, 2020-11-10:
Great, I think I'll go with `rclcpp_generic` next to `rclcpp` etc.

emersonknapp, 2020-11-17:
> What is meant by "adaptive QoS" in the issue description?

In DDS (and therefore in ROS 2), the ability to subscribe to a topic is dependent on 3 things:
* Name of the topic (e.g. `/chatter`)
* Type of the topic (e.g. `std_msgs/String`)
* Quality of service: Compatibility of QoS Profile offered by publisher, vs QoS Profile requested by subscription

In order to generically subscribe to a topic in an "I don't care what the QoS is" way, you need more than just choosing the type of the topic, you need to understand what QoS profiles are being offered by the publishers in the system! This is what I mean by adaptive QoS.

One approach you could take is to request the **least restrictive possible qos** profile at all times. However, this means that you don't receive any of the benefits of a higher quality of service, even if all publishers in the system are offering it. 

In rosbag2 we currently have logic that chooses the highest-quality setting that will still be compatible with all known publishers. I would like to make this logic available to the rest of the system (for example, I think `ros2 topic echo --i-dont-care-qos` should exist, and probably be the default behavior!) I do have this on my backlog to try and contribute for G-turtle, but it's not the top of my backlog at the moment - I'd be happy to collaborate on exposing that functionality more generally. I think `rcl` would be a good place for it so that all client libraries can take advantage of it.

ivanpauno, 2020-11-17:
> One approach you could take is to request the least restrictive possible qos profile at all times. However, this means that you don't receive any of the benefits of a higher quality of service, even [...]

I think that in some cases you can have logic that chooses the QoS correctly, but for a lot of cases it's not obvious what is the "right" choice (DDS have a pretty "complicated" QoS matching logic).
But for things like `ros2 topic echo` or `topic_tools`, I agree that having some logic that almost always choose the right QoS is a good idea.

emersonknapp, 2020-11-17:
> I think that in some cases you can have logic that chooses the QoS correctly, but for a lot of cases it's not obvious what is the "right" choice 

From my evaluation, this is not the case for all QoS profiles exposed within the ROS 2 RCL API. I believe that there is an trivially-calculated correct choice for all options within the ROS 2 API. When there are multiple publishers on a given topic with different offered QoS profiles, there is still an easily-determinable choice that will match with all publishers. However, it may not get the advantage of some of the "higher quality" publishers (like guaranteed deliver and latched messages, for example). I believe the logic that I already built in rosbag2 is robust in this regard - it just needs to be exposed to the ROS 2 API in general

ivanpauno, 2020-11-17:
If you have two publishers, one with volatile durability and the other with transient local durability, what is the right choice for a relay node?

The subscription should obviously have transient local durability, but the publisher is not obvious.
The "correct way" is two create two, one with transient local durability and the other with volatile, and forward the message with one publisher or the other based of the durability of the original publisher.

I haven't think of all the combinations of other qos settings, but I don't think it's obvious.

emersonknapp, 2020-11-17:
> what is the right choice for a relay node

Oh, you're absolutely right - I was only considering the intake, not the output.

This problem also exists in rosbag2 for playback - it's not 100% accurate to the input because it "cheats down" - without a way to subscribe to individual publishers instead of topics... yeah I don't think there is a good answer to this. We just provided a way to override so the user could customize.

nnmm, 2020-11-20:
Thanks for explaining! So if I understand correctly, it's problematic to have one subscriber with "lowest" QoS because you want to take advantage of higher QoS where you can. And it's problematic to have one publisher with "highest" QoS because it might forward messages from a lower-QoS publisher to a higher-QoS subscriber, which wouldn't happen if they were directly connected?

I'll look into detecting the QoS of connected publishers and subscribers. By the way, a PR for the generic pub/sub exists and is waiting for a review: https://github.com/ros2/rclcpp/pull/1452

emersonknapp, 2020-11-24:
> Thanks for explaining! So if I understand correctly, it's problematic to have one subscriber with "lowest" QoS because you want to take advantage of higher QoS where you can. And it's problematic to [...]

That's exactly right - I think the relay can only be considered 100% correct if a subscription gets exactly the same behavior as if it subscribed to the original publisher. This is a difficult case given the potential mixed-qos situation. I think that an early solution to this could just be to refuse when the QoS of publishers are not all the same.

> I'll look into detecting the QoS of connected publishers and subscribers. 

Please see the logic existing in rosbag2.
* Adaptive subscription logic:
  * https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/qos.hpp
  * https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/qos.cpp
* Subscription side
  * https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L211

nnmm, 2020-11-26:
@wjwwood @mabelzhang @ivanpauno as the maintainers of `rclcpp`, should I add all or some of you as maintainers in the `package.xml` of the new `rclcpp_generic` package? Should I be a maintainer or author there?

ivanpauno, 2020-11-30:
> Should I be a maintainer or author there?

You should definitely add yourself as an `author`.
I guess it's fine if you add yourself as `maintainer` too, but you don't have to (maintainers get notifications from buildfarm failures).

> should I add all or some of you as maintainers in the package.xml of the new rclcpp_generic package?

Yes, all please.

wjwwood, 2020-11-30:
Hmm, I'm not 💯 percent happy with the separate package. I was thinking about when/if we split `rclcpp` up into parts, I would have a `rclcpp_topics`, or something similar, and I think the generic publishers would go there, not in a separate package. So in the meantime, I think it might make more sense for it to be part of `rclcpp`, especially if it is in this repository. Originally, I was thinking that we'd only use a separate package if we wanted it in a separate repository as well.

Don't act on this yet, I'd like to hear from @ivanpauno and/or @mabelzhang first.

ivanpauno, 2020-11-30:
> Don't act on this yet, I'd like to hear from @ivanpauno and/or @mabelzhang first.

Sounds good to me

wjwwood, 2020-11-30:
Ok, well @nnmm, unless you care to weigh in, let's drop the new package and just add the new API to `rclcpp` directly. I think that's more appropriate for now.

nnmm, 2020-12-16:
Sorry for the silence – sounds good to me, I'll edit the PR.

nnmm, 2021-04-13:
I'd suggest a new repository for topic_tools, since I didn't see an existing one where it would fit in (except maybe message_filters), and some of its [sibling packages in ROS 1](https://github.com/ros/ros_comm/tree/noetic-devel/tools) are now standalone repos. What do you think?

emersonknapp, 2021-04-13:
I would agree that a new `topic_tools` repository would be appropriate.

It might be worth considering managing `topic_tools` under the [Tooling WG](github.com/ros-tooling) umbrella - it's peripheral enough that I wouldn't expect it to be a dependency for core packages, but useful enough to have a place to triage issues, have design discussions, and manage releases.

nnmm, 2021-04-16:
I'm not sure what the implications of that organization vs ros2 are, but a priori I don't have a preference. Should I ping anyone specific to get consensus on that and set it up, @emersonknapp?

emersonknapp, 2021-04-16:
There's no real harm in creating it anywhere specific to start out, we can always transfer the repository to a different github organization - it's easier to do this before the first bloom release though, just to avoid any confusion.

I've created this PR https://github.com/ros-tooling/community/pull/30 to start the conversation, it can be discussed in the next WG meeting next Friday. 

I'll want feedback from Open Robotics folks before going ahead with anything, given that it was a `ros_comm` tool in ROS. Perhaps the ROS `topic_tools` maintainers @jacobperron @mjcarroll @sloretz would like to weigh in?

jacobperron, 2021-04-22:
A separate `topic_tools` package under the ros-tooling sounds okay to me. Unless, there were ideas about adding such functionality to `rclcpp` (@wjwwood)?

emersonknapp, 2021-04-22:
>  Unless, there were ideas about adding such functionality to rclcpp

My understanding was that design-wise we would want the topic tools to be component nodes - it would make sense, to me, to have those live in their own ros package, I'm not sure that vending specific-purpose utility nodes is in the scope of the rclcpp client. Open to alternate view though.

wjwwood, 2021-04-23:
Nah, nothing needs to be in rclcpp I think. Perhaps features will have to be added to rclcpp to implement them. But not the tools themselves.

emersonknapp, 2021-04-23:
Ok - after the WG discussion we agreed it makes sense to maintain it there - I've created https://github.com/ros-tooling/topic_tools repo and configured it for development. It's an empy `ament_cmake` package so far - PRs welcome!

Should we close this issue in favor of opening new feature requests on that repo moving forward?

jacobperron, 2021-04-27:
> Should we close this issue in favor of opening new feature requests on that repo moving forward?

Sounds reasonable to me.

clalancette, 2021-04-27:
> Should we close this issue in favor of opening new feature requests on that repo moving forward?

Yep, me too.  Going ahead and closing this one.

closed 2021-04-27 by clalancette

emersonknapp, 2021-04-27:
Cool - I've opened tickets at https://github.com/ros-tooling/topic_tools/issues for the major features that should be relatively easy to accomplish with the new `GenericPublisher`/`GenericSubscription` tools.