ros2/ros2 #861: [Eloquent]use_intra_process_comms dont work

dongxuanlb, 2020-01-28:
[component_container-1] terminate called after throwing an instance of 'std::invalid_argument'
[component_container-1]   what():  intraprocess communication allowed only with volatile durability

launch file :

vision_channel_pipeline = launch_ros.actions.ComposableNodeContainer(
        node_name="vision_channel_pipeline",
        package='rclcpp_components',
        node_namespace='',
        node_executable='component_container',
        emulate_tty=True,
        composable_node_descriptions=[
            ComposableNode(
                package='realsense_ros',
                node_plugin='realsense::RealSenseNodeFactory',
                node_name='realsense',
                parameters=[get_package_share_directory('vision_channel_node') + '/config/realsense/t265.yaml'],
                extra_arguments=[{'use_intra_process_comms': **True**}]),
            ComposableNode(
                package='realsense_ros',
                node_plugin='realsense::RealSenseNodeFactory',
                node_name='realsense',
                parameters=[get_package_share_directory('vision_channel_node') + '/config/realsense/d435i.yaml'],
                extra_arguments=[{'use_intra_process_comms': **True**}]),

dirk-thomas, 2020-02-03:
Please provide more context what exactly you do in what environment as well as reproducible steps.

dongxuanlb, 2020-02-05:
the environment is latest ros2(Eloquent), when i run :
https://github.com/intel/ros2_openvino_toolkit/blob/robot_devkit/openvino_node/launch/rs_od.launch.py

when set ComposableNode's extra_arguments=[_**{'use_intra_process_comms':'true'**_}, it will give:

[INFO] [realsense_node-1]: process started with pid [47019]
[realsense_node-1] [INFO] [camera]: Device with serial number 837212070949 was found.
[realsense_node-1] [INFO] [camera]: Create a node for D435i Camera
[realsense_node-1] _terminate called after throwing an instance of 'std::invalid_argument'_
[realsense_node-1] _what(): intraprocess communication allowed only with volatile durability_
[ERROR] [realsense_node-1]: process has died [pid 47019, exit code -6, cmd '/home/dongxuan/dev_ws/install/realsense_node/lib/realsense_node/realsense_node --ros-args --params-file /tmp/launch_params_k0e9tlse'].

and the node file is :

`namespace openvino
{
OpenVINOFactory::OpenVINOFactory(const rclcpp::NodeOptions & node_options)
: Node("openvino", "/", node_options)
{
  init();
}

OpenVINOFactory::OpenVINOFactory(const std::string & node_name, const std::string & ns, const rclcpp::NodeOptions & node_options)
: Node(node_name, ns, node_options)
{
  init();
}`

the lauch file is :

`def generate_launch_description():
    print(get_package_share_directory('openvino_node')+'/config/object_detection.yaml')
    container = ComposableNodeContainer(
            node_name='vision_pipeline',
            node_namespace='',
            package='rclcpp_components',
            node_executable='component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='realsense_ros',
                    node_plugin='realsense::RealSenseNodeFactory',
                    node_name='realsense',
                    parameters=[get_package_share_directory('realsense_examples')+'/config/d435i.yaml'],
                    extra_arguments=[{'use_intra_process_comms':'true'}]),
                ComposableNode(
                    package='openvino_ros',
                    node_plugin='openvino::OpenVINOFactory',
                    node_name='object_detection',
                    remappings=[('/rdk/openvino/detected_objects', '/openvino/detected_objects'), ('/rdk/openvino/image_raw', '/camera/color/image_raw')],
                    parameters=[get_package_share_directory('openvino_node')+'/config/object_detection.yaml'],
                    extra_arguments=[{'use_intra_process_comms':'true'}])
            ],
            output='screen',
    )
`

fujitatomoya, 2020-03-12:
@dongxuanlb 

> intraprocess communication allowed only with volatile durability

changing QoS would solve your issue, i do not think that it is supported right now.

closed 2020-03-12 by dirk-thomas

tgreier, 2020-08-26:
We are currently having the exact same problem. Is there a solution to this. The response from @fujitatomoya  does not describe a workaround.