ros2/ros2 #862: Using exact topic names

ivanpauno, 2020-01-28:
## Feature request

#### Feature description

This was triggered by a discussion in the implementation of [get_info_by_topic methods](https://github.com/ros2/rclcpp/pull/960#discussion_r368239648).
Related old issue: https://github.com/ros2/ros2/issues/496.

Summary of the current state:
- There's a way of ignoring ros conventions (i.e. avoiding adding `rt/` in topic names, etc) when creating a `Publisher/Subscription/Service/Client`, by using [ignore_ros_conventions](https://github.com/ros2/rmw/blob/980493fcb0878e335c3bcd524a1a5829ac9c3280/rmw/include/rmw/types.h#L334) qos option.
- In all the graph API, there's a way to specify if you want to query the "nice" (i.e.: demangled) ros names, or if you want to get the raw DDS names (or middleware specific names).
- The new `rmw_get_publishers_info_by_topic` has a combination of both usages. A topic name is passed, and part of the output is the type name. The first iteration used a `no_mangle` option, that indicates both if ros conventions have to be avoided or not, and wheather the type name is demangled or not (`no_mangle=false`: ros conventions + demangled output, `no_mangle=true`: don't apply ros conventions + show raw dds output without demangling).
- This is inconsistent with `rmw_count`, where there's no way to avoid ros conventions.
- `rclcpp` [count_publishers](https://github.com/ros2/rclcpp/blob/3ec882cd2db97d84330197bb946bd31572215489/rclcpp/include/rclcpp/node_interfaces/node_graph_interface.hpp#L84) does expansion, but does not remapping.

The issues are the following ones:
- Name expansion can't be avoided, so when avoiding ros conventions the result of a `foo` input is `/foo` (same as reported in https://github.com/ros2/ros2/issues/496).
- If `remapping` is applied to `publisher/subscription/etc` creation, it should also be applied in the graph query functions. Don't doing so can led to inconsistencies when composing nodes. e.g.: one node creates a publisher in a topic, another uses the graph query api. The first gets the topic name remapped, the later doesn't.

Targeted users:
- Normal ros users: wants expansion+remaping+ros conventions applied.
- DDS user: don't care about any of them.
- DDS user that likes ros features: don't want expansion and ros conventions, but wants to be able to use remapping.

To satisfy the three use cases, the following options might be implemented:
- Implement only the first two use cases, switching between them with a boolean. Expansion and remapping API should be available in all clients.
  The third use case can be manually done by using the `remapping` API and switching off expansion+remapping+ros conventions.
- Use an option with three states.
- Allow expansion/remapping/ros conventions to be configured individually.
- The `+exact` idea could be used: http://design.ros2.org/articles/topic_and_service_names.html.

My vote: option 1. It covers the three use cases, and it's the easier one to implement.
A three state option can be added later, if we have a real use case of that.

#### Implementation considerations

We should review all the related API and apply one of the options to all of them (graph API and `publisher/etc` creation).
It might worth it to create a function that does remapping and expansion in rcl, and wrap it in all the clients. There are many places where we do expansion and remapping combined, so we might get rid of some duplicated code.

ivanpauno, 2020-01-28:
Adding this to the `Foxy` project board, because I think we can consider addressing this inconsistency as part of the API review.
@wjwwood as reviewer, I hope I didn't skip anything from what we talked in the meeting.

clalancette, 2020-03-25:
One other consideration around this area: currently calling the rclpy method `get_topic_names_and_types` with `no_demangle=False` (the default) returns a list of demangled topic names.  Calling it with `no_demangle=True` returns the non-demangled topic *and* service names.  This is because services are built on top of topics.  But even though we asked for the non-demangled names, we still asked for *topic* names.  This is inconsistent.

The simple way to solve this is to make `no_demangle=True` return just the topic names; this would be consistent and most likely what the user wanted.  The more flexible way to do it would be to introduce a 4-way enum:

* DEMANGLE_TOPICS_ONLY - demangle the topic names, and the topic names only
* DEMANGLE_TOPICS_AND_SERVICES - demangle the topic names, but also return services (it's not entirely clear what this would mean)
* NO_DEMANGLE_TOPICS_ONLY - Don't demangle the topic names, and the topic names only
* NO_DEMANGLE_TOPICS_AND_SERVICES - Don't demangle the the topic and service names

dirk-thomas, 2020-03-25:
>  The more flexible way to do it would be to introduce a 4-way enum:

Coupling two orthogonal options into the same enum seems weird to me. The choice to demangle or not has nothing to do with the requested entities (topic or topic & services).

If you prefer a single function those can be passed as two separate arguments.

clalancette, 2020-03-25:
> If you prefer a single function those can be passed as two separate arguments.

Sure, that would work too.

wjwwood, 2020-03-25:
Should it also show pure DDS topics, i.e. topics lacking any prefix like `rt/`?

If so then I think the fact that the topics used to implement services being shown is expected. As they are pure DDS (in my opinion), still topics, and an implementation detail.

If not, then I think an existing use case cannot be served, i.e. listing non-ROS topics with the tool. This is why this option was added, IIRC.

A separate question is "why are you using this option at all" @clalancette? It should be in exceptional cases, and in those cases, I think you have to deal with the fact that there will be things in that list that are implementation dependent and things that cannot easily be subscribed to from the ROS API.

clalancette, 2020-03-25:
> A separate question is "why are you using this option at all" @clalancette? It should be in exceptional cases, and in those cases, I think you have to deal with the fact that there will be things in [...]

The reason I wanted to do this was so that I could use the Fast-RTPS XML configuration file to set QoS settings on a per-topic basis.  There I need to know the true DDS topic name so I can match it in the XML.  I thought it would be a useful-in-general option, so I started implementing a `--no-demangle` option to `ros2 topic list`, and then I noticed that it printed a lot of unrelated stuff.

wjwwood, 2020-03-25:
That's a case where you would, in my opinion, need to deal with things like implementation detail topics, unrelated non-ROS topics (like builtin topics), or different answers based on the rmw implementation you're using.

In short, I don't see a problem with the current behavior to be honest.

mmatthe, 2020-10-04:
Is there any update on this? I was recently trying to use `avoid_ros_namespace_conventions` to create a DDS topic with one exact name, however, ROS2 Foxy always prepends a `/` to the topic name:

```python
        qos = rclpy.qos.QoSProfile(avoid_ros_namespace_conventions=True, depth=10)
        self._sub1 = self.create_subscription(String, "my_raw_topic_sub", handler, qos)
        self._sub2 = self.create_subscription(String, "my_topic_sub", handler, 10)

        self._pub1 = self.create_publisher(String, "my_raw_topic_pub", qos)
```

yields this output in e.g. the RTI Admin Console:

[image]

wjwwood, 2020-10-04:
> Is there any update on this?

No, sorry.

> I was recently trying to use avoid_ros_namespace_conventions to create a DDS topic with one exact name, however, ROS2 Foxy always prepends a / to the topic name

That's expected behavior right now. The `avoid_ros_namespace_conventions` only prevents the `rr` prefix.

ivanperez-keera, 2024-01-26:
Is this a more general version of: https://github.com/ros2/ros2/issues/496?

clalancette, 2024-01-26:
> Is this a more general version of: #496?

Yes (it is stated in the opening comment).