ros2/ros2 #864: Ros2 Eloquent on Mac Mojave building Error - package 'rviz_ogre_vendor' failed

meiTob, 2020-01-31:
**Required Info:**

- Operating System:
  - MacOs 10.4.6 Mojave

- Installation type:
  - from source (as binary wasn´t working either. It had some load entry point errors and someone told me to build from source

- Version or commit hash:
  -  Eloquent

- DDS implementation:
  - don´t realy know

- Client library (if applicable):
  - rviz_ogre_vendor

#### Steps to reproduce issue

- I completly clean installed Mojave and before doing anything else followed this guide:
https://index.ros.org/doc/ros2/Installation/Eloquent/OSX-Development-Setup/

The step: 
**cd ~/ros2_eloquent/
colcon build --symlink-install**
gives me Errors
#### Expected behavior
Building success

#### Actual behavior

Im getting following Error Messages:

```
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> fastcdr
Starting >>> foonathan_memory_vendor
Finished <<< ament_package [2.30s]                             
Starting >>> ament_cmake_core                               
Finished <<< foonathan_memory_vendor [2.60s]                
--- stderr: fastcdr                                         
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: for: /Users/tobiasmeinert/ros2_eloquent/install/fastcdr/lib/libfastcdr.1.0.10.dylib (for architecture x86_64) option "-add_rpath /Users/tobiasmeinert/ros2_eloquent/install/fastcdr/lib" would duplicate path, file already has LC_RPATH for: /Users/tobiasmeinert/ros2_eloquent/install/fastcdr/lib
---
Finished <<< fastcdr [2.73s]
Starting >>> fastrtps                                       
Starting >>> poco_vendor
Finished <<< ament_lint [2.90s]                             
Starting >>> ament_flake8
Finished <<< ament_cmake_core [1.87s]                                     
Starting >>> ament_cmake_libraries                                   
Finished <<< poco_vendor [1.86s]                                  
Starting >>> ament_cmake_export_libraries                         
Finished <<< ament_cmake_libraries [0.88s]                        
Starting >>> ament_cmake_include_directories
Finished <<< ament_cmake_export_libraries [0.72s]                 
Starting >>> ament_cmake_python                                        
Finished <<< ament_flake8 [3.05s]                                      
Starting >>> ament_cmake_export_dependencies
Finished <<< fastrtps [3.27s]
Starting >>> ament_cmake_export_definitions
Finished <<< ament_cmake_include_directories [1.40s]
Starting >>> ament_cmake_export_include_directories                          
Finished <<< ament_cmake_export_dependencies [0.72s]                         
Finished <<< ament_cmake_python [1.35s]
Finished <<< ament_cmake_export_definitions [0.81s]
Starting >>> ament_cmake_test
Starting >>> ament_cmake_export_link_flags
Starting >>> ament_cmake_version         
Finished <<< ament_cmake_export_include_directories [1.36s]
[... 145 lines cut ...]
#   include "Availability.h"
            ^~~~~~~~~~~~~~~~
1 error generated.
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreASTCCodec.cpp.o] Error 1
make[4]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [ogre-v1.12.1-prefix/src/ogre-v1.12.1-stamp/ogre-v1.12.1-build] Error 2
make[1]: *** [CMakeFiles/ogre-v1.12.1.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
---
Failed   <<< rviz_ogre_vendor	[ Exited with code 2 ]
Aborted  <<< resource_retriever                          
Aborted  <<< urdf                                        
Aborted  <<< rcutils                                                 

Summary: 83 packages finished [1min 15s]
  1 package failed: rviz_ogre_vendor
  3 packages aborted: rcutils resource_retriever urdf
  3 packages had stderr output: fastcdr orocos_kdl rviz_ogre_vendor
  211 packages not processed

```

Does anyone know what causes this error, and what I need to do to get rid off it. Or should i maybe install another distribution? I know there are already similar issues already, but they all seem to work now, while Im still getting the error. I am kind of a Neewbie, if this Post here don´t belong here feel free to tell me, and maybe tell me where else I could get help.

mikelynders, 2020-02-02:
I am also having the same issue on Catalina (i know its not supported but I was trying to build it from source)

hirotaka001, 2020-02-03:
I had the same problem. However, it was solved by the following command.

`export SDKROOT="$(xcrun --sdk macosx --show-sdk-path)"`

I hope your problem will be solved.