ros2/ros2 #866: choco install python installs python 3.8 on windows 10

flynneva, 2020-02-11:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - binaries

#### Steps to reproduce issue
- running the following command on Windows 10 installed python version 3.8.
- for ROS2 Dashing patch 5, ros2-script.py file in the Scripts directory calls for specifically python 3.7
- causes demo/getting started instructions to fail
```
choco install python
```
then run
```
ros2 run demo_nodes_cpp talker
```
**without python 3.7 installed** you will get an error.

#### Expected behavior
- demo talker should work

#### Actual behavior
- prints 'failed to create process'

## Feature request
- specify in [installation instructions](https://index.ros.org/doc/ros2/Installation/Dashing/Windows-Install-Binary/) to install python 3.7
- or remove python37 from path in ros2-script.py

flynneva, 2020-02-11:
#771 , #821  same issues....

[looks like a PR](https://github.com/ros2/ros2_documentation/pull/418) was already merged in for Eloquent documentation to fix this...we should probably bring this over to Dashing because Eloquent wont be supported past this fall.

tfoote, 2020-02-20:
There's an PR to update dashing install instructions too: https://github.com/ros2/ros2_documentation/pull/534

closed 2020-02-22 by flynneva