ros2/ros2 #867: remove use of poco for portably loading shared libraries

wjwwood, 2020-02-12:
## Feature request

#### Feature description

Currently we use [the poco library](https://pocoproject.org/) for loading shared libraries dynamically at runtime. However, poco provides a lot of functionality beyond loading shared libraries and is therefore a relatively large dependency for what we use it for.

Several times we have been asked to remove poco in favor of a smaller library or a custom implementation of the functionality we need.

There are several places we use poco for this functionality:

- `rmw_implementation`: optionally used to load rmw implementations when more than one is available
  - https://github.com/ros2/rmw_implementation/blob/afe5761bddc3ca47994957d72e5514c291393a85/rmw_implementation/CMakeLists.txt#L55
- `rosidl_typesupport_c`/`rosidl_typesupport_cpp`: optional counter part to `rmw_implementation` for loading various type supports
  - https://github.com/ros2/rosidl_typesupport/blob/4847322f6a03e5a2ec5cf0498ec8d93d0be8f7b4/rosidl_typesupport_c/CMakeLists.txt#L21
  - https://github.com/ros2/rosidl_typesupport/blob/4847322f6a03e5a2ec5cf0498ec8d93d0be8f7b4/rosidl_typesupport_cpp/CMakeLists.txt#L17
- `class_loader`: required to load classes from shared libraries
  - https://github.com/ros/class_loader/blob/264841b10c94e8ba43a3197b95e333f0d915d0a7/CMakeLists.txt#L22
- `poco_vendor`: vendor package that ensure poco is available, building it if necessary
  - https://github.com/ros2/poco_vendor/blob/546f40ac6ca3efcdc47ac5c244f37a3011b42c3f/CMakeLists.txt#L11
- `rosbag2`: used to dynamically load typesupport libraries at runtime as needed
  - https://github.com/ros2/rosbag2/blob/ec4fb3913233df00850b53a4422d9b804fbab9d7/rosbag2_converter_default_plugins/CMakeLists.txt#L26

The idea would be to replace poco with a smaller library that provides the same functionality, or add our own implementation of these few functions we use to a new package or a package like `rcutils`.

#### Implementation considerations

Unless poco is being used in some way that I'm not aware of, this means the deprecation and removal of `poco_vendor`. Or at least removing it from our default set of packages and all dependencies on it.

Some packages may also need to be updated to avoid catching Poco exceptions and things like that.

This is a summary of the hits for `poco` in my local copy of ROS 2, case-insensitive:

<details>
<pre>
Searching 9331 files for "poco"

/Users/william/ros2_ws/src/micro-ROS/ros_tracing/ros2_tracing/all.repos:
  184      url: https://github.com/ros2/orocos_kinematics_dynamics.git
  185      version: ros2
  186:   ros2/poco_vendor:
  187      type: git
  188:     url: https://github.com/ros2/poco_vendor.git
  189      version: master
  190    ros2/rcl:

/Users/william/ros2_ws/src/ros/class_loader/.appveyor.yml:
   34    - cd ros2
   35    - git clone https://github.com/ros2/console_bridge_vendor.git
   36:   - git clone https://github.com/ros2/poco_vendor.git
   37    - cd ..\..\..
   38    - call "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Auxiliary\Build\vcvarsall.bat" x86_amd64

/Users/william/ros2_ws/src/ros/class_loader/.travis.yml:
   19      if [[ "$TRAVIS_OS_NAME" == "linux" ]]; then
   20          sudo apt-get update -qq
   21:         sudo apt-get install cmake cppcheck libxml2-utils libpoco-dev python3 python3-pip python3-empy python3-pyparsing python3-setuptools wget
   22      else
   23          brew update
   24          brew outdated cmake || brew upgrade cmake || brew install cmake
   25:         brew install cppcheck console_bridge poco uncrustify
   26          brew unlink python@2
   27          brew outdated python3 || brew upgrade python3 || brew install python3
   ..
   44    - cd $AMENT_WS_SRC/ros2
   45    - git clone https://github.com/ros2/console_bridge_vendor.git
   46:   - git clone https://github.com/ros2/poco_vendor.git
   47    - cd $AMENT_WS
   48    # Link source into workspace's source space

/Users/william/ros2_ws/src/ros/class_loader/CHANGELOG.rst:
[... 569 lines cut ...]
   26  
   27  ament_export_dependencies(rosidl_typesupport_interface)
   ..
   36    src/service_type_support_dispatch.cpp
   37    src/type_support_dispatch.cpp)
   38: if(Poco_FOUND)
   39    target_compile_definitions(${PROJECT_NAME}
   40:     PRIVATE "ROSIDL_TYPESUPPORT_C_USE_POCO")
   41  endif()
   42  if(WIN32)
   ..
   49  )
   50  
   51: if(Poco_FOUND)
   52:   ament_target_dependencies(${PROJECT_NAME} "Poco")
   53  endif()
   54  ament_target_dependencies(${PROJECT_NAME} "rcpputils" "rosidl_generator_c")

/Users/william/ros2_ws/src/ros2/rosidl_typesupport/rosidl_typesupport_c/package.xml:
   11  
   12    <depend>rcpputils</depend>
   13:   <build_depend>libpoco-dev</build_depend>
   14:   <build_depend>poco_vendor</build_depend>
   15    <build_depend>rosidl_generator_c</build_depend>
   16

gavanderhoorn, 2020-02-12:
[Boost.DLL](https://www.boost.org/doc/libs/1_69_0/doc/html/boost_dll.html)! :upside_down_face: `\( ﾟヮﾟ)/`.

ahcorde, 2020-03-10:
Issue to track the progress of this change https://github.com/ros2/rcutils/issues/216

dirk-thomas, 2020-04-14:
@ahcorde Can this ticket be closed?

closed 2020-04-15 by ahcorde