ros2/ros2 #868: Dashing/Eloquent: Incompatibility between most recent VS installations and fast-RTPS <= 1.9.3

brawner, 2020-02-13:
This fix was already addressed in https://github.com/eProsima/Fast-RTPS/pull/932, but there is a known incompatibility between Visual Studio's VCTool 16.24 and fast-RTPS versions <= 1.9.3.

See https://github.com/eProsima/Fast-RTPS/issues/927 for more information.

I believe that Dashing and Eloquent may need to target a more recent version of Fast-RTPS with the bug fix.

dirk-thomas, 2020-02-13:
For Eloquent we have already identified a couple other FastRTPS related issues which will likely make us use a newer version of FastRTPS soon (something around current head or to-be-released 1.9.5).

For Dashing it is highly unlikely that we will significantly bump FastRTPS - simply because FastRTPS has significantly changed compared to the version pinned in Dashing. The only option would be to create a fork of FastRTPS and cherry-pick a specific patch necessary on top of the pinned version of FastRTPS for Dashing.

brawner, 2020-02-13:
I figured it would be a harder change for Dashing. Microsoft doesn't keep older versions of Visual Studio Community. If changing FastRTPS would be difficult, then the source install instructions should be updated to reflect this incompatibility.

There are two possible workarounds for users, they can build with VS 2017 or install an older version of VS Buildtools 2019 and work with that. MS does host older versions of the same year of Buildtools (https://docs.microsoft.com/en-us/visualstudio/releases/2019/history). The install instructions are slightly different for Buildtools than they are for Community though to get the right components.

MiguelCompany, 2020-02-14:
@brawner AFAIK, the incompatibility was introduced on Fast RTPS 1.9.0, so Dashing should not be affected.

mjcarroll, 2020-03-02:
Since there is now a `ros2-eloquent` branch in eProsima/Fast-RTPS, I'm  backporting the fix for this to Eloquent: https://github.com/eProsima/Fast-RTPS/pull/1040

ahcorde, 2020-03-02:
It works!. Tested on CI https://ci.ros2.org/job/ci_windows-container/256/

brawner, 2020-03-04:
Closing, as I believe this has been addressed with: 

https://github.com/eProsima/Fast-RTPS/pull/1040
https://github.com/ros2/ros2/pull/872

closed 2020-03-04 by brawner