ros2/ros2 #870: dlopen Library not Loaded /usr/local/opt/poco/lib/libPocoFoundation.63.dylib

JMarple, 2020-02-14:
- Operating System:
  - Mac 10.14.6 
- Installation type:
  - Binary (https://github.com/ros2/ros2/releases/tag/release-eloquent-20200124)

#### Steps to reproduce issue
I followed the steps shown here:
https://index.ros.org/doc/ros2/Installation/Crystal/OSX-Install-Binary/

After installation, running `ros2 -h` returns the errors below.  The demo py/cpp scripts have the same errors.  I attempted to reinstall poco/pcre with brew with no luck.  

Also: 
```csrutil status
System Integrity Protection status: disabled.
```

```Failed to load entry point 'launch': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'info': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'list': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'send_goal': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'show': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'play': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'record': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'list': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'load': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'standalone': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'types': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'unload': dlopen(/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_crystal/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_crystal/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-e [... cut ...]
  param      Various param related sub-commands
  pkg        Various package related sub-commands
  run        Run a package specific executable
  security   Various security related sub-commands
  service    Various service related sub-commands
  srv        Various srv related sub-commands
  test       Run a ROS2 launch test
  topic      Various topic related sub-commands
  trace      Trace ROS nodes to get information on their execution
  wtf        Use `wtf` as alias to `doctor`

  Call `ros2 <command> -h` for more detailed usage.
```

dirk-thomas, 2020-02-14:
You are claiming to use Eloquent but your error messages are showing Crystal (which is already EOL)?

JMarple, 2020-02-14:
Ah thank you, I somehow missed there were different install instructions.  

I rm'd the ros2_crystal files and went through the Eloquent instructions (https://index.ros.org/doc/ros2/Installation/Eloquent/OSX-Install-Binary/), but getting the same errors unfortunately 

```
ros2 -h
Failed to load entry point 'launch': dlopen(/Users/justinmarple/ros2_eloquent/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_eloquent/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_eloquent/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'info': dlopen(/Users/justinmarple/ros2_eloquent/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_eloquent/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_eloquent/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'list': dlopen(/Users/justinmarple/ros2_eloquent/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_eloquent/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found
The C extension '/Users/justinmarple/ros2_eloquent/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'send_goal': dlopen(/Users/justinmarple/ros2_eloquent/ros2-osx/lib/python3.7/site-packages/rclpy/_rclpy.cpython-37m-darwin.so, 2): Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
  Referenced from: /Users/justinmarple/ros2_eloquent/ros2-osx/lib/librosidl_typesupport_c.dylib
  Reason: image not found

etc etc, same as before. 
```

JMarple, 2020-02-14:
```
JustinMarple-C02XKJPJJG5H:lib justinmarple$ pwd
/usr/local/opt/poco/lib
JustinMarple-C02XKJPJJG5H:lib justinmarple$ ls
cmake				libPocoDataSQLite.70.dylib	libPocoFoundation.dylib		libPocoMongoDB.70.dylib		libPocoNetSSL.dylib		libPocoXML.70.dylib
libPocoCrypto.70.dylib		libPocoDataSQLite.dylib		libPocoJSON.70.dylib		libPocoMongoDB.dylib		libPocoRedis.70.dylib		libPocoXML.dylib
libPocoCrypto.dylib		libPocoEncodings.70.dylib	libPocoJSON.dylib		libPocoNet.70.dylib		libPocoRedis.dylib		libPocoZip.70.dylib
libPocoData.70.dylib		libPocoEncodings.dylib		libPocoJWT.70.dylib		libPocoNet.dylib		libPocoUtil.70.dylib		libPocoZip.dylib
libPocoData.dylib		libPocoFoundation.70.dylib	libPocoJWT.dylib		libPocoNetSSL.70.dylib		libPocoUtil.dylib
JustinMarple-C02XKJPJJG5H:lib justinmarple$ 
```

Looks like I might have mis-matching versions from brew

JMarple, 2020-02-14:
Similar when trying to run ros2, I get 

```dyld: Library not loaded: /usr/local/opt/assimp/lib/libassimp.4.dylib
  Referenced from: /Users/justinmarple/ros2_eloquent/ros2-osx/lib/rviz2/rviz2
  Reason: image not found
```

```
JustinMarple-C02XKJPJJG5H:~ justinmarple$ ls /usr/local/opt/assimp/lib/
cmake			libIrrXML.dylib		libassimp.5.0.0.dylib	libassimp.5.dylib	libassimp.dylib		pkgconfig
```

Is there a proper way of managing on these versions?

dirk-thomas, 2020-02-14:
I will need to install the same version as used by the precompiled binaries. Since macOS / brew is a rolling distribution these kind of changes are happening frequently.

@mjcarroll Either the install instructions should be updated to use a pinned version or the macOS machines needs to be updated to use the latest version.

nuclearsandwich, 2020-02-14:
@JMarple which version of poco do you currently have installed? `brew info poco` will tell you.

JMarple, 2020-02-14:
```
$ brew info poco
poco: stable 1.10.0 (bottled), HEAD
C++ class libraries for building network and internet-based applications
https://pocoproject.org/
/usr/local/Cellar/poco/1.10.0 (879 files, 14.9MB) *
  Poured from bottle on 2020-02-14 at 13:44:17
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/poco.rb
==> Dependencies
Build: cmake ✔
Required: openssl@1.1 ✔
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 853 (30 days), 1,808 (90 days), 6,320 (365 days)
install-on-request: 719 (30 days), 1,537 (90 days), 5,187 (365 days)
build-error: 0 (30 days)
```

balzes, 2020-03-20:
On MacOS Catalina 10.15.3
to solve:
`Library not loaded: /usr/local/opt/poco/lib/libPocoFoundation.63.dylib`
for me helped creating a symbolic link:
```
ln -s /usr/local/opt/poco/lib/libPocoFoundation.71.dylib /usr/local/opt/poco/lib/libPocoFoundation.63.dylib
```

in my case
`ls /usr/local/opt/poco/lib`
 contained `*.71.dylib` instead of yours `*.70.dylib` files

this solution was found here: https://stackoverflow.com/a/60379217/6097221

mjcarroll, 2020-03-20:
Just as a note, we don't currently support `Catalina` for `eloquent` or `foxy`, only `mojave`.

mjcarroll, 2020-03-20:
The ROS2 buidfarm is currently using `poco stable 1.10.1`

mjcarroll, 2020-03-20:
I'm going to close this as our `macOS` builds are now using the latest version of poco, so there should be no need to rely on using a pinned version.

I'll make a note that checking for `brew` updates should be a regular part of the buildfarmer task.

mjcarroll, 2020-03-20:
I take that back, this will be resolved when we build new binaries for the second `eloquent` patch release, which should happen relatively soon (in the next week or two).

In the meantime, I suggest either building from source, or using the symlink workaround that @balzes posted.  I'm not sure of the potential side effects of the symlink approach, but they could exist.