ros2/ros2 #875: ROS2 Eloquent: Pointcloud2 crashes (core dumped) with increasing large width size

JEGroeneweg, 2020-03-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries 

#### Steps to reproduce issue
Create a Pointcloud2 msg which increases in size every callback. As a test I published a Pointcloud2 msg with a starting width of 4000 which I increased with 50 every iteration of the callback. The error did not occur with constant large sizes. 

```
class PublisherNode : public rclcpp::Node
{
public:
  PublisherNode() : Node("minimal_publisher_node"), t_(int64_t{ 0 }), nPt_(uint64_t{ 4000 })
  {
    publisher_ = this->create_publisher<sensor_msgs::msg::PointCloud2>("/test_pointcloud", rclcpp::QoS(10));

    auto timer_callback = [this]() -> void {
      t_ += 1e9;
      nPt_ += 50;
      auto pc = generate_point_cloud(nPt_, t_);
      this->publisher_->publish(*pc);
    };
    timer_ = this->create_wall_timer(250ms, timer_callback);
  }

private:
  rclcpp::Publisher<sensor_msgs::msg::PointCloud2>::SharedPtr publisher_;
  rclcpp::TimerBase::SharedPtr timer_;
  int64_t t_;
  uint64_t nPt_;
};
```
The generate pointCloud function generates a pointcloud with random data, has 9 fields all of type PointField::FLOAT32,  point_step: sizeof(PointField::FLOAT32), row_step: number_of_points * point_step, bigendian = false, is_dense = false

#### Expected behavior
Using the dashing distribution I did not experience any problems. (no code changes between distributions)

#### Actual behavior
After some time, width passes the 5000, the width when the error occurs is not constant.

```
free(): invalid next size (normal)
Aborted (core dumped)
```

dirk-thomas, 2020-03-04:
This might be the same as ros2/ros1_bridge#240. Please see this comment: https://github.com/ros2/ros1_bridge/issues/240#issuecomment-594223000

There has been a patch to Fast-RTPS recently which addressed a memory problem. What version of `ros-eloquent-fastrtps` are you using? Up to 1.9.3-1 has the memory problem. As of version 1.9.3-2 it should work correctly.

JEGroeneweg, 2020-03-05:
I indeed still used the 1.9.3-1 version. I updated and tested it and so far nothing has crashed while having a Pointcloud2 msg with a width up to 30.000 points. Thanks.

closed 2020-03-05 by JEGroeneweg