ros2/ros2 #878: Install Failed

YoJn, 2020-03-11:
## System: ubuntu18.04
- Install the ros2 version: Eloquent Elusor
- When I used colcon to compile the source package, I got the following problem

[image]

- If the above picture is not clear, what I intercept below is the main part

[image]

- Looking forward to your answer! Thanks for your help !

claireyywang, 2020-03-19:
Hi, what's your OS version? Are you following instructions here https://index.ros.org/doc/ros2/Installation/Eloquent/ ? Could you provide more specific steps for us to reproduce this error?

YoJn, 2020-03-19:
@claireyywang Thanks your reply
- System environment: linux ubuntu18.04
- I installed it according to the link below
 > https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/

- There was a problem executing the following command
> colcon build --symlink-install

closed 2020-05-13 by YoJn