ros2/ros2 #882: Dashing Patch Release 7

nuclearsandwich, 2020-03-19:
This issue will track the broad status of the next Dashing patch release. The date for this patch release is not currently set and will be determined by a critical mass of desired changes ready to be released.

We're going to use a GitHub Project board for this patch release again: [Dashing Patch Release 7](https://github.com/orgs/ros2/projects/19)

As with prior releases, if you want to contribute fixes please open pull requests targeting the latest development branch and mention to the reviewer that you'd like the pull request to be considered for backporting into Dashing.

MiguelCompany, 2020-04-07:
@nuclearsandwich Could I suggest to bump Fast CDR to v1.0.13, and Fast-RTPS to v1.8.4?

nuclearsandwich, 2020-04-07:
> Could I suggest to bump Fast CDR to v1.0.13, and Fast-RTPS to v1.8.4?

From a quick glance those should be backportable releases. I see from the changelogs that these releases contain mostly bugfixes. @MiguelCompany are these changes all compatible with the current Dashing release of rmw_fastrtps and are they ABI/API compatible with the previously released versions of Fast-CDR (1.0.11) and Fast-RTPS (1.8.2)? We don't intentionally break ABI within a released distribution unless there are extraordinary circucmstances.

nuclearsandwich, 2020-04-07:
@ThijsSassen with the releases in https://github.com/ros/rosdistro/pull/24404 and https://github.com/ros/rosdistro/pull/24405 could you please provide a summary of the major changes since the last releases?

ThijsSassen, 2020-04-07:
@nuclearsandwich The main changes since the last release are support for deadline, lifespan and liveliness qos which makes cyclonedds more feature complete.

nuclearsandwich, 2020-07-22:
Packaging jobs:

* Ubuntu Bionic AMD64 [image]
* Ubuntu Bionic ARM64 [image]
* Ubuntu Bionic ARMHF [image]
* CentOS AMD64 [image]
* MacOS :no_entry_sign: No longer supported in Dashing
* Windows AMD64 [image]
* Windows Debug AMD64 [image]

nuclearsandwich, 2020-07-22:
The sync is [out](https://discourse.ros.org/t/new-packages-and-patch-release-for-dashing-diademata-2020-07-22/15541)!

closed 2020-07-22 by nuclearsandwich