ros2/ros2 #883: Different sigint behavior for cpp and python talker in bash background

Parallel-Hao, 2020-03-23:
## Bug report
I write a bash script to run a talker program in background, then send sigint to the talker process. The weird thing is that the talker has different behavior for python and cpp.

**Required Info:**

- Operating System:
  -  Ubuntu 18.04
- Installation type:
  -  binaries
- Version or commit hash:
  -  foxy
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  -  rclcpp and rclpy

#### Steps to reproduce issue
The test.bash file:
```
# For python test:
python3 /opt/ros/foxy/lib/python3.6/site-packages/demo_nodes_py/topics/talker.py &

# For cpp test:
#../opt/ros/foxy/lib/demo_nodes_cpp/talker &
```
Reproduce in terminal:
1. bash test.bash
2. find the talker pid, and kill -2 talker_pid

#### Expected behavior
cpp test: talker is killed.
python test: talker is killed.
#### Actual behavior
cpp test: talker is killed.
python test: talker continues running.
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

claireyywang, 2020-04-02:
closing due to duplicate

closed 2020-04-02 by claireyywang

sloretz, 2020-04-02:
Duplicate of ros2/rclpy#527